raspberrypi/linux #1347: 4.4.5 loads rtc-ds1307 but doesn't activate it.

DougieLawson, 2016-03-14:
To get my RTC running I had to use
```
echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
```

Config.txt has
```
dtoverlay=i2c-rtc,ds1307
gpu_mem=16
initramfs initrd.sda followkernel
max_usb_current=1
dtparam=i2c=on,spi=on
```
As soon as I do the echo command it all comes alive
```
[130676.800914] rtc-ds1307 1-0068: rtc core: registered ds1307 as rtc0
[130676.801133] rtc-ds1307 1-0068: 56 bytes nvram
[130676.801335] i2c i2c-1: new_device: Instantiated device ds1307 at 0x68
```

pelwell, 2016-03-14:
It is more likely that the module isn't loaded. What does `sudo vcdbg log msg |& grep -v -E  "(gpioman|HDMI|clock):"` show?

DougieLawson, 2016-03-15:
It's been booted since 09:22 on 13th, so it looks like your debug log isn't available
```
root@apollo /tmp # vcdbg log msg |& grep -v -E "(gpioman|HDMI|clock):"
Unable to determine the value of __LOG_START
Unable to read logging_header from 0x00000000
root@apollo /tmp #
```

It'll be Friday night before I can reboot it and do more testing.

DougieLawson, 2016-03-18:
Hi Phil,

I've just updated to 4.4.6 and the RTC stuff worked. Thanks.
```
[   30.546305] rtc-ds1307 1-0068: rtc core: registered ds1307 as rtc0
[   30.548262] rtc-ds1307 1-0068: 56 bytes nvram
```

Looks like we can close this issue.

closed 2016-03-18 by DougieLawson