raspberrypi/linux #1351: 7" touchscreen display reinitialization while Pi keeps running?

SebTC, 2016-03-18:
The official 7" touchscreen display works great, but is currently not hot plugable. The Pi + touchscreen run on a big battery where the Pi must run always and the touchscreen only for a short time. It is already possible to switch off the backlight, but the rest of the screen still consumes around 100mA.
I can switch the screen power supply off with external components, but turning it on again won't work (no initialization).
If I understand it correctly, the initialization of the screen (+touch) happens by the GPU over i2c-0. Is it possible to make the screen turn on again, by modifying or writing drivers? Don't know if this initialization is done by the GPU firmware or software over mailbox, and if i can trigger the software to redo it.

Another option would be to send a command to the screen hardware/attiny to go into low power mode (just like the backlight PWM value). Don't know if this is coming or is possible.

Thank you for the suggestions/help.

popcornmix, 2016-03-20:
Really dropping power to a connected peripheral that is not designed to be hot-pluggable is not advisable.It certainly won't work currently and I doubt it is something we will support.

It's not just the initialisation - we may have been in the middle of an I2C operation to read the touchscreen when power was dropped, which may cause issues.

Ping @ghollingworth

SebTC, 2016-03-20:
Seems logical. Before the screen is initialized over i2c, the current @5V to the screen hardware is less than 20mA. Is it then possible to bring the IC's on the board to the (old) standby state with a function or i2c command, or is this something what has to be changed in the firmware? Would also be great just to put the screen in standby mode and when needed back to active mode.

Guess this is not (yet) supported by the firmware, but is it possible?

closed 2017-05-18 by JamesH65