raspberrypi/linux #1352: No audit support in raspbian

rfrht, 2016-03-18:
When trying to set up a audit rule for a given file:

~~~
[root@rasp-rodhome etc]# auditctl -w /DataVolume -p r 
Error - audit support not in kernel
Cannot open netlink audit socket
~~~

Looking for `audit` support in Raspbian kernel:

~~~
[root@rasp-rodhome etc]# modprobe configs ; gunzip -dc /proc/config.gz | grep AUDIT
# CONFIG_AUDIT is not set
# CONFIG_AUDIT_ARCH_COMPAT_GENERIC is not set
~~~

Is it really intended? If it is not, please consider enabling `CONFIG_AUDIT` in next kernel builds.

Ferroin, 2016-03-18:
It probably is intended, auditing support is a seriously huge amount of code (compared to some other features) that would likely provide absolutely zero benefit for a majority of users, and also can't be built as a loadable module.

popcornmix, 2016-03-18:
Yes, it's disabled deliberately. Expensive settings with little benefit to the majority of users are not enabled on the Pi. CONFIG_AUDIT is not clearly a good thing:
https://lwn.net/Articles/600568/
https://lwn.net/Articles/600578/

If you want it enabled you will need to provide justification including measures of cpu and memory usage with and without it.

rfrht, 2016-03-18:
> Yes, it's disabled deliberately. Expensive settings with little benefit to the majority of users are not enabled on the Pi. CONFIG_AUDIT is not clearly a good thing:

OK then, considering that its `Kconfig` is a bool (no module-loading supported), I'll happily let this go. I scripted a workaround for what I needed.

Thanks for your time!

closed 2016-03-18 by rfrht

devinbfergy, 2019-10-30:
@rfrht What work around did you configure?

rfrht, 2019-10-30:
@devinbfergy I wanted to see what was actually being accessed in my /DataVolume, I think I tackled that by using inotify