raspberrypi/linux #1353: USB sounds sometimes fail with "FIQ timed out. Data may have been lost."

andrewsil1, 2016-03-19:
I'm using a Pi 3 at stock speeds with RetroPie (latest), and 3-15 rpi-update kernel.  Unclear whether it's Pi 3 specific or a Jessie problem, though.  Default sound card in this case is a generic C-media Headphone Set from Adafruit, using snd_usb_audio module.

The other 3 external USB ports are consumed with a keyboard and 2 joystick/kbd HID devices (game controllers).

If I turn on sounds within the Emulationstation shell (e.g. Carbon theme, and then enable sounds), when the app tries to play a sound for the first time, I will relatively frequently (not 100% repro) lose the function of all of the USB devices (including keyboard too), and the message in the subject will show up at the end of the dmesg log, e.g. "Transfer to device 4 endpoint 0x2 frame 578 failed - FIQ times out. Data may have been lost."  Never used to see this behavior under Wheezy/Pi2.

Hotplugging the USB devices will not restore any functionality, the only solution at that appears to be to force a reboot from an ssh shell or yanking power.

Happy to provide any additional logs or investigation needed to track it down.

andrewsil1, 2016-03-20:
Actually, the fiq message I described above only has occurred once, the first time I happened to check the log when the sound stutter and USB failures occurred.  Since then I've reproved the behavior many more times, but without anything of interest showing up in the logs.  Also, emulation station is the only application that seems to instigate this behavior: The emulators it launches have shown no such propensity to drag the USB subsystem down no matter how much audio they play. :-/

popcornmix, 2016-03-20:
Any difference in behaviour between Pi2 and Pi3 would be surprising, so it would be useful if you can double check if the exact same sdcard that has the issues on Pi3 also has issues on Pi2.
Might be worth trying with ``dwc_otg.fiq_fsm_mask=0xf`` added to cmdline.txt (on same line).

andrewsil1, 2016-03-20:
I can try both things and report back.

andrewsil1, 2016-03-20:
Well, I've just done a bunch of testing using fsm_mask=0xf, and the problem goes away.  So at least I have a cure, which is a good thing. :-)  The only major differences I can think of in terms of the overall configuration between the Pi3 and the Pi2 setup I was previously using is that I used a Comfast wifi dongle previously, whereas now I'm using the internal Wifi, and since that freed up a port on the Pi, I have now moved the other devices (keyboard, sound adapter, and 2 HID arcade input devices) directly onto the PI's ports, rather than the self-powered hub they were previously connected to.

I'll still try hooking the Pi2 back up and see if I can get the problem to repro without the 0xf hack with Jessie, just for informational reasons.  I was previously running Wheezy and hadn't updated anything in maybe 3-6 months or so, started noticing this after picking up all the latest bits when updating to the Pi3.

closed 2016-03-20 by andrewsil1

amtssp, 2016-03-23:
hi. Would this be the same as using : dwc_otg.fiq_fsm_mask=0x8 or that a different mask setting?

andrewsil1, 2016-03-23:
That wouldn't be the same. If you only used 0x8, you'd be enabling a specific hack (which appears to have helped in my case) but you'd be disabling all of the normal cases that FIQ is enabled for, pretty much undoing the performance benefits that it was meant to provide in the first place.