raspberrypi/linux #1355: WIFI channels 12 and 13 not being recognized by Raspberry 3

blfuentes, 2016-03-20:
Version 4.1.20 of firmware still doesn't recognize channels 12 and 13 in country zones where those should be available.

popcornmix, 2016-03-20:
This seems to be an upstream kernel issue. I believe if you update to 4.4. kernel with ``sudo BRANCH=next rpi-update`` then channels 12 and 13 are usable.

(but be aware that rpi-update is for testers and there may be regressions).

It is likely we will move to 4.4 kernel as a default soon (next week or so for master branch rpi-update with apt-get to follow) so that is the likely route to getting the higher channels.

grigorig, 2016-03-22:
I can confirm that channels 12 and 13 work with kernel 4.4.

pelwell, 2016-03-22:
Thanks for the confirmation.

popcornmix, 2016-04-11:
Latest rpi-update kernel is now 4.4 so shouldn't have this issue.

JamesH65, 2017-05-18:
Closing this issue as questions answered/resolved.

closed 2017-05-18 by JamesH65