raspberrypi/linux #1358: Issue in mmcsd with raspberry pi 1

sobinonline, 2016-03-21:
Hi

I built the kernel, but there is issue in mmcsd. i am getting mmcsd error on booting with the kernel compiled with this version any fix ?

pelwell, 2016-03-21:
rpi-3.6.y is very old. I suggest you update to at least rpi-4.1.y, or even go straight to rpi-4.4.y. These use a different hardware block with a different driver, and it has solved most of the SD Card problems.

Start by updating the firmware by installing a fresh Raspbian image or by using rpi-update. Then download and build your chosen kernel - build instructions can be found [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md).

If you still see errors after updating then try and capture as much of the error as you can and create a new issue.

sobinonline, 2016-03-22:
Hi

I tried building 4.4 kernel (rpi-4.4.y) . I followed below steps

   - irst, clean the source:                                 #make ARCH=arm
   CROSS_COMPILE=${CCPREFIX} distclean
   - Then set the correct config file for the Board, there are lots of
   config file that one can find inside '/arch/arm/configs/' folder.
                     #make ARCH=arm CROSS_COMPILE=${CCPREFIX} RPI_defconfig
   - Otherwise, place the following config file that i have attached in the
   form of .txt here(
   https://drive.google.com/file/d/0B6Cu_2GN5atpMzUxOXJCSmlzOEk/edit?usp=sharing),
   copy and place it inside '/arch/arm/configs' directory of the Linux Kernel
   Source.
   - After, this start the compilation
   - #make ARCH=arm CROSS_COMPILE=${CCPREFIX}

After giving make i am getting belon
Physical address of main memory (PHYS_OFFSET) [] (NEW) 0x20000000
#
# configuration written to .config
#
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
make[1]: `include/generated/mach-types.h' is up to date.
  CC      kernel/bounds.s
gcc: error: unrecognized argument in option ‘-mabi=aapcs-linux’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized command line option ‘-mlittle-endian’
gcc: error: unrecognized command line option ‘-mno-thumb-interwork’
gcc: error: unrecognized command line option ‘-mfpu=vfp’
make[1]: *** [kernel/bounds.s] Error 1
make: *** [prepare0] Error 2

what could be the error, is it could be due to tool chain

regards
Sobin

pelwell, 2016-03-22:
Those are not the instructions I asked you to follow, and they are wrong ("RPI_defconfig" does not exist - "bcmrpi_defconfig" does). Try following [these instructions](https://www.raspberrypi.org/documentation/linux/kernel/building.md) instead.

Ruffio, 2017-02-05:
@sobinonline is this stilll an issue?

closed 2017-05-18 by JamesH65