raspberrypi/linux #1360: Bluetooth and Dual Shock 3 Playstation Remote will crash System.

coniferous1, 2016-03-21:
Hello All, 

Looking to report a bug with the bluetooth on raspberry pi three with dual shock 3 playstation controllers. 

When connected to one controller intermittently I get these errors (around once an hour, but it's pretty random). It seems like a driver issue, But I'll experiment with anything that will help me fix the issue.

Mar 21 21:45:58 retropie kernel: [ 2372.052392] Bluetooth: hci0: Frame reassembly failed (-84)
Mar 21 21:45:58 retropie kernel: [ 2372.052572] Unable to handle kernel paging request at virtual address ffffffc8
Mar 21 21:45:58 retropie kernel: [ 2372.060430] pgd = 80004000
Mar 21 21:45:58 retropie kernel: [ 2372.063467] [ffffffc8] *pgd=2effa821, *pte=00000000, *ppte=00000000

Message from syslogd@retropie at Mar 21 21:45:58 ...
 kernel:[ 2372.070115] Internal error: Oops: 17 [#1] SMP ARM

uname -r reports: 

4.1.19-v7+

Not a linux Guru, if this is the wrong place or you need more information please let me know.

pelwell, 2016-03-22:
If you can post the whole of the crash dump - down to the "---[ end trace ..." if possible - then it might just help to debug this.

coniferous1, 2016-03-23:
Hey @pelwell , 

Working on getting this now. First time I've created a crash dump so I might be a bit, working on acquiring the prerequisite knowledge.

pelwell, 2016-03-23:
I was thinking that you could just include more of the output in /var/log/syslog. I've never tried to get a crash dump for a kernel crash, and I'm not sure it would work.

coniferous1, 2016-03-23:
That's sort of my thinking as well. It seems that IO simply stops when this occurs. Not sure if I can pipe it out the serial when this happens or not, but I'm looking into that. 

In the meantime I'll get more information from /var/log/syslog and post it here as soon as I get home.

pelwell, 2016-03-23:
Alternatively, if you can provoke it with the text console visible (Ctrl-Alt-F1, Ctrl-Alt-F7 to get back to the desktop) then adding `framebuffer_height=2160` to config.txt may allow you get the whole backtrace onto the screen where you can photograph it.

coniferous1, 2016-03-23:
Generally it happens with emulators open and me actively using the game pad, but I'm going to try that and mash on the buttons furiously while running something with IO in the background and see what happens.

coniferous1, 2016-03-24:
I'm having trouble reproducing this error and being able to capture the trace. I can only seem to get it to happen when emulation station is actively doing something, witch stops me from looking at the text console.

drakerui, 2016-03-30:
Having the same issue here, also unable to get any trace other than the one previously mentioned. 

The system only seems to freeze when displaying something heavier than a TTY on screen, I've had these freezes occur while watching video on kodi, playing some games with emulationstation or working with something on a xfce session but never when looking at a TTY, even if something heavy is running in the background.

I did once get only a partial bluetooth fail while attempting to trigger the freeze:
```
Mar 30 01:53:08 PIPPLWARE dbus[429]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service'
Mar 30 01:53:08 PIPPLWARE systemd[4992]: Failed at step EXEC spawning /usr/lib/bluetooth/bluetoothd: Permission denied
Mar 30 01:53:08 PIPPLWARE systemd[1]: bluetooth.service: main process exited, code=exited, status=203/EXEC
Mar 30 01:53:08 PIPPLWARE systemd[1]: Failed to start Bluetooth service.
Mar 30 01:53:08 PIPPLWARE systemd[1]: Unit bluetooth.service entered failed state.
Mar 30 01:53:33 PIPPLWARE dbus[429]: [system] Failed to activate service 'org.bluez': timed out
```
Another thing of note is that the bluetooth service often fails to start once before actually starting during system startup.

Neutrosider, 2016-03-30:
I'm pretty sure i have the exact same problem, but mine is triggered from a nodejs-program, therefore i'm in the console when it happens, and thus i have logs. I hope they help:

```Bash
[ 1299.902456] Bluetooth: hci0 unexpected SMP command 0x02 from c8:0f:10:28:ab:51
[ 1300.037106] Bluetooth: hci0 unexpected SMP command 0x03 from c8:0f:10:28:ab:51
[ 1300.292100] Bluetooth: hci0 unexpected SMP command 0x04 from c8:0f:10:28:ab:51
[ 1300.697121] Bluetooth: hci0 unexpected SMP command 0x06 from c8:0f:10:28:ab:51
[ 1300.925752] Bluetooth: hci0: Frame reassembly faild (-84)
[ 1300.931428] Bluetooth: hci0 unexpected SMP command 0x07 from c8:0f:10:28:ab:51
[ 1301.111148] Unable to handle kernel paging request at virtual address ffffffc8
[ 1301.118563] pgd = 80004000
[ 1301.121372] [ffffffc8] *pgd=3dffa821, *pte=00000000, *ppte=00000000
[ 1301.127833] Internal error: Oops: 17 [#1] SMP ARM

Entering kdb (current=0xbbcb6d40, pid 367) on processor 0 Oops: (null)
due to oops @ 0x07f3c8c20
CPU: 8 PID: 367 Comm: kwoerker/0:3 Not tainted 4.1.20-v7+ #867
Hardware name: BCM2709
Workqueue: events flush_to_ldisc
task: bbcb6d40 ti: b9db4000 task.ti: b9db4000
PC is at h4_recv_buf+0x34/0x22c [hci_uart]
LR is at h4_recv+0x50/0x98 [hci_uart]
pc : [<7f3c8c20>]    lr : [<7f3c8e68>]    psr: a0000013
sp : b9db5e38  ip : b9db5e68  fp : b9db5e64
r10: 00000008  r9 : 00000008  r8 : bb22d000
r7 : 7f3ca194  r6 : bb067c6c  r5 : 00000003  r4 : ffffffac
r3 : 00000008  r2 : bb067c6c  r1 : ffffffac  r0 : bb22d000
Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Control: 10c5383d  Table: 3bf4406a  DAC: 00000015
CPU: 0 PID: 367 Comm: kworker/0:3 Not tainted 4.1.20-v7+ #867
Hardware name: BCM2709
Workqueue: events flush_to_ldisc
[<800185e0>] (unwind_backtrace) from [<80013f48>] (show_stack+0x20/0x24)
[<80013f48>] (show_stack) from [<80572ddc>] (dump_stack+0xd4/0x118)
[<80572ddc>] (dump_stack) from [<80010d18>] (show_regs+0x1c/0x20)
[<80010d18>] (show_regs) from [<800b4a70>] (kdb_main_loop+0x3ec/0x800)
```

There it just stops. but i have a longer log from a previous crash that seems to have the same cause:

```Bash
[  156.026986] Bluetooth: hci0: Frame reassembly faild (-84)
[  156.060272] Bluetooth: hci0 unexpected SMP command 0x07 from c8:0f:10:28:af:c1
Entering kdb (current=0xbc77cac0, pid 96) on processor 0 Oops: (null)
due to oops @ 0x07f3c8c20
CPU: 8 PID: 367 Comm: kwoerker/0:2 Not tainted 4.1.20-v7+ #867
Hardware name: BCM2709
Workqueue: events flush_to_ldisc
task: bc77cac0 ti: bbf7c000 task.ti: bbf7c000
PC is at h4_recv_buf+0x34/0x22c [hci_uart]
LR is at h4_recv+0x50/0x98 [hci_uart]
pc : [<7f3c2c20>]    lr : [<7f3c2e68>]    psr: a0000013
sp : bbf7de38  ip : bbf7de68  fp : bbf7de64
r10: 0000000a  r9 : 0000000a  r8 : b9de0000
r7 : 7f3c4191  r6 : bb1eb8db  r5 : 00000003  r4 : ffffffac
r3 : 0000000a  r2 : bb1eb8db  r1 : ffffffac  r0 : b9de0000
Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Control: 10c5383d  Table: 39c3806a  DAC: 00000015
CPU: 0 PID: 96 Comm: kworker/0:2 Not tainted 4.1.20-v7+ #867
Hardware name: BCM2709
Workqueue: events flush_to_ldisc
[<800185e0>] (unwind_backtrace) from [<80013f48>] (show_stack+0x20/0x24)
[<80013f48>] (show_stack) from [<80572ddc>] (dump_stack+0xd4/0x118)
[<80572ddc>] (dump_stack) from [<80010d18>] (show_regs+0x1c/0x20)
[<80010d18>] (show_regs) from [<800b4a70>] (kdb_main_loop+0x3ec/0x800)
more> [<800b4a70>] (kdb_main_loop) from [<800b796c>] (kdb_stub+0x2c4/0x510)
[<800b796c>] (kdb_stub) from [<800ad360>] (kgdb_cpu_enter+0x480/0x838)
[<800ad360>] (kgdb_cpu_enter) from [<800ad974>] (kgdb_handle_exception+0x104/0x218)
[<800ad974>] (kgdb_handle_exception) from [<80017aec>] (kgdb_notify+0x40/0x90)
[<80017aec>] (kgdb_notify) from [<80044cd4>] (notifier_call_chain+0x54/0x94)
[<80044cd4>] (notifier_call_chain) from [<80044eec>] (notify_die+0x48/0x54)
[<80044eec>] (notify_die) from [<80014064>] (die+0x118/0x43c)
[<80014064>] (die) from [<80571c48>] (__do_kernel_fault.part.0+0x74/0x8
[<80571c48>] (__do_kernel_fault.part.0) from [<80578bd4>] (do_page_fault+0x28c/0x398)
[<80578bd4>] (do_page_fault) from [<800091e8>] (do_DataAbort+0x48/0xc4)
[<800091e8>] (do_DataAbort) from [<8057811c>] (__dabt_svc+0x3c/0x60)
Exception stack(0xbbf7ddf0 to 0xbbf7de38)
dde0:                                     b9de0000 ffffffac bb1eb8db 0000000a
de00: ffffffac 00000003 bb1eb8db 7f3c4194 b9de0000 0000000a 0000000a bbf7de64
de20: bbf7de68 bbf7de38 7f3c2e68 7f3c2c20 a0000013 ffffffff
[<8057811c>] (__dabt_svc) from [<7f3c2c20>] (h4_recv_buf+0x34/0x22c [hci_uart])
[<7f3c2c20>] (h4_recv_buf [hci_uart]) from [<7f3c2e68>] (h4_recv+0x50/0x98 [hci_uart])
[<7f3c2e68>] (h4_recv [hci_uart]) from [<7f3c20f0>] (hci_uart_tty_receive+0x74/0xb0 [hci_uart])
[<7f3c20f0>] (hci_uart_tty_receive [hci_uart]) from [<80358b48>] (flush_to_ldisc+0x190/0x1a0)
more> [<80358b48>] (flush_to_ldisc) from [<8003e0b8>] (process_one_work+0x154/0x458)
[<8003e0b8>] (process_one_work) from [<8003e410>] (worker_thread+0x54/0x500)
[<8003e410>] (worker_thread) from [<80043bec>] (kthread+0xec/0x104)
[<80043bec>] (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)

[0]kdb> _
```

I was careful, but the logs **may contain errors, because i had to transcribe them by hand**

delize, 2016-04-03:
I am having a similar problem with bluetooth and a Nexus6P device connecting. After pairing the Nexus 6P with blueman and other bluetooth tools, the raspberry pi immediately freezes. Leaving it for ten minutes results in no response, however, I have no logs.

steverandy, 2016-04-06:
I'm having a similar issue with a Logitech bluetooth mouse. If there is an intensive process going on, moving the mouse would often freeze the entire OS. 

I'm new to Linux world. What are the steps I can do to provide debug info here?

Is there an instability issue with Bluetooth hardware on rpi3? Or is this a bug in software?

pelwell, 2016-04-06:
Bluetooth throughput is limited on Pi3 because of a lack of flow control signals to BT modem. The Bluetooth H5 protocol is supposed to handle packet loss, but the current BT modem firmware does not work with the BlueZ stack in H5 mode.

steverandy, 2016-04-06:
Is this issue perhaps related to this? https://github.com/raspberrypi/firmware/issues/553

paulrouget, 2016-04-12:
How can we make this issue actionable?

JamesH65, 2016-04-12:
It's open and on this bug list, so will be looked at when time allows. Depends on the impact of the issues and the number of  people it affects.

paulrouget, 2016-04-12:
> Depends on the impact of the issues

For the record, all my bluetooth devices trigger this crash.

Also, as I was googling, I've seen several people reporting "system freeze" with the PS3 bluetooth controller (example http://blog.petrockblock.com/forums/topic/problems-getting-ps3-controller-to-work-properly/#post-119899 and https://www.reddit.com/r/RetroPie/comments/4aoojj/pi3_crashing_with_kernel_474079425_internal_error/)

paulrouget, 2016-04-12:
If there's anything I can do to get a better log, or try a patch, I'd be happy to help.

BaDvs3viL, 2016-04-12:
I just wanted to add that I am having the same issue.  I have a Pi3 with a fresh RP3.6 and just started using PS3 DS3 controllers through BT yesterday and almost instantly started seeing the freezes.  If I change out the PS3 controller and use a wired 360, I have no issues at all.

paulrouget, 2016-04-12:
(no crash with a bluetooth dongle)

pelwell, 2016-04-12:
We'll get in a DS3 and have a go. Can you say what extra configuration is required beyond pairing and trusing, and perhaps recommend a test application if necessary?

I can see three potential outcomes:
1) If this crash turns out to be due to a bug/incompatibility in the Linux Bluetooth stack then we have a chance of fixing it.
2) If it is a bug/incompatibility in the Broadcom firmware then it becomes much harder to get a fix.
3) If it is due to data loss going through the UART then it may be improved by H5 support, which is somewhere between 1 and 2 in tractability.

BaDvs3viL, 2016-04-12:
All I did was install the required files for the PS3 controllers through the RP-setup and sync.

Prindle19, 2016-04-12:
Also having this issue with onboard Bluetooth and clone dualshock 3 controller (Shanwan) connected via sixad.

It happen sporadically and under different emulators / roms. When it hangs, system can't be interrupted via keyboard, SSH sessions break and pi stips responding to pings

paulrouget, 2016-04-12:
> We'll get in a DS3 and have a go. Can you say what extra configuration is required beyond pairing and trusing, and perhaps recommend a test application if necessary?

Linux retropie 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux
No special configuration. I run PSP games. After a minute or so of playing, freeze.

BaDvs3viL, 2016-04-12:
@paulrouget How did you disable the on board BT so you could use you own?

dtoverlay=pi3-disable-bt?

paulrouget, 2016-04-12:
> @paulrouget How did you disable the on board BT so you could use you own?

I was reporting what I've found in [other forums](https://www.reddit.com/r/RetroPie/comments/4aoojj/pi3_crashing_with_kernel_474079425_internal_error/d13fgae):

> I plugged in a CSR 4.0 bluetooth USB button and am using bluetooth without a hitch - so it seems the issue is with the onboard bluetooth.

BaDvs3viL, 2016-04-13:
I can confirm that I put dtoverlay=pi3-disable-bt in my config.txt and used a USB BT dongle, I played for hours last night without any issues.  I soon as I reverted back and tried, I got a freeze with 5 mins.

KrS14, 2016-04-13:
I'm also seeing this issue with the onboard BT:
[Apr13 16:55] Bluetooth: hci0: Frame reassembly failed (-84)
[  +0.000152] Unable to handle kernel paging request at virtual address ffffffc8
[  +0.007510] pgd = 80004000
[  +0.002857] [ffffffc8] *pgd=2effa821, *pte=00000000, *ppte=00000000
[  +0.006553] Internal error: Oops: 17 [#1] SMP ARM

Hopefully the PS3 driver gets fixed sooner than later, as I don't have a BT dongle to use :/

joestrong, 2016-04-14:
I can confirm that using a bluetooth dongle solves it for me. On-board bluetooth means a freeze within a few minutes, dongle has allowed me to play for hours with no problems. I didn't have to change any text file to disable the pi3 bluetooth, just plugging in a dongle alone seemed to work.

oblique, 2016-04-15:
I don't know if more crash dumps will help, but here they are:

```
[  380.219390] Unable to handle kernel paging request at virtual address ffffffc8
[  380.228822] pgd = 80004000
[  380.233051] [ffffffc8] *pgd=2effa821, *pte=00000000, *ppte=00000000
[  380.241212] Internal error: Oops: 17 [#1] SMP ARM
[  380.247696] Modules linked in: hidp bnep hci_uart btbcm bluetooth brcmfmac snd_usb_audio brcmutil cfg80211 snd_usbmidi_lib snd_hwdep snd_rawmidi snd_seq_device rfkill evdev bcm2835_gpiomem bcm2835_rng uio_pdrv_genirq uio snd_bcm2835 snd_pcm snd_timer snd joydev uinput fuse ipv6
[  380.277738] CPU: 0 PID: 4 Comm: kworker/0:0 Not tainted 4.1.19-v7+ #858
[  380.287198] Hardware name: BCM2709
[  380.293683] Workqueue: events flush_to_ldisc
[  380.300995] task: ad901140 ti: ad91c000 task.ti: ad91c000
[  380.309521] PC is at h4_recv_buf+0x34/0x22c [hci_uart]
[  380.317860] LR is at h4_recv+0x50/0x98 [hci_uart]
[  380.325840] pc : [<7f37fc20>]    lr : [<7f37fe68>]    psr: a0000013
[  380.325840] sp : ad91de38  ip : ad91de68  fp : ad91de64
[  380.344050] r10: 00000011  r9 : 00000011  r8 : ad315000
[  380.352813] r7 : 7f381194  r6 : ad0fbca5  r5 : 00000003  r4 : ffffffac
[  380.363117] r3 : 00000011  r2 : ad0fbca5  r1 : ffffffac  r0 : ad315000
[  380.373280] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[  380.384694] Control: 10c5383d  Table: 2aeb006a  DAC: 00000015
[  380.394601] Process kworker/0:0 (pid: 4, stack limit = 0xad91c210)
[  380.404993] Stack: (0xad91de38 to 0xad91e000)
[  380.413849] de20:                                                       00000011 00000011
[  380.426617] de40: a9c55400 aae0bac0 aae0bd80 ad0a9004 ad0a9000 00000011 ad91de8c ad91de68
[  380.439493] de60: 7f37fe68 7f37fbf8 7f381194 00000003 a9c55400 ad381000 aae0bac0 ad0fbc00
[  380.452647] de80: ad91deac ad91de90 7f37f0f0 7f37fe24 00000011 00000011 ad381000 ad0a9014
[  380.466025] dea0: ad91dee4 ad91deb0 80358e48 7f37f088 ad91df0c ad91dec0 80573e14 ad896380
[  380.479569] dec0: ad0a9004 aefb4080 aefb7f00 00000000 ad896380 00000000 ad91df24 ad91dee8
[  380.492998] dee0: 8003e0b8 80358cc4 aefb4080 aefb4094 ad91c000 00000008 ad896380 aefb4080
[  380.506584] df00: ad896398 aefb4094 ad91c000 00000008 ad896380 aefb4080 ad91df5c ad91df28
[  380.520465] df20: 8003e410 8003df70 aefb4214 80822900 00000000 00000000 ad894c00 ad896380
[  380.534561] df40: 8003e3bc 00000000 00000000 00000000 ad91dfac ad91df60 80043bec 8003e3c8
[  380.548828] df60: ad91df94 00000000 8004a880 ad896380 00000000 00000000 ad91df78 ad91df78
[  380.562945] df80: 00000000 00000000 ad91df88 ad91df88 ad894c00 80043b00 00000000 00000000
[  380.576929] dfa0: 00000000 ad91dfb0 8000faf8 80043b0c 00000000 00000000 00000000 00000000
[  380.590854] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  380.605278] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 55555555 55555555
[  380.619668] [<7f37fc20>] (h4_recv_buf [hci_uart]) from [<7f37fe68>] (h4_recv+0x50/0x98 [hci_uart])
[  380.634601] [<7f37fe68>] (h4_recv [hci_uart]) from [<7f37f0f0>] (hci_uart_tty_receive+0x74/0xb0 [hci_uart])
[  380.650364] [<7f37f0f0>] (hci_uart_tty_receive [hci_uart]) from [<80358e48>] (flush_to_ldisc+0x190/0x1a0)
[  380.666024] [<80358e48>] (flush_to_ldisc) from [<8003e0b8>] (process_one_work+0x154/0x458)
[  380.680568] [<8003e0b8>] (process_one_work) from [<8003e410>] (worker_thread+0x54/0x500)
[  380.695006] [<8003e410>] (worker_thread) from [<80043bec>] (kthread+0xec/0x104)
[  380.708777] [<80043bec>] (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)
[  380.722754] Code: e59b7004 e59b5008 0a00002d e3540000 (11d431bc) 
[  380.735827] ---[ end trace 831fb3fed9cab54d ]---
[  380.747637] Unable to handle kernel paging request at virtual address ffffffec
[  380.747640] pgd = 80004000
[  380.747652] [ffffffec] *pgd=2effa821, *pte=00000000, *ppte=00000000
[  380.747660] Internal error: Oops: 17 [#2] SMP ARM
[  380.747704] Modules linked in: hidp bnep hci_uart btbcm bluetooth brcmfmac snd_usb_audio brcmutil cfg80211 snd_usbmidi_lib snd_hwdep snd_rawmidi snd_seq_device rfkill evdev bcm2835_gpiomem bcm2835_rng uio_pdrv_genirq uio snd_bcm2835 snd_pcm snd_timer snd joydev uinput fuse ipv6
[  380.747714] CPU: 0 PID: 4 Comm: kworker/0:0 Tainted: G      D         4.1.19-v7+ #858
[  380.747716] Hardware name: BCM2709
[  380.747729] task: ad901140 ti: ad91c000 task.ti: ad91c000
[  380.747746] PC is at kthread_data+0x18/0x20
[  380.747758] LR is at wq_worker_sleeping+0x1c/0xe0
[  380.747764] pc : [<8004432c>]    lr : [<8003f284>]    psr: 00000193
[  380.747764] sp : ad91db78  ip : ad91db88  fp : ad91db84
[  380.747767] r10: 2e7b1000  r9 : ad90142c  r8 : 808240a4
[  380.747770] r7 : 80803580  r6 : 80803580  r5 : ad901140  r4 : 00000000
[  380.747773] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : ad901140
[  380.747781] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  380.747784] Control: 10c5383d  Table: 2aeb006a  DAC: 00000015
[  380.747787] Process kworker/0:0 (pid: 4, stack limit = 0xad91c210)
[  380.747791] Stack: (0xad91db78 to 0xad91e000)
[  380.747796] db60:                                                       ad91db9c ad91db88
[  380.747801] db80: 8003f284 80044320 aefb4580 ad901140 ad91dbec ad91dba0 80574048 8003f274
[  380.747808] dba0: 800781ac 80077e7c ad91dc04 ad91dbb8 80028030 80078194 00000000 807ff240
[  380.747813] dbc0: ad910964 ad91c000 ad91d98c ad91dc10 ad9013a8 00000001 00000001 ad900000
[... cut ...]
[  380.748187] [<80358e48>] (flush_to_ldisc) from [<8003e0b8>] (process_one_work+0x154/0x458)
[  380.748196] [<8003e0b8>] (process_one_work) from [<8003e410>] (worker_thread+0x54/0x500)
[  380.748203] [<8003e410>] (worker_thread) from [<80043bec>] (kthread+0xec/0x104)
[  380.748213] [<80043bec>] (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)
[  380.748219] Code: e24cb004 e52de004 e8bd4000 e59032c0 (e5130014) 
[  380.748225] ---[ end trace 831fb3fed9cab54e ]---
[  380.748227] Fixing recursive fault but reboot is needed!
```

lkernan, 2016-04-18:
Hopefully related, i've been able to trigger what appears to the be the same thing by playing music from an iPhone to the Pi 3 onboard BT via Bluetooth A2DP.   

Sitting idle at an empty desktop it will play several songs no problem. 
While still playing, within seconds of starting another task on the pi, it becomes unresponsive to input.

filius-martis, 2016-04-20:
I just want to add that I see the same system freezes with a Sony PS4 Wireless DualShock 4 Controller.

On the other hand, connecting my simple Anker Bluetooth Keyboard A7726 doesn`t seem to trigger the bug.

BaDvs3viL, 2016-04-20:
Not that I think it would matter but I upgraded to Retropie 3.7 and still had the same issue.

coniferous1, 2016-04-20:
@BaDvs3viL this will probably effect all distros until the kernel or firmware is upgraded.

mauruco, 2016-04-22:
Has anyone a tutorial to make a bluetooth dongle works on PI3? 1. Disable the internal bluetooth etc ..
Thanks

mauruco, 2016-04-22:
Done. But the dongle does not work yet and it works perfect in pi2.
Thanks

Prindle19, 2016-04-22:
Probably getting a bit off topic here, but likely need to install pi-bluetooth:
 
https://www.reddit.com/r/RetroPie/comments/4d5hcf/how_to_setup_8bitdo_nes30_fc30_pro_controllers_on/

mauruco, 2016-04-22:
Funciona! Eu adicionei no rc.local "sixad --start &" e funcionou!

mauruco, 2016-04-22:
Do you think the setting're ok?
Thank you for your help!

diogomsantos, 2016-04-22:
@mauruco This is a bug report section, not an "how to" do something. If you don't know how to setup a BT dongle use the apropriate forums. You are spamming everyone interested in getting this bug fixed.

cjbnorg, 2016-04-23:
Can confirm my PI3 B locks up while using onboard BT under RetroPie 3.7 ( raspbian ) with BOTH authentic PS3 DS controllers as well as the Garsia PS3 DS clone controllers.  Disabled onboard BT and all runs fine with a BT USB dongle ( Inateck BTA-BC4B6 ).  Watching this issue with great interest.  Whole reason I got a PI3 was for onboard BT for my controllers.  GLA

vinpac1981, 2016-04-23:
I can officially report that the freeze only happens when playing N64 or PSX games. I sat there and played all other systems with no issue.

cjbnorg, 2016-04-23:
Only play with SNES & MAME here and locks up within 1 to 5 minutes every time.

KrS14, 2016-04-23:
I have the same issue with NES

Raize, 2016-04-23:
I am having the same issue with SNES and 8bitdo wireless controllers on a Pi3. Locks up within 5 to 8 minutes typically. I would be willing to pay a .5 BTC bug bounty on this one since it is affecting my niece and I's ability to game together on weekends. Thanks.

FreakyBigFoot, 2016-04-23:
Same here. My 8bitdo controller is also causing my pi 3 to freeze. It's not just ps3 controllers. 

Side note; I'm glad people are linking others to my how-to for the 8Bitdo controllers on reddit!

erikpena, 2016-04-24:
I'm seeing the same thing with my RPI3 and PS3 controller.

secured, 2016-04-24:
Another crash dump with RPI3, Retropie und PS3 connected via bluetooth.

`[23131.768726] Unable to handle kernel paging request at virtual address ffffffc8
[23131.776103] pgd = 80004000
[23131.778886] [ffffffc8] *pgd=2effa821, *pte=00000000, *ppte=00000000
[23131.785363] Internal error: Oops: 17 [#1] SMP ARM
[23131.790128] Modules linked in: evdev hidp bnep hci_uart btbcm bluetooth brcmfmac brcmutil cfg80211 rfkill bcm2835_gpiomem bcm2835_rng uio_pdrv_genirq uio snd_bcm2835 snd_pcm snd_timer snd joydev uinput fuse ipv6
[23131.809224] CPU: 0 PID: 6716 Comm: kworker/0:0 Not tainted 4.1.19-v7+ #858
[23131.816189] Hardware name: BCM2709
[23131.819637] Workqueue: events flush_to_ldisc
[23131.823962] task: aa4cb3c0 ti: aa7f8000 task.ti: aa7f8000
[23131.829436] PC is at h4_recv_buf+0x34/0x22c [hci_uart]
[23131.834641] LR is at h4_recv+0x50/0x98 [hci_uart]
[23131.839405] pc : [<7f32bc20>]    lr : [<7f32be68>]    psr: a0000013
[23131.839405] sp : aa7f9e38  ip : aa7f9e68  fp : aa7f9e64
[23131.851041] r10: 00000008  r9 : 00000008  r8 : ad103000
[23131.856332] r7 : 7f32d194  r6 : aa420834  r5 : 00000003  r4 : ffffffac
[23131.862946] r3 : 00000008  r2 : aa420834  r1 : ffffffac  r0 : ad103000
[23131.869560] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[23131.876966] Control: 10c5383d  Table: 2a7f406a  DAC: 00000015
[23131.882785] Process kworker/0:0 (pid: 6716, stack limit = 0xaa7f8210)
[23131.889312] Stack: (0xaa7f9e38 to 0xaa7fa000)
[23131.893722] 9e20:                                                       00000008 00000008
[23131.902012] 9e40: ac594580 ac7c4200 ac7c4840 ada8a004 ada8a000 00000008 aa7f9e8c aa7f9e68
[23131.910302] 9e60: 7f32be68 7f32bbf8 7f32d194 00000003 ac594580 ad3d4000 ac7c4200 aa420800
[23131.918591] 9e80: aa7f9eac aa7f9e90 7f32b0f0 7f32be24 00000008 00000008 ad3d4000 ada8a014
[23131.926880] 9ea0: aa7f9ee4 aa7f9eb0 80358e48 7f32b088 aa7f9f0c aa7f9ec0 80573e14 aa599680
[23131.935169] 9ec0: ada8a004 aefb4080 aefb7f00 00000000 aa599680 00000000 aa7f9f24 aa7f9ee8
[23131.943459] 9ee0: 8003e0b8 80358cc4 aefb4080 aefb4094 aa7f8000 00000008 aa599680 aefb4080
[23131.951749] 9f00: aa599698 aefb4094 aa7f8000 00000008 aa599680 aefb4080 aa7f9f5c aa7f9f28
[23131.960038] 9f20: 8003e410 8003df70 8003e3bc 80822900 00000000 00000000 ac63c5c0 aa599680
[23131.968328] 9f40: 8003e3bc 00000000 00000000 00000000 aa7f9fac aa7f9f60 80043bec 8003e3c8
[23131.976617] 9f60: aa7f9f94 00000000 8004a880 aa599680 00000000 00000000 aa7f9f78 aa7f9f78
[23131.984906] 9f80: 00000000 00000000 aa7f9f88 aa7f9f88 ac63c5c0 80043b00 00000000 00000000
[23131.993196] 9fa0: 00000000 aa7f9fb0 8000faf8 80043b0c 00000000 00000000 00000000 00000000
[23132.001485] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[23132.009774] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[23132.018083] [<7f32bc20>] (h4_recv_buf [hci_uart]) from [<7f32be68>] (h4_recv+0x50/0x98 [hci_uart])
[23132.027170] [<7f32be68>] (h4_recv [hci_uart]) from [<7f32b0f0>] (hci_uart_tty_receive+0x74/0xb0 [hci_uart])
[23132.037048] [<7f32b0f0>] (hci_uart_tty_receive [hci_uart]) from [<80358e48>] (flush_to_ldisc+0x190/0x1a0)
[23132.046752] [<80358e48>] (flush_to_ldisc) from [<8003e0b8>] (process_one_work+0x154/0x458)
[23132.055133] [<8003e0b8>] (process_one_work) from [<8003e410>] (worker_thread+0x54/0x500)
[23132.063338] [<8003e410>] (worker_thread) from [<80043bec>] (kthread+0xec/0x104)
[23132.070750] [<80043bec>] (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)
[23132.078072] Code: e59b7004 e59b5008 0a00002d e3540000 (11d431bc) 
[23132.084329] ---[ end trace 501092e1c3e7ed97 ]---
[23132.089321] Unable to handle kernel paging request at virtual address ffffffec
[23132.096655] pgd = 80004000
[23132.096666] [ffffffec] *pgd=2effa821, *pte=00000000, *ppte=00000000
[23132.096673] Internal error: Oops: 17 [#2] SMP ARM
[23132.096711] Modules linked in: evdev hidp bnep hci_uart btbcm bluetooth brcmfmac brcmutil cfg80211 rfkill bcm2835_gpiomem bcm2835_rng uio_pdrv_genirq uio snd_bcm2835 snd_pcm snd_timer snd joydev uinput fuse ipv6
[23132.096720] CPU: 0 PID: 6716 Comm: kworker/0:0 Tainted: G      D         4.1.19-v7+ #858
[23132.096722] Hardware name: BCM2709
[23132.096732] task: aa4cb3c0 ti: aa7f8000 task.ti: aa7f8000
[23132.096741] PC is at kthread_data+0x18/0x20
[23132.096750] LR is at wq_worker_sleeping+0x1c/0xe0
[23132.096754] pc : [<8004432c>]    lr : [<8003f284>]    psr: 00000193
[23132.096754] sp : aa7f9b78  ip : aa7f9b88  fp : aa7f9b84
[23132.096757] r10: 2e7b1000  r9 : aa4cb6ac  r8 : 808240a4
[23132.096759] r7 : 80803580  r6 : 80803580  r5 : aa4cb3c0  r4 : 00000000
[23132.096762] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : aa4cb3c0
[23132.096770] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[23132.096772] Control: 10c5383d  Table: 2a7f406a  DAC: 00000015
[23132.096775] Process kworker/0:0 (pid: 6716, stack limit = 0xaa7f8210)
[23132.096778] Stack: (0xaa7f9b78 to 0xaa7fa000)
[23132.096782] 9b60:                                                       aa7f9b9c aa7f9b88
[23132.096787] 9b80: 8003f284 80044320 aefb4580 aa4cb3c0 aa7f9bec aa7f9ba0 80574048 8003f274
[23132.096793] 9ba0: 800781ac 80077e7c aa7f9c04 aa7f9bb8 80028030 80078194 00000000 807ff240
[23132.096799] 9bc0: aa77f824 aa7f8000 aa7f998c aa7f9c10 aa4cb628 00000001 00000001 ad900000
[23132.096804] 9be0: aa7f9c04 aa7f9bf0 80574324 80573c30 0420806c aa4cb3c0 aa7f9c44 aa7f9c08
[... 31 lines cut ...]
[... cut ...]
[23132.097152] [<80358e48>] (flush_to_ldisc) from [<8003e0b8>] (process_one_work+0x154/0x458)
[23132.097161] [<8003e0b8>] (process_one_work) from [<8003e410>] (worker_thread+0x54/0x500)
[23132.097168] [<8003e410>] (worker_thread) from [<80043bec>] (kthread+0xec/0x104)
[23132.097175] [<80043bec>] (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)
[23132.097181] Code: e24cb004 e52de004 e8bd4000 e59032c0 (e5130014) 
[23132.097187] ---[ end trace 501092e1c3e7ed98 ]---
[23132.097190] Fixing recursive fault but reboot is needed!`

mauruco, 2016-04-27:
@diogomsantos  It is good to know that a bluetooth dongle solves the problem and also how to do it.
I am also interested in the problem and mentioned it in several places.
Relax and try to contribute to the cause. Complaining does not help.

diogomsantos, 2016-04-27:
@mauruco This is not a forum. 
The fact that using a BT dongle solves this problem was already well known, and spread all over the internet. It was also already mentioned here earlier. 
But this not a how to workaround the issue, this is for fixing the issue in the RPi 3 internal BT chip. 
So do you realize that your posts don't help nobody, and you did nothing to fix the issue ? 
It is just spam. Keep this on topic.

mauruco, 2016-04-27:
@diogomsantos I do not know what your problem boy. Just try to be constructive. People also look for workarrounds and found this place! Everyone has understood your message. OK? Have a nice day.

FreakyBigFoot, 2016-04-30:
Here is everything that happened at 5:04. Not sure what's going on but it only happens when I'm using bluetooth. 

`Apr 30 05:04:09 retropie rsyslogd: [origin software="rsyslogd" swVersion="8.4.2" x-pid="433" x-info="http://www.rsyslog.com"] exiting on signal 15.
Apr 30 05:04:13 retropie rsyslogd: [origin software="rsyslogd" swVersion="8.4.2" x-pid="438" x-info="http://www.rsyslog.com"] start
Apr 30 05:04:13 retropie kernel: [    0.000000] Booting Linux on physical CPU 0x0
Apr 30 05:04:13 retropie kernel: [    0.000000] Initializing cgroup subsys cpuset
Apr 30 05:04:13 retropie kernel: [    0.000000] Initializing cgroup subsys cpu
Apr 30 05:04:13 retropie kernel: [    0.000000] Initializing cgroup subsys cpuacct
Apr 30 05:04:13 retropie kernel: [    0.000000] Linux version 4.1.19-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #858 SMP Tue Mar 15 15:56:00 GMT 2016
Apr 30 05:04:13 retropie kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Apr 30 05:04:13 retropie kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Apr 30 05:04:13 retropie kernel: [    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
Apr 30 05:04:13 retropie kernel: [    0.000000] cma: Reserved 8 MiB at 0x2e400000
Apr 30 05:04:13 retropie kernel: [    0.000000] Memory policy: Data cache writealloc
Apr 30 05:04:13 retropie kernel: [    0.000000] [bcm2709_smp_init_cpus] enter (9420->f3003010)
Apr 30 05:04:13 retropie kernel: [    0.000000] [bcm2709_smp_init_cpus] ncores=4
Apr 30 05:04:13 retropie kernel: [    0.000000] PERCPU: Embedded 12 pages/cpu @aefb0000 s20416 r8192 d20544 u49152
Apr 30 05:04:13 retropie kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 190820
Apr 30 05:04:13 retropie kernel: [    0.000000] Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2709.boardrev=0xa22082 bcm2709.serial=0x64aa1861 smsc95xx.macaddr=B8:27:EB:AA:18:61 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet loglevel=3 consoleblank=0
Apr 30 05:04:13 retropie kernel: [    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
Apr 30 05:04:13 retropie kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Apr 30 05:04:13 retropie kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Apr 30 05:04:13 retropie kernel: [    0.000000] Memory: 744496K/770048K available (6064K kernel code, 534K rwdata, 1664K rodata, 444K init, 757K bss, 17360K reserved, 8192K cma-reserved)
Apr 30 05:04:13 retropie kernel: [    0.000000] Virtual kernel memory layout:
Apr 30 05:04:13 retropie kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Apr 30 05:04:13 retropie kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Apr 30 05:04:13 retropie kernel: [    0.000000]     vmalloc : 0xaf800000 - 0xff000000   (1272 MB)
Apr 30 05:04:13 retropie kernel: [    0.000000]     lowmem  : 0x80000000 - 0xaf000000   ( 752 MB)
Apr 30 05:04:13 retropie kernel: [    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
Apr 30 05:04:13 retropie kernel: [    0.000000]       .text : 0x80008000 - 0x807945f0   (7730 kB)
Apr 30 05:04:13 retropie kernel: [    0.000000]       .init : 0x80795000 - 0x80804000   ( 444 kB)
Apr 30 05:04:13 retropie kernel: [    0.000000]       .data : 0x80804000 - 0x80889b10   ( 535 kB)
Apr 30 05:04:13 retropie kernel: [    0.000000]        .bss : 0x8088c000 - 0x809497dc   ( 758 kB)
Apr 30 05:04:13 retropie kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Apr 30 05:04:13 retropie kernel: [    0.000000] Hierarchical RCU implementation.
Apr 30 05:04:13 retropie kernel: [    0.000000] 	Additional per-CPU info printed with stalls.
Apr 30 05:04:13 retropie kernel: [    0.000000] NR_IRQS:608
Apr 30 05:04:13 retropie kernel: [    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
Apr 30 05:04:13 retropie kernel: [    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Apr 30 05:04:13 retropie kernel: [    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Apr 30 05:04:13 retropie kernel: [    0.000026] Switching to timer-based delay loop, resolution 52ns
Apr 30 05:04:13 retropie kernel: [    0.000233] Console: colour dummy device 80x30
Apr 30 05:04:13 retropie kernel: [    0.000431] console [tty1] enabled
Apr 30 05:04:13 retropie kernel: [    0.000465] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Apr 30 05:04:13 retropie kernel: [    0.000485] pid_max: default: 32768 minimum: 301
Apr 30 05:04:13 retropie kernel: [    0.000788] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
Apr 30 05:04:13 retropie kernel: [    0.000803] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
Apr 30 05:04:13 retropie kernel: [    0.001767] Initializing cgroup subsys blkio
Apr 30 05:04:13 retropie kernel: [    0.001793] Initializing cgroup subsys memory
Apr 30 05:04:13 retropie kernel: [    0.001818] Initializing cgroup subsys devices
[... cut ...]
Apr 30 05:04:18 retropie kernel: [    9.937629] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
Apr 30 05:04:18 retropie kernel: [    9.937664] Bluetooth: HIDP socket layer initialized
Apr 30 05:04:22 retropie kernel: [   14.214266] input: 8Bitdo NES30 Pro as /devices/platform/soc/3f201000.uart/tty/ttyAMA0/hci0/hci0:12/0005:3820:0009.0001/input/input0
Apr 30 05:04:22 retropie kernel: [   14.215087] hid-generic 0005:3820:0009.0001: input,hidraw0: BLUETOOTH HID v1.00 Keyboard [8Bitdo NES30 Pro] on b8:27:eb:00:b2:cb
`

FreakyBigFoot, 2016-04-30:
Where are the logs kept that show the error(s)?

konradsa, 2016-05-02:
Has anybody tried a 4.4 kernel yet to see if that fixes the issue?

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=144087

FreakyBigFoot, 2016-05-02:
I don't even see a mention of bluetooth there ^

konradsa, 2016-05-02:
Bluetooth doesn't need to be mentioned there, these are the instructions on how to upgrade to a 4.4 kernel. There is a high likelihood a lot of patches and improvements have been done for the Bluetooth drivers in the new kernel version.

Prindle19, 2016-05-02:
What is in there is a note about Device Tree: 

"Notes:
Device tree is required for 4.4 kernel. Settings like "dtoverlay=" to disable device tree will cause failure to boot."

So, safe to assume that current configs that disable onboard bt would cause issues? 

"dtoverlay=pi3-disable-bt"

pelwell, 2016-05-02:
Why would you think that? What is the connection?

And FYI, for Foundation kernels the device_tree= setting is now ignored.

konradsa, 2016-05-02:
@Prindle19 : I think they are talking about "device_tree=" here, as pelwell pointed out.

konradsa, 2016-05-02:
Btw, to revert to an earlier version:

1) Find the sha hash here: https://github.com/Hexxeh/rpi-firmware/commits/master
2) sudo rpi-update <hash>

For instance, to revert to 4.1.19 use the following hash: 6e8b794818e06f50724774df3b1d4c6be0b5708c

filius-martis, 2016-05-02:
To clear things up, I already upgraded to the latest 4.4 kernel branch (https://github.com/Hexxeh/rpi-firmware) and no matter which kernel is in place, I always see the same behavior:

Connecting my PS4 controller to the integrated bluetooth controller -> the system always freezes within minutes.

Connecting my PS4 controller to an usb bluetooth stick while the internal bluetooth controller still is active -> no freezes at all

Connecting my PS4 controller to an usb bluetooth stick while the internal bluetooth controller is deactivated (dtoverlay=pi3-disable-bt-overlay in /boot/config.txt) -> no freezes at all

[63 more replies]

closed 2016-05-13 by coniferous1