raspberrypi/linux #1362: please add ntp with PPS included

beta-tester, 2016-03-22:
hi, the kernel supports PPS (pulse per second) from GPIO... see https://github.com/raspberrypi/linux/issues/109
but the apt-get package **ntp** for RPi raspbian is compiled without the necessary option to take PPS into account.
so if i want to use PPS for my ntp server, i always have to download the source and have to compile ntp newly to have that the ability to use PPS.
```
sudo apt-get install libcap-dev
wget http://archive.ntp.org/ntp4/ntp-4.2/ntp-4.2.8p6.tar.gz
tar xvfz ntp-4.2.8p6.tar.gz
cd ntp-4.2.8p6/
./configure --enable-linuxcaps
make
sudo make install
```
can you please add a compiled version of ntp to the packages for raspbian that will take PPS into account...
... thank you.

XECDesign, 2016-03-22:
This should be filed with debian, since we don't provide any special ntp package. Unless raspbian specifically disables it, in which case this should be filed with the raspbian project.

popcornmix, 2016-03-23:
Closing as this is not related to linux kernel.

closed 2016-03-23 by popcornmix