raspberrypi/linux #1363: Cannot move ACT LED functionality to GPIO pin

Cristov9000, 2016-03-23:
Previously on the Pi2 You could move the ACT led functionality to another GPIO pin by adding dtparam=act_led_gpio=XX to /boot/config.txt.  This functionality is lost on the Pi3 with the GPIO extender.  Looking for a work around.  Thank you!

pelwell, 2016-03-23:
The easy method for now is to create an overlay that overwrites the GPIO reference of the ACT LED with the original &gpio. The dtparam will then work as before. It's a bit of a hack, though.

I would have liked to be able to do this from the existing dtparam, but the __override__ syntax isn't expressive enough to allow that yet; this issue is on my list of test cases for when I get around to improving it.

pelwell, 2016-03-23:
I'll reply again when I've pushed some suitable overlays.

pelwell, 2016-03-23:
The active branches - rpi-4.1.y, rpi-4.4.y and rpi-4.5.y now include the pi3-act-led overlay. From the README:
```
Name:   pi3-act-led
Info:   Pi3 uses a GPIO expander to drive the LEDs which can only be accessed
        from the VPU. There is a special driver for this with a separate DT
        node, which has the unfortunate consequence of breaking the
        act_led_gpio and act_led_activelow dtparams.
        This overlay changes the GPIO controller back to the standard one and
        restores the dtparams.
Load:   dtoverlay=pi3-act-led,<param>=<val>
Params: activelow               Set to "on" to invert the sense of the LED
                                (default "off")

        gpio                    Set which GPIO to use for the activity LED
                                (in case you want to connect it to an external
                                device)
                                REQUIRED
```
The source will be built into future releases, but you can download a precompiled version for 4.1 [here](https://drive.google.com/open?id=0B8VsfKAD4-NOQ0d3ekVkU3ZoQWc), and for 4.4 and 4.5 [here](https://drive.google.com/file/d/0B8VsfKAD4-NON3l2RkRZbTRPVFU/view?usp=sharing).

Cristov9000, 2016-03-23:
Thank you!
If I got your correctly, I just need to add your precompiled overlay to /boot/dts/overlays and then in /boot/config.txt insert a line dtoverlay=pi3-act-led, gpio=18. Then it should work?  I will report back if I could get it working in a few.

pelwell, 2016-03-23:
Yes, that's correct.

Cristov9000, 2016-03-24:
This worked perfectly.  Thank you!

pelwell, 2016-03-24:
You're welcome.

closed 2016-03-24 by pelwell

solidcinu, 2017-01-27:
@Cristov9000 could You please help me with that??  i connected led to 17 and 12 pins on gpio, place both script compilations (4.1, and 4.4 , 4.5) on sd card (boot/overlays on this small 50mb boot partition) and line "dtoverlay=pi3-act-led, gpio=18" to config file in /boot and it does not work. The original light for sdcard stop to blink. it blink again when i comment this line i put in config file.

pelwell, 2017-01-27:
Try connecting the LED to header pins 12 (GPIO18) and 14 (GND). Be careful to get it the right way round.

solidcinu, 2017-01-27:
@pelwell thank You for You unswer:) but still dont work, but led stop ti light  - it is 100% off now, any other idea? i connect power to 12 and ground to 14

pelwell, 2017-01-27:
Can you post the content of config.txt?

solidcinu, 2017-01-27:
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details
dtoverlay=pi3-act-led, gpio=18
disable_splash=1

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1
gpu_mem=512

pelwell, 2017-01-27:
```
dtoverlay=pi3-act-led, gpio=18
```
Remove the space after the comma.

solidcinu, 2017-01-27:
@pelwell YOU ARE THE GENIUS!!!!!!! coma :) now it is working:) THANK YOU VERY MUCH!!!