raspberrypi/linux #1368: Overclocking : setting core_freq lower the SD card speed

zeludo, 2016-03-23:
Hello,

I noticed a poor sd card speed on my RPI2 setup.
After a long time investigating i found it was related to core clock speed and scaling governor.

To demonstrate this i made a simple test with different setups.

For all tests i used only these two common settings :
```
arm_freq=1000
over_voltage=5
```
then after a reboot i launched  
```bash
dd if=/dev/mmcblk0 iflag=direct of=/dev/null bs=1M count=512
```

with different core_feq and scaling governor values :

Test | core_freq | core_freq_min | scaling governor | Read spped (MB/s)
-------|----------|---------------|------------------|---------------------
1 |          |               | ondemand         | 17.3
2 |          |               | performance      | 22.1
3 |500       |               | ondemand         | **11.8**
4 |500       |               | performance      | 23.4
5 |500       | 500           | ondemand         | **13.0**
6 |500       | 500           | performance      | 23.4

**freqs left blank means not set.**

As you can see, setting core_freq lower the sd card speed !

I watched clocks during tests and with ondemand governor they seemed to always stay at their min value. Also with core_freq = core_freq_min it seemed core_freq_min was ignored.
Temperature was always under 50°C

On my real setup I have allowed a wider range for clocks (ie allowing dynamic overclocking and downclocking) and sd card speed only reach about 10 MB/s for sequential reads and 6MB/s when writing. So for now i'll leave core_freq to default.

zeludo, 2016-03-24:
I forgot to mention I'm running Arch Linux ARM with firmware de7aa7e95d4b8fdca7eea2a66ca811d54f34aeaa (2013.03.22)

pelwell, 2016-03-24:
This is the expected behaviour. The "SD Host" interface (the other SD controller in BCM2835) is clocked off the core clock, so if the VPU/GPU clock changes going into or out of turbo mode then the SD card clock scales proportionately. For obvious reasons the clock divisor has to be tuned to the maximum core clock, which means that when not in turbo mode the SD clock is reduced by the ratio between the two VPU/GPU clock frequencies.

There are a number of options:
1) Use the performance governor.
2) Use the ondemand governor, but enable the "io_is_busy" feature so that any drive activity will cause the processor to go into turbo mode for the duration. This works really well, and is what Raspbian does by default:
```
      printf " No. Switching to ondemand scaling governor"
      SYS_CPUFREQ_GOVERNOR=/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
      if [ -e $SYS_CPUFREQ_GOVERNOR ]; then
        echo "ondemand" > $SYS_CPUFREQ_GOVERNOR
        echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
        echo 100000 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_rate
        echo 50 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_down_factor
        echo 1 > /sys/devices/system/cpu/cpufreq/ondemand/io_is_busy
      fi
```
3) Fix the VPU (core) clock to the minimum (`core_freq=250`), so that it never scales.

popcornmix, 2016-04-11:
Latest (rpi-update) firmware/kernel now doesn't require io_is_busy. Can you try testing again?

zeludo, 2016-04-11:
Arch Linux has not switched to latest firmware/kernel yet, commit dff460c81cf76ec265b8ef7d0487f0077dfd093b is not included, but I'll test as soon as it is available.
By the way thanks to @pelwell for the clear explanation and for giving workarounds. It made me notice I could tune the ondemand governor to better suit my needs :)

zeludo, 2016-04-14:
I've just updated and tested, I confirm it's working exactly as I expected it to ! 
Thanks a lot guys, great work !!

closed 2016-04-14 by zeludo