raspberrypi/linux #1369: Can no longer redirect PWR LED GPIO on Pi 2B or B+

izx2, 2016-03-24:
This is on the older Pis (2B/B+) where it used to work before. From the _vcdbg_ log, it appears something is explicitly overriding _pwr_led_gpio_ to the default pin 35 before config.txt dtparams are read; as a result, the redirection in config.txt is not honored.

In relevant part, setting _pwr_led_gpio=24_ in config.txt gives this vcdbg log output on both Pi 2B/B+:

    001423.027: dtparam: pwr_led_gpio=35
    001424.818: dtdebug: Found override pwr_led_gpio
    001424.857: dtdebug:   override pwr_led_gpio: cell target gpios @ offset 4 (size 4)
    ...
    001438.254: dtparam: pwr_led_gpio=24
    001440.056: dtdebug: Found override pwr_led_gpio
    001440.094: dtdebug:   override pwr_led_gpio: cell target gpios @ offset 4 (size 4)

As expected, changing the *pwr_led_gpio* pin in the base DTB or in cmdline.txt doesn't work either. What is this black magic overriding *pwr_led_gpio=35* when this is the default assignment in the DTB??

pelwell, 2016-03-24:
I think you are mistaken. It works as advertised, but don't expect to see anything if you don't also change the trigger from "input" (a special trigger that only works on the original GPIO pin that is used to detect under-voltage conditions) to something else.

Try this:
```
dtparam=pwr_led_gpio=24,pwr_led_trigger=heartbeat
```

Yes, the firmware does set the initial value (this is necessary because the pin moved between some board revisions), but the user specified value is applied afterwards so is the one that is used.

izx2, 2016-03-24:
D'oh! I was mistaken. I should have made the connection that the UV detection wouldn't transfer over from the Pi 3 issue, but the red LED stayed on fooling me. Is there a way to turn the onboard red LED off after changing the GPIO? Thanks!

pelwell, 2016-03-24:
Not yet, unfortunately. In order to do so we have to change the GPIO direction to be an output, and that would have the effect of disabling the under-voltage detection, but it might be something we add in the future (with a warning).

Hmm, come to think of it, you could permanently under-power your Pi3! (Only joking).

izx2, 2016-03-26:
If only I could afford a bench power supply for every one of my Pis...! (joking). For now, a little piece of tape or a strategic dab of permanent marker should suffice to make the red light less blinding.

But it looks like I can have my cake and eat it too! On the B+/2B, undervoltage indication continues to work even after the GPIO is changed. Nice ~~bug~~ feature! Thank you again for the quick response...closing.

closed 2016-03-26 by izx2