raspberrypi/linux #1370: IPV6_SUBTREES not enabled in kernel

dtaht, 2016-03-24:
the IETF homenet working group has mandated that source specific routing be available in homenet devices.... and while it can be emulated using the ip -6 rule database, it is simpler, cleaner, and safer
to just enable IPV6_SUBTREES=y in the kernel build. 

This functionality is used by babel and other source specific routing protocols.

Can this kernel option be enabled in future builds?

popcornmix, 2016-04-06:
This is now added. Can you rpi-update and test?

dtaht, 2016-04-06:
I can confirm IPV6_SUBTREES works on this kernel 4.1.21-v7+ #872 on both the rpi2 and rpi3. The rpi3 is still experiencing #1371, but that is another issue entirely.

ip -6 route | grep "from"

default from 2001:551:6045:bd:1ccf:b9bb:e8da:2c77 via fe80::100d:7fff:fe64:c990 dev eth0  proto 42  metric 1024 
default from 2601:641:4103:56c0::/60 via fe80::100d:7fff:fe64:c990 dev eth0  proto 42  metric 1024 
2601:641:4103:56c0::/60 from 2601:641:4103:56c0::/64 via fe80::100d:7fff:fe64:c990 dev eth0  proto 42  metric 1024

closed 2016-04-06 by dtaht