raspberrypi/linux #1373: "apt-get dist-upgrade" on a clean Raspbian 1.9 install outputs many pypy-upstream removal warnings

DanTup, 2016-03-25:
I've tried this twice (once was a network install from NOOBS Lite, and more recently by downloading the full NOOBS install that has Raspbian 1.9 included from a few days ago). When I run "apt-get dist-upgrade" it says it's going to remove pypy-upstream-doc and update a bunch of other python-related stuff.

It then fills the screen with failures to remove non-entry directories (screenshot attached).

It's not clear whether these are leaving anything in a bad state, but I presume it's not expected.

[image]

XECDesign, 2016-03-25:
This is mostly harmless issue. It seems like the pypy-upstream package wasn't done very well and has been outdated for a while. We switched to Debian's pypy package, but when pypy-upstream removes itself, it leaves behind some cruft. This can be removed by (carefully) running `sudo rm -rf /usr/lib/pypy-upstream`.

Also, please note that this repo is for the linux kernel. If you would like to report other distro issues or continue the discussion on this particular issue, the best place for that is here
https://github.com/RPi-Distro/repo/issues/

closed 2016-03-25 by XECDesign

DanTup, 2016-03-25:
Oops sorry, didn't realise (bit of a Pi/Linux noob!).

Thanks for the info, I wasn't sure if the new libs put stuff in the `pypy-upstream` folder or it was safe to remove. I'll just wipe it out in my setup script after the updates.

Thanks!