raspberrypi/linux #1374: Pi 3 freeze during load

patraulea, 2016-03-25:
Fresh install of raspbian-lite 2016-03-18 image on a sandisk extreme pro (SDSDQXP-016G-X46). Using a 2A 5V microusb power supply, will post the model when I get home. No addons, only ethernet and power attached, and wifi disabled (blacklist brcmfmac, blacklist brcmutil).

All is well until I start transferring data over ethernet, or do some ffmpeg decoding. Have downloaded about 20GB using rtorrent so far at 5MB/s, no problems yet. 

Have received the Pi yesterday, saw one crash while copying data over samba from it, and another after 5 min of ffmpeg -i input.mp4 -f null /dev/null. Connected HDMI after crash, no signal on display. Will test again with power from a desktop USB port.

Two questions:

- Is there a way to save a dmesg output on panic, or do I have to have HDMI connected at all times ?

- How do I measure TP1 & TP2 voltage on Pi 3 ? I couldn't find where the test points are on the new board.

popcornmix, 2016-03-25:
No, dmesg won't be flushed on a panic. A uart is once solution. HDMI the other.

Measure 5V from the GPIO header:
https://www.element14.com/community/docs/DOC-73950/l/raspberry-pi-3-model-b-gpio-40-pin-block-pinout

patraulea, 2016-03-27:
Tried with power from desktop USB port, still seeing crashes after running idle a few hours. TP1-TP2 voltage was 4.99 idle, 4.91 with one core at 100%.

Messages from tail -f /var/log/syslog:

> Mar 26 13:04:10 raspberrypi kernel: [ 3787.484330] mmc0: timeout waiting for hardware interrupt.
> Mar 26 13:04:10 raspberrypi kernel: [ 3787.484356] [e110e20f] REQ> b8cc3ce0 0

and fs was remounted r/o. `cd /usr` works, running anything (e.g. `ls`) outputs 'No such file or directory'. 

The sdcard serial number I mentioned initially seems to be for a full-size SD (store possibly mixed them up). More likely it's this one: SDSDQXP-016G-AFFP (mentioned on http://elinux.org/RPi_SD_cards as "boots but random umounts occur and requires reboots, no work around found").

Question: is there a possible fix for this sdcard family that hasn't made it into jessie from 2016-03-18 (the build I'm using) ?

Ruffio, 2017-02-05:
@patraulea is this still an issue?

closed 2017-05-18 by JamesH65