raspberrypi/linux #1375: kernel trace caused by using hostapd on Pi3

Darren, 2016-03-26:
pi@rpi:~ $ cat /boot/.firmware_revision 
692dde0c1586f7310301379a502b9680d0c104fd

[31211.436419] ------------[ cut here ]------------
[31211.436503] WARNING: CPU: 2 PID: 344 at drivers/net/wireless/brcm80211/brcmfmac/core.c:1144 brcmf_netdev_wait_pend8021x+0xfc/0x108 [brcmfmac]()
[31211.436512] Modules linked in: tun ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ip_tables x_tables bnep hci_uart btbcm bluetooth brcmfmac brcmutil cfg80211 rfkill snd_bcm2835 bcm2835_rng bcm2835_gpiomem snd_pcm snd_timer snd uio_pdrv_genirq uio fuse ipv6
[31211.436637] CPU: 2 PID: 344 Comm: hostapd Not tainted 4.1.20-v7+ #867
[31211.436646] Hardware name: BCM2709
[31211.436682] [<800185e0>] (unwind_backtrace) from [<80013f48>] (show_stack+0x20/0x24)
[31211.436705] [<80013f48>] (show_stack) from [<80572ddc>] (dump_stack+0xd4/0x118)
[31211.436727] [<80572ddc>] (dump_stack) from [<800271b4>] (warn_slowpath_common+0x98/0xc8)
[31211.436746] [<800271b4>] (warn_slowpath_common) from [<800272a0>] (warn_slowpath_null+0x2c/0x34)
[31211.436805] [<800272a0>] (warn_slowpath_null) from [<7f2596bc>] (brcmf_netdev_wait_pend8021x+0xfc/0x108 [brcmfmac])
[31211.436895] [<7f2596bc>] (brcmf_netdev_wait_pend8021x [brcmfmac]) from [<7f248878>] (send_key_to_dongle+0xa4/0xf8 [brcmfmac])
[31211.436953] [<7f248878>] (send_key_to_dongle [brcmfmac]) from [<7f248eb0>] (brcmf_cfg80211_add_key+0x16c/0x314 [brcmfmac])
[31211.437041] [<7f248eb0>] (brcmf_cfg80211_add_key [brcmfmac]) from [<7f134b8c>] (nl80211_new_key+0x100/0x268 [cfg80211])
[31211.437111] [<7f134b8c>] (nl80211_new_key [cfg80211]) from [<804c839c>] (genl_rcv_msg+0x26c/0x3ec)
[31211.437130] [<804c839c>] (genl_rcv_msg) from [<804c7584>] (netlink_rcv_skb+0xb0/0xcc)
[31211.437148] [<804c7584>] (netlink_rcv_skb) from [<804c8120>] (genl_rcv+0x34/0x44)
[31211.437165] [<804c8120>] (genl_rcv) from [<804c6ec8>] (netlink_unicast+0x180/0x244)
[31211.437182] [<804c6ec8>] (netlink_unicast) from [<804c7360>] (netlink_sendmsg+0x30c/0x378)
[31211.437203] [<804c7360>] (netlink_sendmsg) from [<8047d528>] (sock_sendmsg+0x24/0x34)
[31211.437224] [<8047d528>] (sock_sendmsg) from [<8047dfec>] (___sys_sendmsg+0x1dc/0x1e4)
[31211.437241] [<8047dfec>] (___sys_sendmsg) from [<8047ed44>] (__sys_sendmsg+0x4c/0x7c)
[31211.437258] [<8047ed44>] (__sys_sendmsg) from [<8047ed8c>] (SyS_sendmsg+0x18/0x1c)
[31211.437277] [<8047ed8c>] (SyS_sendmsg) from [<8000fa20>] (ret_fast_syscall+0x0/0x54)
[31211.437288] ---[ end trace b47cbbf8c50b64b5 ]---
[31213.436454] brcmfmac: send_key_to_dongle: wsec_key error (-52)
[31215.436477] brcmfmac: brcmf_cfg80211_del_station: SCB_DEAUTHENTICATE_FOR_REASON failed -52

asavah, 2016-03-27:
Same here,
`Linux rpi3 4.4.6-v7+ #1 SMP Sat Mar 26 20:35:03 EET 2016 armv7l GNU/Linux`
built on 6ec2115ccc87288132b345ab2cd05aed2d3beb97,  bcm2709_defconfig

Firmware `version e16cd015026a8bda22bb50dc30b30ce07f52a202 (clean) (release)`

Homegrown OS.

eth0 bridged with wlan0 into br0 via systemd network files,  no nat, no iptables rules, plain AP bridged into my LAN

There was no wireless traffic at that point of time, and my android smartphone is set up to use this AP.  

Everything continued working fine after the backtrace, AP works, ethernet works, nothing crashed.

hostapd config:

```
ssid=PI3_TEST
wpa_passphrase=snip
macaddr_acl=0
auth_algs=1
interface=wlan0
bridge=br0
auth_algs=3
channel=7
driver=nl80211
hw_mode=g
#logger_stdout=-1
#logger_stdout_level=2
max_num_sta=10
rsn_pairwise=CCMP
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
channel=6            # the channel to use
ieee80211d=1          # limit the frequencies used to those allowed in the country
country_code=DE       # the country code
ieee80211n=1          # 802.11n support
wmm_enabled=1         # QoS support
ht_capab=[HT20][SHORT-GI-20]
```

backtrace:
```
Mar 27 08:54:34 rpi3 kernel: eth0: hw csum failure
Mar 27 08:54:34 rpi3 kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C      4.4.6-v7+ #1
Mar 27 08:54:34 rpi3 kernel: Hardware name: BCM2709
Mar 27 08:54:34 rpi3 kernel: [<8001852c>] (unwind_backtrace) from [<80013edc>] (show_stack+0x20/0x24)
Mar 27 08:54:34 rpi3 kernel: [<80013edc>] (show_stack) from [<8031cf70>] (dump_stack+0xd4/0x118)
Mar 27 08:54:34 rpi3 kernel: [<8031cf70>] (dump_stack) from [<804bfdd0>] (netdev_rx_csum_fault+0x44/0x48)
Mar 27 08:54:34 rpi3 kernel: [<804bfdd0>] (netdev_rx_csum_fault) from [<804b3ad4>] (__skb_checksum_complete+0xb4/0xb8)
Mar 27 08:54:34 rpi3 kernel: [<804b3ad4>] (__skb_checksum_complete) from [<805adf48>] (ipv6_mc_validate_checksum+0xa8/0x158)
Mar 27 08:54:34 rpi3 kernel: [<805adf48>] (ipv6_mc_validate_checksum) from [<804b08f4>] (skb_checksum_trimmed+0x9c/0x190)
Mar 27 08:54:34 rpi3 kernel: [<804b08f4>] (skb_checksum_trimmed) from [<805ae0fc>] (ipv6_mc_check_mld+0x104/0x348)
Mar 27 08:54:34 rpi3 kernel: [<805ae0fc>] (ipv6_mc_check_mld) from [<7f20e56c>] (br_multicast_rcv+0x68/0xb58 [bridge])
Mar 27 08:54:34 rpi3 kernel: [<7f20e56c>] (br_multicast_rcv [bridge]) from [<7f2050e0>] (br_handle_frame_finish+0x1b4/0x5b0 [bridge]
Mar 27 08:54:34 rpi3 kernel: [<7f2050e0>] (br_handle_frame_finish [bridge]) from [<7f205654>] (br_handle_frame+0x178/0x94c [bridge])
Mar 27 08:54:34 rpi3 kernel: [<7f205654>] (br_handle_frame [bridge]) from [<804bd7fc>] (__netif_receive_skb_core+0x3a0/0xb74)
Mar 27 08:54:34 rpi3 kernel: [<804bd7fc>] (__netif_receive_skb_core) from [<804bfe94>] (__netif_receive_skb+0x20/0x7c)
Mar 27 08:54:34 rpi3 kernel: [<804bfe94>] (__netif_receive_skb) from [<804c0ed0>] (process_backlog+0xb4/0x16c)
Mar 27 08:54:34 rpi3 kernel: [<804c0ed0>] (process_backlog) from [<804c0534>] (net_rx_action+0x28c/0x3f4)
Mar 27 08:54:34 rpi3 kernel: [<804c0534>] (net_rx_action) from [<80028d1c>] (__do_softirq+0x190/0x3d0)
Mar 27 08:54:34 rpi3 kernel: [<80028d1c>] (__do_softirq) from [<800292fc>] (irq_exit+0xdc/0x140)
Mar 27 08:54:34 rpi3 kernel: [<800292fc>] (irq_exit) from [<80070ad0>] (__handle_domain_irq+0x70/0xc4)
Mar 27 08:54:34 rpi3 kernel: [<80070ad0>] (__handle_domain_irq) from [<800108f0>] (handle_IRQ+0x28/0x2c)
Mar 27 08:54:34 rpi3 kernel: [<800108f0>] (handle_IRQ) from [<80009518>] (bcm2836_arm_irqchip_handle_irq+0xb8/0xbc)
Mar 27 08:54:34 rpi3 kernel: [<80009518>] (bcm2836_arm_irqchip_handle_irq) from [<805e8f04>] (__irq_svc+0x44/0x5c)
Mar 27 08:54:34 rpi3 kernel: Exception stack(0x80897f08 to 0x80897f50)
Mar 27 08:54:34 rpi3 kernel: 7f00:                   00000000 aafa8348 00000000 808994ec 80896000 808985dc
Mar 27 08:54:34 rpi3 kernel: 7f20: 80898500 80898580 805ede7c 808fd598 8086fa30 80897f64 80897f58 80897f58
Mar 27 08:54:34 rpi3 kernel: 7f40: 800109b0 800109b4 60000013 ffffffff
Mar 27 08:54:34 rpi3 kernel: [<805e8f04>] (__irq_svc) from [<800109b4>] (arch_cpu_idle+0x34/0x4c)
Mar 27 08:54:34 rpi3 kernel: [<800109b4>] (arch_cpu_idle) from [<80063a50>] (default_idle_call+0x34/0x48)
Mar 27 08:54:34 rpi3 kernel: [<80063a50>] (default_idle_call) from [<80063c7c>] (cpu_startup_entry+0x218/0x2b4)
Mar 27 08:54:34 rpi3 kernel: [<80063c7c>] (cpu_startup_entry) from [<805e3888>] (rest_init+0x88/0x8c)
Mar 27 08:54:34 rpi3 kernel: [<805e3888>] (rest_init) from [<8081ed34>] (start_kernel+0x3dc/0x3e8)

```

pelwell, 2016-03-29:
Duplicate of https://github.com/raspberrypi/linux/issues/1083?

pelwell, 2016-03-29:
@Darren can you give more information about what you did so we can try to reproduce this?

Darren, 2016-03-30:
@pelwell

I try to reproduce this using iperf to make the network load saturated, but it did not happen.

But I noticed that it occurs when I watch youtube over my Apple TV which is connected to Pi3's wifi.
I have seen this happens for three times so far.

as opposed to asavah's case, when kernel show the trace above, wifi disappears and  the Apple TV's wifi got disconnected untill Pi 3 is rebooted.

asavah, 2016-03-30:
> @asavah Duplicate of #1083?

@pelwell looks like it, I'll try to apply the patches/commits mentioned and report back (in the right issue if it's the case), and I forgot to mention I do have a public routable ipv6 prefix in my lan.

raintonr, 2016-04-03:
Sounds a bit like #1313?

Darren, 2016-05-01:
it still happens with 4.4.8-v7+

JamesH65, 2017-05-04:
Please retest with the latest Raspbian - we have a recent fix for bridged wifi. Please report back and close if appropriate.

closed 2017-05-18 by JamesH65