raspberrypi/linux #1376: SD error after latest kernel update

ezar, 2016-03-26:
[16046.468483] mmc0: tried to reset card
[17268.064776] mmcblk0: r/w command failed, status = 0x80000900
[17268.133000] mmc0: tried to reset card
[17578.346068] mmcblk0: r/w command failed, status = 0x80000900
[17578.414149] mmc0: tried to reset card
[17889.414099] mmcblk0: r/w command failed, status = 0x80000900
[17889.485312] mmc0: tried to reset card
[19730.076646] mmcblk0: r/w command failed, status = 0x80000900
[19730.142663] mmc0: tried to reset card
[20350.603860] mmcblk0: r/w command failed, status = 0x80000900
[20350.675079] mmc0: tried to reset card
[21581.684853] mmcblk0: r/w command failed, status = 0x80000900
[21581.750182] mmc0: tried to reset card
[28347.661371] mmcblk0: r/w command failed, status = 0x80000900
[28347.730299] mmc0: tried to reset card
[29268.895503] mmcblk0: r/w command failed, status = 0x80000900
[29268.964539] mmc0: tried to reset card
[30309.870086] mmcblk0: r/w command failed, status = 0x80000900
[30309.939286] mmc0: tried to reset card

popcornmix, 2016-03-26:
Pi1, Pi2 or Pi3?
What does ``vcgencmd version`` and ``uname -a`` report?

ezar, 2016-03-26:
- Pi2
- _uname -a_
  Linux minibian 4.1.20-v7+ #867 SMP Wed Mar 23 20:12:32 GMT 2016 armv7l GNU/Linux
- _vcgencmd version_
  Mar 24 2016 12:40:17 
  Copyright (c) 2012 Broadcom
  version e16cd015026a8bda22bb50dc30b30ce07f52a202 (clean) (release)

ezar, 2016-03-28:
Please close this issue. 
I disable webmin and this fails gone.
Sorry.

pelwell, 2016-03-28:
Thanks for telling us.

closed 2016-03-28 by pelwell