raspberrypi/linux #1377: Where can I increase the USB receive buffer size?

IeiuniumLux, 2016-03-26:
I’m connecting a [TeraTower](http://www.teraranger.com/teraranger-tower/) with eight(8) cameras to the RPi2B and the USB receive buffer of the real-time kernel seems to be too small to handle the TeraTower transfer rate.  Can I increase the buffer in the /drivers/usb/core/devio.c file (i.e. USB_SG_SIZE)?  Otherwise, can you please advise?

P33M, 2016-03-29:
Please provide a `lsusb -v` listing.

abencomo, 2016-03-29:
pi@navio:~ $ lsusb -v

Bus 001 Device 005: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x7392 Edimax Technology Co., Ltd
  idProduct          0x7811 EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           46
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           4
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
[... 261 lines cut ...]
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

abencomo, 2016-03-29:
And the Kernel7 installed in the SD card was built out of this repo:

https://github.com/emlid/linux-rt-rpi

P33M, 2016-03-29:
What's the expected data rate to/from from the ST micro device? I see it's emulating a UART but this doesn't mean much to the lower layers of the software stack.

Full-speed bulk IN/OUT has a few caveats regarding maximum transfer speed. We can manage an absolute maximum of ~45% theoretical throughput per-device so you won't get more than ~5mbit/s. 

There's also an aggressive back-off to prevent interrupt storms - a delay of 1ms is inserted when a bulk endpoint either reports that it can't accept any more data (out) or can't provide any more data (in). If your device has shallow buffers, then you could be running into this latency limitation.

P33M, 2016-03-29:
Pending confirmation of your expected data rate, there is a dial you can turn:

In `/boot/cmdline.txt` `dwc_otg.nak_holdoff=x` controls the hold-off interval for low/full-speed endpoints. 8 microframes is the default. Try setting this to 2 and see if this improves things.

abencomo, 2016-03-29:
The expected data rate is 1kHz (in fast mode) - up to 600Hz (in precision mode).  I'm currently running them in precision mode since in fast mode the usb buffer seems to overflow.

Let me set **dwc_otg.nak_holdoff** to **2** and run a quick test and thanks for your help to troubleshoot this.

abencomo, 2016-03-29:
My original /boot/cmdline.txt contains these:

`console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait`

However, when I add dwc_otg.nak_holdoff=2 and reboot they Pi, the entry is dwc_otg.nak_holdoff gone.

Is that normal?  Sorry, Linux is not my forte.

P33M, 2016-03-29:
Are you rebooting via commandline or by power cycle? Hint: you should reboot via commandline.

abencomo, 2016-03-29:
I was doing the latter.  With "_sudo reboot_" the **dwc_otg.nak_holdoff** entry is persisted.  Unfortunately, adding **dwc_otg.nak_holdoff=2** didn't resolve the problem.

I noticed that the forked [cdc-acm.c](https://github.com/emlid/linux-rt-rpi/blob/rpi-3.18.9-rt5/drivers/usb/class/cdc-acm.c) file doesn't have the latest fixes as the one in the raspberrypi [repo](https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/usb/class/cdc-acm.c).

Could those changes have fixed the buffer size problem?

P33M, 2016-03-29:
We don't offer support for bespoke images. There's no guarantee that some bug could have been fixed in our own releases, which closely track official kernel releases (plus a variable amount of glue to keep a small number of out-of-tree things working). I note that the branch in question in the 3rd-party repo is 3.18 - we haven't used that for about a year.

Does your project work with the official Raspberry Pi-provided kernel/firmware?

abencomo, 2016-03-29:
We're using the [NAVIO2](http://www.emlid.com) hat as a Linux autopilot for the Raspberry Pi and that is the RT linux kernel that Emlid provides or recommends to use with their hardware.  Therefore, our project is not using the official Raspberry Pi-provided kernel/firmware since it's not trimmed down as RT ready kernel by default.

I also understand your point about not offering support for bespoke images so I really appreciate you took the time to provide some suggestions.  Feel free to close this issue.  Thanks again!

P33M, 2016-03-29:
RT-kernels in general meddle with so much stuff at the interrupt-handling layer that I'm not surprised things do go wrong. The dwc_otg driver in particular has never been proven to work reliably under -rt patches. I was sort-of hoping for a simple case where it was a throughput issue, but apparently this is not so.

closed 2016-03-29 by P33M