raspberrypi/linux #1378: Default "fast boot" setting is confusing and causes services to be unreliable

DanTup, 2016-03-27:
I don't know if this is the right place for this and I don't think this will be changed, but I wanted to provide feedback in any case. I've spent many hours yesterday evening and most of today trying to figure out why I had services randomly failing at startup - they appeared to be starting during the DHCP process when there was no DNS (postfix takes a copy of /etc/resolv.conf into its chroot during the DHCP process which results in a broken postfix that can't resolve any DNS).

I eventually stumbled upon this magic setting:

[image]

And suddenly everything was explained!

Since I can't script raspi-config, I've taken the code [from here](https://github.com/raspberrypi/rc_gui/blob/bf829a6cd4a130eaa421e1a39cee6388301f8414/raspi-config#L868) and added it to my setup script. This has fixed both postfix, and all of the other DNS-related errors that fill syslog on a default Raspian install.

I understand that people want faster boots, but to me it seems a little crazy to have a faster boot at the expense of working services and filling syslog with errors at boot. Maybe I'm the first person to spend > 10 hours debugging before finding this setting; maybe I'm not. It's frustrating in any case. I think the default should be the safe option, and if people want faster boots, they're probably already making tweaks and they can change this (and understand the implications).

popcornmix, 2016-03-27:
This repo is for linux kernel issues, whereas this issue is due to raspberry pi specific parts of distribution.
Can you open this issue here: https://github.com/RPi-Distro/repo/issues/ where the right devs will see it.

closed 2016-03-27 by popcornmix

DanTup, 2016-03-27:
Will do; thanks (and sorry!) :)