raspberrypi/linux #1380: RPi3 Wi-Fi error when establishing a lease

phelum, 2016-03-28:
Hi,
I am seeing many problems when establishing a Wi-Fi connection.  After a network is found a lease is requested.  When granted and reported the system receives an error saying the network is down.  The connection establishment process starts again and is terminated once a lease is again granted.  Generally the end result is that the connection fails.  Occasionally it does work.

If I use a dongle instead of the 43430 everything works fine.

Here is a log of the failure then success when using the 43430:
`Mar 27 23:05:26 localhost dhclient: Internet Systems Consortium DHCP Client 4.2.2
Mar 27 23:05:26 localhost dhclient: Copyright 2004-2011 Internet Systems Consortium.
Mar 27 23:05:26 localhost dhclient: All rights reserved.
Mar 27 23:05:26 localhost dhclient: For info, please visit https://www.isc.org/software/dhcp/
Mar 27 23:05:26 localhost dhclient: 
Mar 27 23:05:26 localhost dhclient: Listening on LPF/wlan0/b8:27:eb:b5:74:80
Mar 27 23:05:26 localhost dhclient: Sending on   LPF/wlan0/b8:27:eb:b5:74:80
Mar 27 23:05:26 localhost dhclient: Sending on   Socket/fallback
Mar 27 23:05:28 localhost ifplugd(wlan0)[1671]: Link beat detected.
Mar 27 23:05:28 localhost ifplugd(wlan0)[1671]: Executing '/etc/ifplugd/ifplugd.action wlan0 up'.
Mar 27 23:05:28 localhost ifplugd(wlan0)[1671]: client: /sbin/ifup: interface wlan0 already configured
Mar 27 23:05:28 localhost kernel: [   21.963038] cfg80211: Calling CRDA to update world regulatory domain
Mar 27 23:05:28 localhost ifplugd(wlan0)[1671]: Program executed successfully.
Mar 27 23:05:28 localhost dhclient: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 2
Mar 27 23:05:28 localhost dhclient: DHCPOFFER from 192.168.1.1
Mar 27 23:05:29 localhost dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Mar 27 23:05:29 localhost dhclient: DHCPACK from 192.168.1.1
Mar 27 23:05:29 localhost dhclient: bound to 192.168.1.101 -- renewal in 20698 seconds.
Mar 27 23:05:30 localhost kernel: [   23.591735] cfg80211: Calling CRDA to update world regulatory domain
Mar 27 23:05:30 localhost ifplugd(wlan0)[1671]: Link beat lost.
Mar 27 23:05:30 localhost dhclient: receive_packet failed on wlan0: Network is down
Mar 27 23:05:30 localhost kernel: [   24.148410] brcmfmac: power management disabled
Mar 27 23:05:30 localhost kernel: [   24.186949] cfg80211: Calling CRDA to update world regulatory domain
Mar 27 23:05:30 localhost dhclient: Internet Systems Consortium DHCP Client 4.2.2
Mar 27 23:05:30 localhost dhclient: Copyright 2004-2011 Internet Systems Consortium.
Mar 27 23:05:30 localhost dhclient: All rights reserved.
Mar 27 23:05:30 localhost dhclient: For info, please visit https://www.isc.org/software/dhcp/
Mar 27 23:05:30 localhost dhclient: 
Mar 27 23:05:30 localhost dhclient: Listening on LPF/wlan0/b8:27:eb:b5:74:80
Mar 27 23:05:30 localhost dhclient: Sending on   LPF/wlan0/b8:27:eb:b5:74:80
Mar 27 23:05:30 localhost dhclient: Sending on   Socket/fallback
Mar 27 23:05:31 localhost kernel: [   25.123059] cfg80211: Calling CRDA to update world regulatory domain
Mar 27 23:05:31 localhost dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Mar 27 23:05:32 localhost ifplugd(wlan0)[1671]: Link beat detected.
Mar 27 23:05:33 localhost ntpd[1901]: Listen normally on 2 wlan0 192.168.1.101 UDP 123
Mar 27 23:05:33 localhost ntpd[1901]: peers refreshed
Mar 27 23:05:33 localhost dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Mar 27 23:05:33 localhost dhclient: DHCPACK from 192.168.1.1
Mar 27 23:05:33 localhost dhclient: bound to 192.168.1.101 -- renewal in 18390 seconds.
`

Here is a successful start (same access point) when using a dongle:
`Mar 27 22:51:40 localhost dhclient: Internet Systems Consortium DHCP Client 4.2.2
Mar 27 22:51:40 localhost dhclient: Copyright 2004-2011 Internet Systems Consortium.
Mar 27 22:51:40 localhost dhclient: All rights reserved.
Mar 27 22:51:40 localhost dhclient: For info, please visit https://www.isc.org/software/dhcp/
Mar 27 22:51:40 localhost dhclient: 
Mar 27 22:51:40 localhost dhclient: Listening on LPF/wifi/00:0f:55:a8:c5:48
Mar 27 22:51:40 localhost dhclient: Sending on   LPF/wifi/00:0f:55:a8:c5:48
Mar 27 22:51:40 localhost dhclient: Sending on   Socket/fallback
Mar 27 22:51:41 localhost kernel: [   21.514017] wifi: authenticate with c0:b0:27:dc:02:03
Mar 27 22:51:41 localhost kernel: [   21.557992] wifi: send auth to c0:b0:27:dc:02:03 (try 1/3)
Mar 27 22:51:41 localhost kernel: [   21.559477] wifi: authenticated
Mar 27 22:51:41 localhost kernel: [   21.563102] wifi: associate with c0:b0:27:dc:02:03 (try 1/3)
Mar 27 22:51:41 localhost kernel: [   21.566092] wifi: RX AssocResp from c0:b0:27:dc:02:03 (capab=0x411 status=0 aid=1)
Mar 27 22:51:41 localhost kernel: [   21.570399] wifi: associated
Mar 27 22:51:42 localhost kernel: [   21.853106] cfg80211: Calling CRDA to update world regulatory domain
Mar 27 22:51:42 localhost ifplugd(wifi)[1788]: Link beat detected.
Mar 27 22:51:42 localhost ifplugd(wifi)[1788]: Executing '/etc/ifplugd/ifplugd.action wifi up'.
Mar 27 22:51:42 localhost ifplugd(wifi)[1788]: client: /sbin/ifup: interface wifi already configured
Mar 27 22:51:42 localhost ifplugd(wifi)[1788]: Program executed successfully.
Mar 27 22:51:43 localhost dhclient: DHCPDISCOVER on wifi to 255.255.255.255 port 67 interval 5
Mar 27 22:51:43 localhost dhclient: DHCPOFFER from 192.168.1.1
Mar 27 22:51:44 localhost dhclient: DHCPREQUEST on wifi to 255.255.255.255 port 67
[... 68 lines cut ...]
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: Copyright 2004-2011 Internet Systems Consortium.
Mar 28 09:18:47 localhost dhclient: All rights reserved.
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: All rights reserved.
Mar 28 09:18:47 localhost dhclient: For info, please visit https://www.isc.org/software/dhcp/
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: For info, please visit https://www.isc.org/software/dhcp/
Mar 28 09:18:47 localhost dhclient: 
Mar 28 09:18:47 localhost dhclient: Listening on LPF/wlan0/b8:27:eb:b5:74:80
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: Listening on LPF/wlan0/b8:27:eb:b5:74:80
Mar 28 09:18:47 localhost dhclient: Sending on   LPF/wlan0/b8:27:eb:b5:74:80
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: Sending on   LPF/wlan0/b8:27:eb:b5:74:80
Mar 28 09:18:47 localhost dhclient: Sending on   Socket/fallback
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: Sending on   Socket/fallback
Mar 28 09:18:47 localhost dhclient: DHCPRELEASE on wlan0 to 192.168.8.1 port 67
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: DHCPRELEASE on wlan0 to 192.168.8.1 port 67
Mar 28 09:18:47 localhost dhclient: send_packet: Network is unreachable
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: send_packet: Network is unreachable
Mar 28 09:18:47 localhost dhclient: send_packet: please consult README file regarding broadcast address.
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: send_packet: please consult README file regarding broadcast address.
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: client: Reloading /etc/samba/smb.conf: smbd only.
Mar 28 09:18:47 localhost ifplugd(wlan0)[1720]: Program executed successfully.
`
Previously I thought the error occurred when I had multiple access points defined in my wpa_supplicant conf file.  I no longer suspect this is the cause of the trouble.

Cheers,
Steven

phelum, 2016-03-28:
My apologies that the line feeds in the code sections apparently do not work.

-- Steven

XECDesign, 2016-03-28:
Fixed the line breaks. It works if you use three backticks instead of one. Single one is for inline code.

phelum, 2016-03-28:
Thanks for the hint about line breaks.  My RPi3 is working now and my problem could be user error.  I'll close this issue and re-post it if I'm sure it is not my fault.  Apologies for the noise.

Cheers,
Steven

closed 2016-03-28 by phelum