raspberrypi/linux #1381: [4.4] kernel crash with omxplayer

fabled, 2016-03-28:
I tried the 4.4.6 kernel today, and it crashed with the following dump:
```
[  140.332952] Unhandled fault: page domain fault (0x01b) at 0xa484fbc0
[  140.332990] pgd = dba48000
[  140.333003] [a484fbc0] *pgd=24cd3831, *pte=2ca477df, *ppte=2ca47e7f
[  140.333035] Internal error: : 1b [#1] PREEMPT SMP ARM
[  140.333083] Modules linked in: ipv6 nls_utf8 cfg80211 rfkill snd_soc_wm5102 snd_soc_wm_adsp snd_soc_arizona gpio_arizona arizona_micsupp i2c_dev af_packet evdev arizona_spi arizona_ldo1 snd_soc_rpi_wsp snd_soc_bcm2835_i2s snd_soc_wm8804_i2c snd_soc_wm8804 regmap_i2c snd_soc_core snd_pcm_dmaengine snd_pcm rotary_encoder snd_timer regmap_spi gpio_keys fixed rtc_ds1307 bcm2835_gpiomem bcm2835_rng snd bcm2835_wdt uio_pdrv_genirq uio uas fb_st7735r(C) fbtft(C) fb_sys_fops syscopyarea sysfillrect sysimgblt gpio_mcp23s08 spi_bcm2835 i2c_bcm2708 squashfs lz4_decompress xz_dec
[  140.333434] CPU: 1 PID: 1324 Comm: omxplayer Tainted: G         C      4.4.6-0-rpi2 #1-Alpine
[  140.333471] Hardware name: BCM2709
[  140.333492] task: dcc0db00 ti: dbada000 task.ti: dbada000
[  140.333528] PC is at memcpy+0x48/0x330
[  140.333556] LR is at __copy_from_user_memcpy+0xa8/0x1ec
[  140.333582] pc : [<b02954a8>]    lr : [<b02a175c>]    psr: 80000013
[  140.333582] sp : dbadbd94  ip : 00000000  fp : b07a2a2c
[  140.333626] r10: a484fbc0  r9 : 00000054  r8 : 00000000
[  140.333650] r7 : de8ad430  r6 : dbada000  r5 : 00000054  r4 : de290dc8
[  140.333675] r3 : 0000013c  r2 : ffffffd4  r1 : a484fbc0  r0 : de8ad430
[  140.333702] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  140.333729] Control: 10c5387d  Table: 2ba4806a  DAC: 00000051
[  140.333754] Process omxplayer (pid: 1324, stack limit = 0xdbada210)
[  140.333778] Stack: (0xdbadbd94 to 0xdbadc000)
[  140.333804] bd80:                                              00000054 dbada000 de8ad430
[  140.333846] bda0: 00000000 de8ad430 de290dc8 b02a175c 00002920 0000013c dda4e980 00000054
[  140.333886] bdc0: 00000051 00000002 dbadbee0 dbadbed0 00000003 de8ad428 00000008 b0333314
[  140.333927] bde0: b07ac1a8 dcdae200 00000002 b032a370 00000000 00000002 dbada000 b0089000
[  140.333969] be00: 00000010 60000013 00000005 b07b06d4 de8ad420 05010009 de880194 b070e300
[  140.334010] be20: 00000010 00000080 fff9fee9 ffffffff 00000000 00000000 b06f0040 00000003
[  140.334051] be40: a484fbc0 dcdae200 dbadbed0 a484f980 dcdae200 00000000 dbadbed0 b032c860
[  140.334093] be60: 00000003 0000005c 00000001 d4c36600 a484f980 00000051 400cc404 b070e30c
[  140.334134] be80: d4c36600 b03307a8 dda4b844 00000001 a6f741bc b0088ed4 008fe494 dcd1b228
[  140.334175] bea0: dcd1b228 a49f5000 dcc5f500 00000c88 00000000 00000001 00000081 a49f5c88
[  140.334217] bec0: 00000001 00011010 00000003 a484fa14 a484f9f4 00000004 a484f9f8 00000004
[  140.334257] bee0: a484fbc0 00000054 00001779 00000000 b06f4f20 a43a1d1c dbadbf14 b0040910
[  140.334298] bf00: 00000000 00000000 0066c274 00000000 dd864e00 a484f980 ddb41720 dccef840
[  140.334340] bf20: 00000004 a484f980 dbada000 00000000 7600d1c4 b0126b94 00000000 00000081
[  140.334381] bf40: a49f5c88 00000001 dbada000 00000000 a6f741bc b008b420 a484f95c 00000000
[  140.334421] bf60: ffffffff 00000003 dccef840 b01308dc dccef841 00000004 dccef840 400cc404
[  140.334465] bf80: a484f980 dbada000 00000000 b0126e2c a6c03294 a484f980 a6f741bc 00000036
[  140.334507] bfa0: b000f724 b000f560 a6c03294 a484f980 00000004 400cc404 a484f980 a484f970
[  140.334549] bfc0: a6c03294 a484f980 a6f741bc 00000036 00000003 7600cfb0 00000000 7600d1c4
[  140.334592] bfe0: a6c02f34 a484f958 a6befc94 a6f133bc 80000010 00000004 2effa861 2effac61
[  140.334651] [<b02954a8>] (memcpy) from [<b02a175c>] (__copy_from_user_memcpy+0xa8/0x1ec)
[  140.334700] [<b02a175c>] (__copy_from_user_memcpy) from [<b0333314>] (vchiq_copy_from_user+0x68/0x8c)
[  140.334750] [<b0333314>] (vchiq_copy_from_user) from [<b032a370>] (queue_message+0x538/0x968)
[  140.334794] [<b032a370>] (queue_message) from [<b032c860>] (vchiq_queue_message+0xf4/0x128)
[  140.334836] [<b032c860>] (vchiq_queue_message) from [<b03307a8>] (vchiq_ioctl+0x368/0x1824)
[  140.334881] [<b03307a8>] (vchiq_ioctl) from [<b0126b94>] (do_vfs_ioctl+0x42c/0x690)
[  140.334923] [<b0126b94>] (do_vfs_ioctl) from [<b0126e2c>] (SyS_ioctl+0x34/0x5c)
[  140.334969] [<b0126e2c>] (SyS_ioctl) from [<b000f560>] (ret_fast_syscall+0x0/0x3c)
[  140.335011] Code: ba000002 f5d1f03c f5d1f05c f5d1f07c (e8b151f8) 
[  140.335038] ---[ end trace f0d7b31389dfd2c7 ]---
[  140.335084] note: omxplayer[1324] exited with preempt_count 1
```

This is probably related to my kernel build having CONFIG_CPU_SW_DOMAIN_PAN=y.

fabled, 2016-03-28:
Confirmed, everything works with CONFIG_CPU_SW_DOMAIN_PAN is turned off. Please fix the drivers to work properly with this security feature enabled.

pelwell, 2016-03-28:
To save time, what is the minimal set of steps required to cause a crash? Does it vary with content?

fabled, 2016-03-28:
Just compile kernel with CONFIG_CPU_SW_DOMAIN_PAN=y. For me the above backtrace is reproduced when playing any mp4 video with omxplayer. Apparently the semantics how to copy data to/from userland are much stricter with domain pan is enabled.

notro, 2016-03-28:
CONFIG_CPU_SW_DOMAIN_PAN and CONFIG_UACCESS_WITH_MEMCPY are currently incompatible, see https://github.com/raspberrypi/linux/pull/1178#issuecomment-163651327

pelwell, 2016-03-28:
Thanks for the hint and link, @notro - that should make the task much easier.

pelwell, 2016-03-29:
I can confirm that I get a crash when CONFIG_CPU_SW_DOMAIN_PAN is set (booting to the desktop is all that is required), and that removing CONFIG_UACCESS_WITH_MEMCPY prevents the crash.

UACCESS_WITH_MEMCPY gives a boost to performance, so it isn't something we would want to disable by default. We could make UACCESS_WITH_MEMCPY dependent on !CPU_SW_DOMAIN_PAN, but that is a change to purely upstream code which we try to avoid. Given that you have to explicitly add CPU_SW_DOMAIN_PAN, explicitly remove UACCESS_WITH_MEMCPY at the same time.

@popcornmix Would you agree? The change to arch/arm/Kconfig is trivial.

popcornmix, 2016-03-29:
I'm okay with that. It really is an upstream issue so it would be nice if it were fixed there, but in the meantime we could make CPU_SW_DOMAIN_PAN remove UACCESS_WITH_MEMCPY to avoid others hitting this issue.

fabled, 2016-03-29:
Did not realize it's upstream issue. Yeah, maybe we should bug them again about this. The Kconfig issue sounds good enough workaround. Do you have any numbers on what kind of performance boost UACCESS_WITH_MEMCPY would give?

fabled, 2016-03-29:
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=c014953d84ec21a4df9a43be2378861ea6e9246e made it to 4.4 This is not a full fix? Or there's more problems with the compatibility?

popcornmix, 2016-03-29:
I seem to recall a network file transfer measurement that increased by about 10% with UACCESS_WITH_MEMCPY. This was probably Pi1 where CPU is more of a bottleneck. But there is a measurable improvement.

popcornmix, 2016-03-29:
@pelwell Does "Improve __copy_to_user and __copy_from_user performance" need anything from "ARM: fix uaccess_with_memcpy() with SW_DOMAIN_PAN" (I've not looked, but I wonder if the upstream fix needs to handle another code path).

pelwell, 2016-03-29:
"ARM: fix uaccess_with_memcpy() with SW_DOMAIN_PAN" (c014953d) only appears in rpi-4.4.y and above, so won't help 4.1. I can push the config fix there, or we can just leave it since 4.1 is obsolescent.

"Improve __copy_to_user and __copy_from_user performance" adds arm_copy_from_user, which also need to modified. I have a patch for this which allows omxplayer to work.

pelwell, 2016-03-29:
I've pushed patches for this to 4.1, 4.4 and 4.5. Please test, and close if happy.

fabled, 2016-03-30:
4.1 does not need to be patched. The domain pan features is new for 4.4.

popcornmix, 2016-04-04:
The 4.1 patch is harmless. @fabled okay to close?

closed 2016-04-11 by popcornmix