raspberrypi/linux #1383: error during make

maxyclu, 2016-03-29:
I try to compile a kernel with IMA enabled but fail after "make -j4 zImage modules dtbs" on my raspberry pi 2B. The error messages read as follows. Please help.

scripts/Makefile.modbuiltin:25: drivers/media/pci/dm1105/Makefile: No such file or directory
make[4]: *** No rule to make target 'drivers/media/pci/dm1105/Makefile'.  Stop.
scripts/Makefile.modbuiltin:54: recipe for target 'drivers/media/pci/dm1105' failed
make[3]: *** [drivers/media/pci/dm1105] Error 2
scripts/Makefile.modbuiltin:54: recipe for target 'drivers/media/pci' failed
make[2]: *** [drivers/media/pci] Error 2
scripts/Makefile.modbuiltin:54: recipe for target 'drivers/media' failed
make[1]: *** [drivers/media] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:1110: recipe for target 'drivers/modules.builtin' failed
make: *** [drivers/modules.builtin] Error 2
make: *** Waiting for unfinished jobs....

pelwell, 2016-03-29:
It looks like your tree is corrupt in some way - drivers/media/pci/dm1105/Makefile should exist in all kernel trees since 2012.

I suggest you follow the kernel building instructions [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md), and make sure that you have enough space on your device.

maxyclu, 2016-03-30:
In previous attempt, I download the zip and unzip it to compile. Apparently it doesn't work this way. I follow your suggestion and it's compiling now (though very slowly). Thank you very much.

closed 2016-03-30 by maxyclu

pelwell, 2016-03-30:
Make sure you include "-j4" in the make command to use all of the cores of the Pi2 - it's mentioned in a comment, but not in the example command line.

maxyclu, 2016-03-30:
Yes, i use -j4. Thanks.