raspberrypi/linux #1384: RPI-3 brmfmac driver - Kernel trace while setting up Ad-hoc mode

s13884, 2016-03-29:
I have been trying to setup the Ad-hoc mode using brmfmac but got following logs it crashes while configuring.

[ 161.652031] ------------[ cut here ]------------
[ 161.652121] WARNING: CPU: 3 PID: 97 at net/wireless/ibss.c:67 cfg80211_ibss_joined+0x17c/0x180 [cfg80211]()
[ 161.652130] Modules linked in: joydev brcmfmac evdev brcmutil cfg80211 rfkill snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpi omem bcm2835_rng uio_pdrv_genirq rpi_ft5406 rpi_backlight uio videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common vi deodev media i2c_dev fuse ipv6
[ 161.652243] **CPU: 3 PID: 97 Comm: kworker/3:2 Not tainted 4.1.20-v7+ #867**
[ 161.652252] Hardware name: BCM2709
[ 161.652300] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[ 161.652338] <800185e0>    (unwind_backtrace)  from  <80013f48>   (show_stack+0x20/0x24)
[ 161.652362] <80013f48>  (show_stack) from <80572ddc>  (dump_stack+0xd4/0x118)
[ 161.652384] <80572ddc>  (dump_stack) from <800271b4>  (warn_slowpath_common+0x98/0xc8)
[ 161.652403] <800271b4>  (warn_slowpath_common) from <800272a0>   (warn_slowpath_null+0x2c/0x34)
[ 161.652472] <800272a0>  (warn_slowpath_null) from <7f1b1f68> (cfg80211_ibss_joined+0x17c/0x180 [cfg80211])
[ 161.652575] <7f1b1f68>  (cfg80211_ibss_joined [cfg80211]) from <7f2f37dc> (brcmf_notify_connect_status+0x284/0x3bc [brcm fmac])
[ 161.652653] <7f2f37dc>  (brcmf_notify_connect_status [brcmfmac]) from <7f2faddc> (brcmf_fweh_call_event_handler+0x48/0xa 4 [brcmfmac])
[ 161.652735] <7f2faddc>  (brcmf_fweh_call_event_handler [brcmfmac]) from <7f2faea0> (brcmf_fweh_event_worker+0x68/0x2d0 [ brcmfmac])
[ 161.652788] <7f2faea0>  (brcmf_fweh_event_worker [brcmfmac]) from <8003e0b8> (process_one_work+0x154/0x458)
[ 161.652811] <8003e0b8>  (process_one_work) from <8003e410> (worker_thread+0x54/0x500)
[ 161.652831] <8003e410>  (worker_thread) from <80043bec> (kthread+0xec/0x104)
[ 161.652851] <80043bec>  (kthread) from <8000faf8>](ret_from_fork+0x14/0x3c)
[ 161.652863] ---[ end trace 5fd34bdc406f077e ]---

Unable to join the ibss. The power management is already off. So anyone ever noticed the same?

pelwell, 2016-03-29:
Can you give more information about what you did so we can try to reproduce this?

s13884, 2016-03-29:
First disabled the wlan0 from /etc/network/interfaces 
#allow-hotplug wlan0
#iface wlan0 inet manual
#   wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

Command to start the Ad-hoc Mode 
iwconfig wlan0 mode ad-hoc essid my-mesh ap 02:CA:FE:FE:FE:40 channel 11

pelwell, 2016-03-29:
Thanks for that. I can reproduce the crash on rpi-4.1.y.

What makes you think that this is supposed to work? I'm not sure that it is. You may be better off asking about this on the Raspberry Pi Forums, where people have had success using hostapd: https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=140997&p=940642

s13884, 2016-03-30:
Don't you think that is the different part than ad-hoc mode? I think, they are discussing about disable the SSID broadcasting in AP mode. I have already posted my issue on Raspberry Pi forum. https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=142306

I have tested ad-hoc mode on RPI2 using wipi. Please find below the same. 
**Command to start the ad-hoc mode**
sudo iwconfig wlan0 mode ad-hoc essid my-mesh ap 02:CA:FE:FE:FE:40 channel 3
sudo ifconfig wlan0 up

**Logs for above commands**
[  111.362520] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  111.363641] wlan0: Created IBSS using preconfigured BSSID 02:ca:fe:fe:fe:40
[  111.363654] wlan0: Creating new IBSS network, BSSID 02:ca:fe:fe:fe:40
[  111.401614] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

**iw dev wlan0 station dump**
connected stations on wlan0
Station XX:XX:XX:01:66:6a (on wlan0)
        inactive time:  90 ms
        rx bytes:       14139
        rx packets:     224
        tx bytes:       0
        tx packets:     0
        tx retries:     0
        tx failed:      0
        signal:         -59 dBm
        signal avg:     -58 dBm
        tx bitrate:     1.0 MBit/s
        rx bitrate:     1.0 MBit/s
        authorized:     yes
        authenticated:  yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
Station XX:XX:XX:ae:5a:1c (on wlan0)
        inactive time:  100 ms
        rx bytes:       16026
        rx packets:     254
        tx bytes:       0
        tx packets:     0
        tx retries:     0
        tx failed:      0
        signal:         -47 dBm
        signal avg:     -47 dBm
        tx bitrate:     1.0 MBit/s
        rx bitrate:     1.0 MBit/s
        authorized:     yes
        authenticated:  yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no

So I am just trying to do the same thing on RPI3 as **iw phy0 info** says it supports the ad-hoc IBSS mode.

s13884, 2016-04-06:
brcmfmac source has following callback for ibss mode. 
.join_ibss = brcmf_cfg80211_join_ibss,
.leave_ibss = brcmf_cfg80211_leave_ibss,

Not sure but some issue with channel as per	if (WARN_ON(!channel)) return; at net/wireless/ibss.c:67
Do you think issue with channel?

s13884, 2016-04-06:
@pelwell  did you get a time to reproduce the crash?

pelwell, 2016-04-06:
Nope - not yet.

s13884, 2016-04-26:
@pelwell  any suggestion to try and check?

pelwell, 2016-04-26:
It doesn't crash on 4.4, but it also doesn't work - instead you get:
```
[  319.461328] brcmfmac: brcmf_cfg80211_change_iface: iface validation failed: err=-16
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Device or resource busy.
```

Since we've made our last 4.1 release I'd rather work on the 4.4 tree. To update the kernel and firmware you can use:
```
sudo rpi-update
```

s13884, 2016-04-27:
Tried 4.4 
iw phy0 info says it supports IBSS mode but not in the list of "valid interface combination". It supports AP, Managed and p2p. Is there any chance to support IBSS?

pelwell, 2016-04-27:
I'm looking at that next.

pelwell, 2016-04-28:
I've been slightly diverted into working out why channels 12 & 13 don't work in the 4.6 kernel, but I'll get back to IBSS ASAP.

pelwell, 2016-04-29:
IBSS appears to be supported on the 4.5 and 4.6 kernels. I can create an Ad Hoc network, even though I can't do much with it; my Android phone won't connect, and my laptop connects but probably needs some additional networking configuration to be able to make use of it.

I can see the failure 4.4; I don't have a fix yet, but I'm getting closer.

pelwell, 2016-04-29:
I've just back-ported two commits to the 4.4 tree that seems to make it work as well as 4.5 for Ad Hoc mode. They will get picked up in the next build.

s13884, 2016-05-10:
IBSS mode works! Thanks
Cannot see the connected station list while running "iw dev wlan0 station dump"

mildmojo, 2016-12-02:
Did the fix make it into a released kernel? On attempting ad-hoc mode, I'm seeing a `brcmfmac` kernel trace on a Raspberry Pi B+ v1.2 with kernel 4.4.35 (latest available from running `rpi-update`). I'm using the [Redbear IoT HAT](https://github.com/redbear/IoT_pHAT) which uses the same driver as the RasPi 3's wifi chip.

```
$ uname -a
Linux raspberrypi 4.4.35+ #931 Sat Nov 26 13:00:50 GMT 2016 armv6l GNU/Linux
```

```
$ iwconfig wlan0 mode ad-hoc essid my-mesh ap 02:12:34:56:78:9A channel 1
$ dmesg
... <snip> ...
[   62.429425] ------------[ cut here ]------------
[   62.430644] WARNING: CPU: 0 PID: 61 at net/wireless/sme.c:850 cfg80211_roamed+0x88/0x98 [cfg80211]()
[   62.430862] Modules linked in: bnep hci_uart btbcm bluetooth joydev evdev hid_logitech_hidpp 8192cu hid_logitech_dj brcmfmac brcmutil snd_bcm2835 snd_pcm snd_timer cfg80211 snd rfkill bcm2835_gpiomem bcm2835_wdt uio_pdrv_genirq uio batman_adv ipv6
[   62.430995] CPU: 0 PID: 61 Comm: kworker/0:2 Not tainted 4.4.35+ #931
[   62.431012] Hardware name: BCM2708
[   62.431391] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[   62.431586] [<c0016cfc>] (unwind_backtrace) from [<c0013c20>] (show_stack+0x20/0x24)
[   62.431696] [<c0013c20>] (show_stack) from [<c02e43bc>] (dump_stack+0x20/0x28)
[   62.431743] [<c02e43bc>] (dump_stack) from [<c0021ef4>] (warn_slowpath_common+0x8c/0xc4)
[   62.431779] [<c0021ef4>] (warn_slowpath_common) from [<c0021fe8>] (warn_slowpath_null+0x2c/0x34)
[   62.432936] [<c0021fe8>] (warn_slowpath_null) from [<bf10fcfc>] (cfg80211_roamed+0x88/0x98 [cfg80211])
[   62.434429] [<bf10fcfc>] (cfg80211_roamed [cfg80211]) from [<bf2bbf38>] (brcmf_notify_roaming_status+0x134/0x1c0 [brcmfmac])
[   62.435089] [<bf2bbf38>] (brcmf_notify_roaming_status [brcmfmac]) from [<bf2c3b38>] (brcmf_fweh_call_event_handler+0x44/0x9c [brcmfmac])
[   62.435561] [<bf2c3b38>] (brcmf_fweh_call_event_handler [brcmfmac]) from [<bf2c3d10>] (brcmf_fweh_event_worker+0x180/0x358 [brcmfmac])
[   62.435917] [<bf2c3d10>] (brcmf_fweh_event_worker [brcmfmac]) from [<c00376a4>] (process_one_work+0x11c/0x39c)
[   62.436009] [<c00376a4>] (process_one_work) from [<c0037964>] (worker_thread+0x40/0x4d0)
[   62.436058] [<c0037964>] (worker_thread) from [<c003d2e8>] (kthread+0xdc/0xf8)
[   62.436106] [<c003d2e8>] (kthread) from [<c000f8a8>] (ret_from_fork+0x14/0x2c)
[   62.436127] ---[ end trace 8b5ff760734b4080 ]---
```

(The 8192cu module is loaded because wlan1 is a USB rtl8188cus chip; the trace reproduces with or without the second adapter connected.)

I can't reproduce it on a Raspberry Pi 2 B v1.1 with the same kernel version (4.4.35-v7+). It seems to work fine there.

s13884, 2016-12-02:
This is something different. I think, failed to get the bss information while roaming.

mildmojo, 2016-12-02:
Ah, thanks; I didn't catch the meaning of the difference in the stack traces. I've opened #1742 for this.

JamesH65, 2017-05-18:
Closing this issue as questions answered/resolved.

closed 2017-05-18 by JamesH65