raspberrypi/linux #1385: Kernel Panic w/ microchip enc28j60 ethernet driver w/ Patch

davarus, 2016-03-31:
There is a NULL pointer dereference kernel panic in the enc28j60 ethernet driver. I made a quick patch that prevents the kernel panic, but I'm not sure if there's something else that's off.  To reproduce the kernel panic is pretty easy:  Connect the pi ethernet adapter directly to another system.  (The one we used was a toughbook.)  If it's connected to a switch of any kind it works happy, happy.  Otherwise after enough traffic it will panic.  This patch solves that, but as I said, I'm not sure if it's the correct fix as I'm not familiar with the linux kernel really.
[enc28j60-panic.patch.txt](https://github.com/raspberrypi/linux/files/198321/enc28j60-panic.patch.txt)

popcornmix, 2016-03-31:
This is upstream code. Ideally if you could submit it here:
https://www.kernel.org/doc/Documentation/SubmittingPatches

the developers who really know this code can comment on the fix.
Once they have agreed on the correct fix, we can cherry-pick it.

davarus, 2016-04-01:
Thanks for getting back so quick, submitted patch upstream yesterday and will keep you informed.

AirRietveld, 2016-04-22:
Any updates on the submitting process? I've got the same kernel panic error here. It's kinda driving me insane :-) But it's a relief to see that it's not only me who has this.

hecko, 2016-04-24:
@davarus can you please link any info on the patch you have submitted to the mainstream kernel so we can track it independently? thanks!

davarus, 2016-04-26:
Sorry, I submitted the patch, but checking the list it didn't go through.  I'm assuming I have to subscribe to the list and resubmit.  I'll be doing that tomorrow and will post the info here.

davarus, 2016-05-04:
Sorry, got busy with some other stuff.  Here is a link to what made it to the list as of a few minutes ago... http://marc.info/?l=linux-netdev&m=146238430627231&w=2

rdicroce, 2016-06-16:
I think I just ran into this. I followed the mailing list thread, and it looks like a good patch was created, but AFAICT it was never applied. Am I missing something here?

hecko, 2016-06-16:
Yes, is was never applied as its really not a proper kernel patch - it
apparently fixes the issue, but was neve accepted do streamline kernel as
the real problem is somewhere else in the code.

Try to search net-dev korg mailing list for ENC28J60 - there were multiple
recent patches applied to the driver so some of the MIGHT fix the issue. I
would love to test this, however Im not able to replicate the panic :(

nlubello, 2016-06-16:
I've found somewhere a post explaining that the cause might be due to an
hardware bug
I solved it just changing the SPI clock to an higher one (I don't have the
config.txt file available now to check the correct value) an tested the
raspberry with a load simulator of a web page served with nginx.

Hope this can help!
Yes, is was never applied as its really not a proper kernel patch - it
apparently fixes the issue, but was neve accepted do streamline kernel as
the real problem is somewhere else in the code.

Try to search net-dev korg mailing list for ENC28J60 - there were multiple
recent patches applied to the driver so some of the MIGHT fix the issue. I
would love to test this, however Im not able to replicate the panic :(

https://github.com/

------------------------------------------------------
This message is not intended to have
contractual effect
------------------------------------------------------

rdicroce, 2016-08-15:
FYI for anyone still interested in this issue: I think a recent commit on the 4.4.y branch (also applied upstream) fixes this issue. See https://github.com/raspberrypi/linux/commit/7547e72d51b51267e6a1450c33e3ca9818f3abd7

Unfortunately I don't have any way of reliably reproducing the problem, so I can't say for sure if this fixes it.

lurch, 2016-08-30:
http://raspi.tv/2015/ethernet-on-pi-zero-how-to-put-an-ethernet-port-on-your-pi#comment-62101 confirms that the rpi-update kernel removes the kernel-panic from the enc28j60 driver, so this issue can probably be closed.

closed 2016-08-30 by popcornmix