raspberrypi/linux #1386: SPI no such device

criss2525, 2016-03-31:
I run : sudo modprobe spi-bcm2708

and I get this: modprobe: ERROR: could not insert 'spi_bcm2708': No such device

in the blacklist: sudo nano /etc/modprobe.d/raspi-blacklist.conf	
everything is set like needed

Any ideas?

pelwell, 2016-04-01:
Blacklists are so 2014 - it's all about Device Tree these days. If you want to enable the SPI device you need to add `dtparam=spi=on` to your config.txt. This will then automatically load the driver (which is now spi_bcm2835, not spi_bcm2708). If you'd rather load the driver by hand when you need it then you can still blacklist the driver to prevent it being loaded automatically - just remember to call it spi_bcm2708.

criss2525, 2016-04-01:
the thing is, I also did this and I still get this: NO SUCH DEVICE. ideas?

pelwell, 2016-04-01:
spi_bcm2708 won't load any more because the platform device it is looking for doesn't exist. Instead we are using spi_bcm2835.

What does `ls /dev/spi*` show? I think that SPI is already working for you, and that the instructions you are following are out of date (a constant problem with this rate of change).

criss2525, 2016-04-01:
thank you so much! you saved my life))) have a nice day)

closed 2016-04-01 by criss2525