raspberrypi/linux #1391: kernel 4.4 dtc issues

Noltari, 2016-04-04:
Hello,

I can't get past the boot splash screen on linux 4.4 (OpenWrt) unless I revert the following patches:
https://github.com/raspberrypi/linux/commit/fa95cd11c54c77d6151b7588b1f41457d8775c60
https://github.com/raspberrypi/linux/commit/8ed747466ab2fac41d870582b149385d3e8c2d43

Looks like the upstream dtc generates malformed dtbs or something...

Regards,
Álvaro.

pelwell, 2016-04-04:
Please unpack that sentence a bit:

1) How is the kernel  you are building different from what you would get following our standard build instructions?

2) What are your build products - kernel, dtbs, overlays etc.?

3) What is in config.txt and cmdline.txt?

4) Which firmware release are you using?

Noltari, 2016-04-04:
1) It's based on OpenWrt modified kernel with these patches:
https://github.com/openwrt-es/openwrt/tree/master/target/linux/generic/patches-4.4
https://github.com/openwrt-es/openwrt/tree/master/target/linux/brcm2708/patches-4.4
Those last patches are the brcm2708 specific ones that I extracted from rpi-4.4.y (my local patches are updated to 08532d242d7702ae0add95096aa49c5e96e066e2).
2) kernel, dtbs and overlays.
3) https://github.com/openwrt-es/openwrt/blob/master/target/linux/brcm2708/image/cmdline.txt
https://github.com/openwrt-es/openwrt/blob/master/target/linux/brcm2708/image/config.txt
4) https://github.com/raspberrypi/firmware/commit/046effa13ebc4cc7601df4f06f4834bd0eebb0f8

pelwell, 2016-04-04:
Which config do you build with?

If you upload the contents of the boot directory somewhere we can take a look.

Noltari, 2016-04-04:
https://github.com/openwrt-es/openwrt/blob/master/target/linux/brcm2708/bcm2710/config-4.4
+CONFIG_OF_CONFIGFS=y

Boot directory:
https://files.noltari.es/openwrt/rpi3/rpi3_boot.tgz

Noltari, 2016-04-05:
Take a look at the kernel image generation:
https://github.com/openwrt-es/openwrt/blob/master/target/linux/brcm2708/image/Makefile#L16

I assume "--dtok --270x" are automatically detected and maybe ddtk is missing, which conflicts with the new overlay configfs?
I don't know if it affects or not, but since I can't find any documentation at all...

Noltari, 2016-04-07:
I would appreciate any ideas with this issue, since I just don't get why reverting to the "dirty" version of dtc makes everything work again and with the upstream version it doesn't get past the boot splash screen...
I insist this is looks like a dtc issue since it's the only thing that I'm changing.
Did anyone actually test the 4.4 kernel with the .dtb generated by the rpi-4.4.y branch and not the ones provided by the firmware repo?

clivem, 2016-04-07:
> Did anyone actually test the 4.4 kernel with the .dtb generated by the rpi-4.4.y branch and not the ones provided by the firmware repo?

Yes, works for me.

pelwell, 2016-04-07:
I should be able to test your image today.

pelwell, 2016-04-07:
OK - I've got some results. Your main problem is the config.txt setting `device_tree_address=0x100`. This doesn't work at the moment because the new dtc configuration is generating larger output files because it is including fixup symbols (possibly not necessary - I'll check this later).

There is only a 16K window - 0x100 to 0x4000 - for the DTB; any longer and something will write over the end. I managed to extract some output - see before.txt [here](https://gist.github.com/pelwell/da91b7ec9fa9f2e34206929b66e878b2) (the ???s are where I had to manually patch over a dirty cache line). As you can see, DT isn't happy.

Current firmware loads the DTB at the end of memory unless you give it an explicit address, so if you comment out this line booting gets further - see after.txt  [here](https://gist.github.com/pelwell/da91b7ec9fa9f2e34206929b66e878b2). I got this output from the UART after loading the pi3-miniuart-bt overlay - pi3-disable-bt would also work.

With that change, the problem now seems to be related to interrupt controllers and IRQ allocation.

If you want to try the new dtc but without the fixup symbols, try this patch:
```
diff --git a/arch/arm/boot/dts/Makefile b/arch/arm/boot/dts/Makefile
index fdc450f..02db01a 100644
--- a/arch/arm/boot/dts/Makefile
+++ b/arch/arm/boot/dts/Makefile
@@ -813,5 +813,5 @@ clean-files := *.dtb
 
 # Enable fixups to support overlays on BCM2708 platforms
 ifeq ($(RPI_DT_OVERLAYS),y)
-       DTC_FLAGS ?= -@
+#      DTC_FLAGS ?= -@
 endif
```

Noltari, 2016-04-07:
Thanks, it was indeed a config issue.
After commenting out the following from config.txt it worked like a charm:
- disable_commandline_tags
- kernel_address
- device_tree_address

Regards,
Álvaro.

closed 2016-04-07 by Noltari