raspberrypi/linux #1399: Hardware UART name changed in Pi3

patricktokeeffe, 2016-04-07:
Code which used to be portable across RPi versions no longer is because the hardware UART was renamed (`ttyAMA0`-->`ttyS0`). To further complicate user's lives, the new Bluetooth serial port has the old UART's name (`ttyAMA0`).

What was wrong with the new Bluetooth serial port being `ttyS0` or even `ttyAMA1`? This device name swapping makes about as much sense to me as naming the new wifi `eth0` and making the wired interface `wlan0`. Given the RPi is intended for novices, it seems somewhat flippant to turn around and require knowledge of device tree overlays just to get a consistent experience across board versions.

*If I've made this report in the wrong place, please kindly direct me to the correct forum. Thanks*

pelwell, 2016-04-08:
Nothing has been renamed, and that is the problem. What is different about Pi3 is that by default the more capable UART (UART0 - a version of the ARM PL011, named by its driver ttyAMA0) is used to communicate with the Bluetooth modem. This UART is preferred because it has a stable clock (not linked to the VPU core clock) and larger FIFOs. The other UART, UART1, is a cut-down 8250 clone, with smaller FIFOs and a variable clock (which is why the `uart_enable=1` config.txt setting fixes the cpu_freq at 250MHz if ttyS0 is the primary, otherwise characters get garbled and lost as the baudrate fluctuates).

Linux does not AFAIK have a standard mechanism to name or rename devices. In the case of UARTs the driver contains a hard-coded stub (ttyAMA for UART0, ttyS for UART1) to which the TTY subsystem adds an instance number. This is a problem for us because we didn't want to break existing systems when upgrading to a Pi3. You could imagine making the TTY name a DT parameter, but that would be completely non-standard (I've already had to modify the PL011 driver to stop it being called ttyAMA1 on a Pi3), and it would make it harder to know which physical UART is being addressed.

We've tried to ameliorate the problems in three ways:
1) We've added aliases - serial0 and serial1 - where serial0 is the primary UART that one would run a console on, and serial1 is the auxilliary UART used for Bluetooth. These aliases are created as symbolic links in /dev, based on DT settings.
2) The firmware will re-write the kernel command line, replacing =serial0 and =serial1 with the appropriate real TTY name - =ttyAMA0 or =ttyS0. In addition, on a Pi3 =ttyAM0 is automatically converted to =ttyS0 if ttyAMA0 is being used for Bluetooth.
3) We've released two DT overlays - pi3-miniuart-bt and pi3-disable-bt - that switch the roles back, at the cost of reducing the BT performance.

These weren't decisions taken lightly - they were engineering choices made to give the best Bluetooth performance on the Pi3 hardware.

[ It wasn't really the best place - firmware issues or RPi Forums would have been better - but I'll let  you off. ]

patricktokeeffe, 2016-04-09:
Bummer, I thought I had started to understand devices and name assignments in Linux but this answer revises so much.. now my complaint sounds silly. Thank you for being informative in your response. 

4.1.19-v7+ #858

closed 2016-04-09 by patricktokeeffe