raspberrypi/linux #1401: Bluetooth HSP with pulseaudio, does not connect

adcpc, 2016-04-11:
Hi,

I've been trying to use Raspbian (2016-03-18) on a Raspberry 2 with a USB Bluetooth dongle to connect with a HSP headset. Since bluez 5.x does not natively support HSP/HFP, I'm using pulseaudio (built with native HSP support) to enable audio through HSP. 

The bluetooth part works well. I can discover and pair my device and even get audio through an A2DP stream.

However, when I switch to HSP (using pavucontrol), I get the following error : "backend-native.c: connect(): Function not implemented".

According to the feedback that I got from the pulseaudio mailing list, the "function not implemented" message is related to a missing support on the kernel : https://lists.freedesktop.org/archives/pulseaudio-discuss/2016-March/025813.html

Let me know if additional details are required.

adcpc, 2016-04-15:
I tried the same operation using a RPi 3 with its built-in bluetooth 4.1 controller. 

The result was different since I didn't have the "function not implemented" errror. In fact, I was able to connect the HSP profile. However, I couldn't get an audio through HSP, just some kind of low level cracking noise. A2DP was still working.

I also tried with a bunch of other bluetooth dongles : 
- Targus USB dongle : Broadcom Corp. BCM20702A0 Bluetooth 4.0 -> Same as built-in controller, can connect HSP but no audio.
- Maxxtro USB dongle : Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) -> No HSP connection ("function not implemented")
- PTS USB dongle : USB Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) -> HSP audio works ! Input and output. I can even switch the profile (HSP <-> A2DP) while playing a fil.

I'm a bit surprised that the built-in bluetooth controller is not working correctly or not fully supported. Any chance to have anyone from the dev team looking at this issue ?

JamesH65, 2016-04-15:
Might be worth checking out issues

#1360
#1402

jenhro, 2016-06-18:
same problem
bluetooth on Pi 3 can't works on HSP/HFP (A2DP is OK)
still trying solutions.....
well, Targus USB dongle seem to works when a firmware be added
[BCM20702A0](https://www.raspberrypi.org/forums/viewtopic.php?t=104579&p=725766)

denics, 2017-03-02:
@adcpc did you manage to use HSP with Raspbian Jessie? can you point me to some documentation please?

YOUN3SS, 2017-03-27:
This may help for HSP, not the ideal solution but still working.

http://youness.net/raspberry-pi/bluetooth-headset-raspberry-pi-3-ad2p-hsp

bmx666, 2017-10-01:
HSP works only with
- kernel v4.2+
- blueZ v5.43 only (+[patches for debian/ubuntu](https://packages.ubuntu.com/source/zesty/bluez)), v5.44 and higher are broken
- ofono v1.17 and higher (patches are not required)
- pulseaudio v10.0 (+[patches for ubuntu](https://anonscm.debian.org/git/pkg-pulseaudio/pulseaudio.git/tree/debian/patches?h=ubuntu/10.0-1ubuntu2))

YOUN3SS, 2017-10-04:
You need ofono for **HFP** only.

mayankc7h7, 2017-10-13:
@bmx666 "HSP works only with..."
you mean with inbuilt bluetooth? Can you please detail?

pelwell, 2017-10-13:
@bmx666 is giving a long list of requirements if you want to use headset profile. The version of BlueZ that comes with Jessie and Stretch does not support it - don't ask me why, it seems a ridiculous decision on the part of the BlueZ devs.

bluetiger9, 2018-01-21:
Solution to use the **HSP profile** with **Raspberry Pi 3**’s built in **BCM43438** chip:
https://github.com/raspberrypi/linux/issues/2229#issuecomment-359233111

YOUN3SS, 2018-02-08:
As said @bluetiger9 there is a solution for that, check above link.

closed 2018-08-31 by JamesH65