raspberrypi/linux #1402: Pi3 bluetooth audio stutters with Wifi enabled

SebTC, 2016-04-11:
Hello,

I'm testing with streaming music using a2dp over bluetooth on the Pi3. When Wifi is enabled, I get constant buffer underruns wit Pulseaudio (Blueman shows a downstream of around 34kB/s). As soon as I disable the Wifi interface (ifdown wlan0), the audio starts to play normally and the downstream is around 42kB/s (which is correct high quality stereo audio if I see http://soundexpert.org/news/-/blogs/bluetooth-audio-quality-a2dp). 
Also tried making the buffer much larger, changing resampling type, realtime-scheduling etc. Also tried latest Pulseaudio, no difference. Seems like it is a Raspberry problem.

First thought it was because the Wifi and Bluetooth both use UART, but the is not true (and would be too slow if Wifi was over the 921600 baud if I see it correctly). They still share the same chip (BCM43438). Is there a known reason why I (and also heard others) have this problem?

pbros, 2016-04-17:
I've been having the exact same issue. Disabling WLAN0 fixed the audio issues. However, I would very much like to be able to use the wifi...

REY-01, 2016-04-19:
Same is here.  Took me 3 days, 2 distros to figure out it was the build in WiFi. Same error appears when i use a WiFi stick direktly at the USB Ports. When i use a USB connection cable with the USB stick everything works fine. So i simply think it comes from the build in antenna that the two 2.4 GHz services interfer each other.  :-/

pbros, 2016-04-19:
I was able to get A2DP to work by disabling the on-board wifi and using a Wi-Pi USB adapter, without any extension cable.

Ferroin, 2016-04-19:
This brings up a rather interesting question:  Does the on-board WiFi chip support Bluetooth co-existence, does the driver support this, and does it work correctly?  Based on what I've seen from multiple sources, there's considerably better latency when you either disable the on-board WiFi or disable the on-board Bluetooth and use a USB adapter instead, and that sounds to me like either the on-board chip doesn't correctly implement BT co-existence, or the driver doesn't properly support it.

pelwell, 2016-04-19:
The BCM43438 has a co-existence interface between the WiFi and Bluetooth interfaces - no software support is required.

WayneKeenan, 2016-04-19:
@Ferroin From my experience I would say /fundamentally/ yes it does, although I'm not an authoritative source and I do not demand much on the Bluetooth side....    Whilst developing Bluetooth LE central and peripheral applications on a Pi 3  I run a VNC X session, 2x SSH sessions and have NFS share mounted all via WiFi and all are ok.

rschulman, 2016-10-07:
+1 on this, as I've just discovered tonight. Took wlan0 down and the audio played just fine. Has anyone gotten any new word since August on what's going on here and if there's a fix?

adn77, 2016-11-14:
+1 me too, "ifdown wlan0" and pulseaudio streams fine via a2dp

SchorschM, 2016-11-19:
+1, anything new on this?

ladygrey, 2016-12-12:
+1, just updated today, using an Anker Sound Core bluetooth speaker. Plays beautifully if I turn off wifi, but that's a pretty big workaround. It's annoying but workable for this project (<passiveaggressive>OKAY FINE, I'll connect via hdmi instead of vncserver I GUESS </passiveaggressive>) but I too am waiting on a fix because it seriously limits my ability to make my projects mobile. VNCserver is a must.

vgallissot, 2017-01-07:
+1 that gave me headaches finding this problem out !  

I needed WiFi so I just:  
1) Use a USB dongle as WiFi adapter
2) Disable onboard WiFi adapter in /etc/network/interfaces

No more sound problems.

daryltucker, 2017-01-12:
I'm excited to see any progress on this, but as a reminder, you can Subscribe to this thread and add a reaction to the original post.  It is **not** recommended to post a response of +1.

jeffleus, 2017-01-24:
Agreed that no WiFi is crippling to the base Pi3.  Adding a usb dongle defeats one of the big gains w/ the Pi3 of onboard WiFi/BT.  :-(

mohitkalraniit, 2017-01-24:
I have also tested the behavior and facing same issue as reqorted here. Planning to add additional USB WiFi adapter  to overcome the problem. Hope pi would support second WiFi without much problems.

hurda, 2017-02-28:
I guess the Zero W will suffer from the same issues regarding Bluetooth and WLAN, as its using the same chips?
Using USB-devices as workarounds ain't that easy with the Zero W, though.

BaReinhard, 2017-03-03:
Is this happening to every Raspberry Pi? How is the music being played? (Pi hat DACs, sound cards, BCM?)

abatardi, 2017-03-03:
Only an issue when both are going. WiFi actively transmitting and then try to use Bluetooth.  Bluetooth + LAN is no problem.  So most people and applications won't see the issue.

kalramohit, 2017-03-03:
I have added secondary WiFi receiver and made it primary and using embedded WiFi as bluetooth receiver. This is a cheapest way to get this working.

hurda, 2017-03-03:
> Bluetooth + LAN is no problem.

Please show me the LAN-port on the Pi0W.

pradeepprakash, 2017-03-06:
Onboard Bluetooth + Wi-Fi  is still a problem. 

+1

BaReinhard, 2017-03-06:
Has anyone tried renicing pulseaudio to have a higher priority?

pradeepprakash, 2017-03-07:
Yes, I tried with with higher priority with no discernible difference in the outcome.

pradeepprakash, 2017-03-07:
Hi,
  Can you please let me know a workable-configuration if you have one for
the above problem i.e  Wifi - needed, Bluetooth Speaker pairing in A2DP
mode.
From your profile, It looks like you have extensively played around in that
area.

Thanks.

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Cheers,
Pradeep
http://pradeepclicks.com/

Michiman, 2017-03-07:
I am also trying to solve this problem.  The choppiness seems to change a bit among different BT speakers/headphones, but it's still there using a WiFi dongle and disabling the onboard WiFi.  Even using a BT dongle, the choppiness is still there while playing a local mp3 or using Pithos (Pandora).  I also used a lower bitrate mp3 file, and the choppiness improved.

mwestphal, 2017-03-08:
I experience the same issue.

Michiman, 2017-03-09:
One thing you have to make sure is that you turn off bluetooth scanning (scan off) while in the bluetoothctl prompt.  That resolved my issue and I was able to stream nicely with a Pi Zero W, Pi3, using the built-in wifi/BT and Pi Zero + redbear IoT PiHat.

mwestphal, 2017-03-09:
@Michiman : I'm 100% sure that i tried it without scanning on at the same time. still had the issue.

rvoogd, 2017-04-04:
+1 
same here it's definitely the combination of onboard wifi + bluetooth.
Setup: pi zero w + phat dac

Onboard bluetooth+wifi enable -> audio stutters very badly
onboard wifi disabled -> audio plays perfectly without stuttering

I guess i need to start investigating how this all works on a low level- which forms a nice challenge to learn this properly

maklotski, 2017-04-17:
I also had terrible sound issues trying to stream audio using a2dp tutorials based on pulseaudio.
I tried suggestions of tuning buffer sizes and disabling internal WLAN.
Sound quality improved greatly, but still not to the point that I would use this as a real listening device - at best I would get a pop or stutter every few seconds.

I found another github project that overcomes the issue by avoiding pulseaudio entirely:
https://github.com/lukasjapan/bt-speaker
After disabling the internal WLAN, audio is quite reasonable using that method, and no need to login at boot (I have it running in the background of my retropie image).

StudentSA, 2017-04-18:
@maklotski , We have already established that the issue is caused when both Wi-Fi and Bluetooth are on at the same time. Yes if you do not need WiFi then the solution is to disable this. However I need to stream audio from online so WiFi is critical. I'm surprised that RPF has not released any useful information on this issue to date.

pelwell, 2017-04-18:
We have released all the useful information we have, i.e. not very much. Cypress (was Broadcom) have two parallel driver stacks - dhd and brcmfmac. They are supposedly close to finishing an updated dhd driver which improves coexistence, but a) it is still being tested and b) we use brcmfmac. As soon as there is an improved brcmfmac driver we will push it out.

vimagick, 2017-04-27:
+1 same here

lnorton89, 2017-05-07:
+1 same

JamesH65, 2017-05-18:
Simply adding `+1` to this issue is of no use. It just makes the comments list longer for no reason.  As soon as we have any information it will be posted.

mattnthat, 2017-05-18:
+1 to continue putting this on the radar and hopefully escalating priority
for a fix

P33M, 2017-05-18:
This github thread will be updated when information relevant to the issue is available. We are somewhat dependent on Broadcom (now Cypress) providing driver updates as the coexistence support on-chip is a function of the chip's firmware or firmware setup.

Degrading the signal-to-noise ratio of the thread with spammy responses is just irritating. Comments that do not offer anything to the discussion surrounding investigation or resolution of the issue are likely to be summarily deleted.

spalthammer, 2017-05-25:
I wrote a little script to use inotify to switch on and off wlan0 if bluetooth connects/disconnect. Ok it is
a workaround but i can live with it.

`#!/bin/bash

while true
do
        RES=`inotifywait -q -e CREATE,DELETE /dev/input/`
        case "$RES" in
                "/dev/input/ DELETE event1")
                        ifconfig wlan0 up
                ;;
                "/dev/input/ CREATE event1")
                        ifconfig wlan0 down
                ;;
        esac
done &
`

StudentSA, 2017-05-31:
So here's the work(alltheway)around I'd like to share at the expense of being laughed at. 
Run `pacat /dev/zero` in the background
Now play some audio and after the crackling stops +-30 seconds then play some more audio and enjoy the clear playback until you stop pacat.
If you worried about all the zeros flying over bluetooth then maybe consider installing "pv" with:
`sudo apt-get install pv`
Run the following in the background instead `cat /dev/zero | pv -qL 2k | pacat` to limit the zeros to a certain bitrate.
Would like to know how this works for you.

BaReinhard, 2017-05-31:
I don't use the project all that much anymore do to the nature of my home life for the time being but when I do have a client that uses this project on my recommendation and has been experiencing bad studder on android devices. I'll try to implement this fix at some point this week and see if this solves any of the issues and let you know. Thank you for the due diligence on this issue, as its alluded me for many months.

blacktigersoftware, 2017-05-31:
Interesting, all.  I've been working on a headless Pi Zero/W -- No X11.  And I can have two/three ssh shells going over wifi, and Bluetooth is as clean as it can be.  I did notice that excessive polling of the Bluetooth device, (ie getting Bluetooth info) causes stutter.  Have you tried just booting to cli?

blacktigersoftware, 2017-05-31:
Well, I just realized that the following comment wasn't really helpful without context. Sorry, been banging away at the keyboard all night ----

1 - Pi Zero/W and Pi 3 identical in terms of Bluetooth/Wifi, at least as far as the kernel is concerned.
2 - Running Jessie Lite - recently updated,  and kernel 4.9.29+
3 - Running NetBeans on desktop and remote Debugging on Pi.
4 - Stress testing Frame rates with a TFT display --- really cranking that SPI bus.
5 - Polling input events for touch screen and dumping results to stderr, which gets piped to NetBeans -- testing jitter on touchscreen
6 - Running mpg123_to_out123 example program from mpg123 tarball over Bluetooth playing Billy Joel's "An Innocent Man" from SD card.
7 - No X11 in sight.

Running as smooth as a pie, raspberry flavored.   Been doing this soo long I hum Billy Joel in my sleep.
Have noticed that that forcing a query for Bluetooth connection status made things bad.  

Suggest eliminating as much "other" code as possible.

thmang82, 2017-06-04:
Hi,
there is definitely a serious problem with the PI (Zero W) Bluetooth.

I moved a python script that detects phones via bluetooth from a C.H.I.P to a Pi Zero W.
The result was crazy, it jammed my whole Home Wifi when Bluetooth was accessed :-(

The script uses the following command to detect if a phone is in range:
      result = bluetooth.lookup_name(mac, timeout=5)

I run this in a loop with two phones. The loop starts every 15 seconds and tests both phones.
I first notified that a) SSH over Wifi was unresponsive sometimes and b) my WiFi LED Light were not responding sometimes after setting up the Pi Zero W.
Strange, so i tried to ping the Wifi Lights, result: Timeouts for about 5 Seconds every 15 seconds.
Then I tried to ping the PI Zero W: Ping times of about 2000-4000ms during those 5 second windows, sometimes also even timeouts.

So I disabled the script running the bluetooth detection: Anything was fine. 
Re-starting the script: The timeouts occurred again.

This is crazy! The bluetooth scanning for the phones (It's basically a "are you there?" to a paired bluetooth device) basically breaks my whole Home Wifi.
I know that Bluetooth and Wifi are on the same frequency. But Bluetooth is standardized to use extensive frequency hopping to prevent just such interference. Not so on the Pi Zero W?

It's definitely reproducible, just try the python script below.

My best guess about the reason: The bluetooth radio disturbs the Wifi, not the other way around. The reason could be a problem in the bluetooth stack regarding frequency hopping. This would also explain the reported bluetooth audio problems: when bluetooth does remain on one frequency, the wifi is more likely to disturb it's signal.
However, I might be wrong, I know WiFi pretty well as I did my PhD on a topic dealing with the Phy Layer of Wiifi, but I am no expert on the Bluetooth Phy.
__________________________________________________________________________
A short python test script that reproduces the problem. Just ping the Pi while running it.

import time
import bluetooth
mac = "00:00:00:00:00:00"
while True:
    print("Search Bluetooth for %s ..." % mac)
    try:
        result = bluetooth.lookup_name(mac, timeout=5)
    except bluetooth.btcommon.BluetoothError as e:
        print("\nERORR: Bluetooth request failed, error: %s" % e)
    print("Result:  %s is: %s" % (mac, result))
    time.sleep(15)

blacktigersoftware, 2017-06-05:
Tomorrow, (Monday evening EST), if you want, I'll post a youtube showing how well it works.  However, just double/triple confirming (just 5 minutes ago) ---- the only problems occur during "Discoverable" and "Scanning".  If you make your device non-discoverable and not actively scanning (discovering) WiFi and Bluetooth work just fine together on a Pi Zero W. I get s steady 4-5 ms ping over WiFi while connected via Bluetooth and ssh.   Need to figure out how to record sound for a youtube video, but I can plainly hear it jitter free.

FWIW - I'm working on a Bluetooth Audio application, so this really does concern me.  In my app, I was polling the connected device's info to get RSSI, etc.  I had to take the polling out, because of the problems already noticed by soo many people here.

Unless you are in control of all the apps in your session that might do a poll (D-Bus) on the Bluetooth connection, you can't rule them out as being complicit in the problems.  I'm not running X11 - therefore I'm much closer to the hardware and what happens.  Granted PulseAudio is still a "blackbox", but aside from that, I'm basically in control of the whole deal and it does work quite well.

Now -- I'm not saying that the firmware doesn't have problems, but in reality, the apps should be better behaved.

AudricManaud, 2017-06-06:
Hey,
I'd really interested in a youtube video if you have time :)
I'm also using a Pi Zero W, but when I disable the Wifi it sill stutters a little...

Nabla128k, 2017-09-04:
Hi, just a note - my Zero W suffers from the same problem - skipping BT audio when streaming by wifi - even for 9.1/Stretch release of Raspbian

pelwell, 2017-09-04:
Cypress are hoping to improve the "coexistence" between WiFi and BT, but they've been focusing on some WiFi stability issues first.

pawKer, 2017-11-07:
Hi, any updates on this?

pelwell, 2017-11-07:
Starting with the most recent Raspbian Stretch image, run:
```
sudo apt-get update
sudo apt-get install bluez bluez-firmware
```
This will pull in a new Bluetooth firmware and an updated BlueZ that, together, should improve the WiFi and Bluetooth coexistence.

pelwell, 2017-11-07:
While you are at it, get the latest kernel for improved Bluetooth reliability:
```
sudo apt-get install raspberrypi-bootloader raspberrypi-kernel
```

Michiman, 2017-11-07:
I'd love to see a side by side video of the performance of BT/WiFi
together. If someone hasn't made one, I'll have to work on it.

pitpit, 2017-11-08:
@pelwell I upgraded bluez bluez-firmware raspberrypi-bootloader raspberrypi-kernel to the last version, as advised by you.

However I'm still facing issue with sound streamed over bluetooth to the raspberry zero W when wifi is up. If I shutdown wifi (`sudo iwconfig wlan0 txpower off`), it works well and no more crackling happens.

Please let me know if I can be of any help.

I'm using bt-speaker. Related issue reported here :  https://github.com/lukasjapan/bt-speaker/issues/4

pelwell, 2017-11-08:
Are you saying you saw no improvement?

pitpit, 2017-11-08:
unfortunately, no improvement :(

MrSirSr, 2017-12-09:
Is anyone having this same type of issue with PS3 controllers using Retropie?  I have what seems to be random interference where sometimes the controllers work fine and sometimes it's as though I didn't press anything at all.  Makes it a bit difficult to play games that way!

ronkupersonos, 2018-01-22:
Today I updated a Pi Zero W to all the latest and can confirm the problem still exists.
`pi@raspberrypi:~ $ dpkg -l | grep -i bluetooth
ii  bluealsa                              0.6                                  armhf        Bluetooth ALSA Audio backend
ii  bluez                                 5.43-2+rpt2+deb9u2                   armhf        Bluetooth tools and daemons
ii  bluez-firmware                        1.2-3+rpt2                           all          Firmware for Bluetooth devices
ii  libbluetooth3:armhf                   5.43-2+rpt2+deb9u2                   armhf        Library to use the BlueZ Linux Bluetooth stack
ii  lxplug-bluetooth                      0.4                                  armhf        Bluetooth plugin for lxpanel
ii  pi-bluetooth                          0.1.6                                armhf        Raspberry Pi 3 bluetooth
ii  pulseaudio-module-bluetooth           10.0-1+deb9u1                        armhf        Bluetooth module for PulseAudio sound server`

nicokaiser, 2018-01-24:
The BCM43438 seems to have problems with multiple connections, either with BT+WiFi or with two BT connections:

When switching off WiFi (`ifconfig wlan0 down` or `dtparam=pi3-disable-wifi`) Bluetooth A2DP audio works quite fine. However when two devices are connected, the audio begins to stutter badly.

With an external USB Bluetooth adapter, multiple devices can connect via A2DP and stream audio, event simultaneously.

So I assume, this is a limitation of the chip, not a software thing... (But I'd love to be proven wrong in a future kernel update)

pelwell, 2018-01-24:
Make sure you are running with the latest BT firmware (`sudo apt-get update; sudo apt-get install bluez-firmware`)  - there have been some improvements.

ronkupersonos, 2018-01-24:
I last did that 2 days ago, has it changed since then?

-Ron

pelwell, 2018-01-24:
No - that will be the latest (1.2-3+rpt1).

ronkupersonos, 2018-01-24:
Thanks! In the meantime I bought a USB wifi dongle as a workaround.

Does anyone happen to know if the chipset driver is supposed to (in theory) take steps to avoid RF interference between these 2 radios?

-Ron

[87 more replies]