raspberrypi/linux #1404: gl driver fails in 4.4.y

ric96, 2016-04-14:
4.4.y shows blank screen when gl driver is enabled
uart output: http://paste.ubuntu.com/15826837/

probably has to do with: [    0.000000] cma: Failed to reserve 256 MiB

falling back to 4.1.y dea25fa62132365c11087e51e416df656db28bf3 works.

Further investigation shows that 4.1.y has this as output
[    0.000000] cma: Reserved 256 MiB at 0x20000000
but will show [    0.000000] cma: Failed to reserve 256 MiB when i manually add gpu_mem=256 to config.txt and blank screen.

joaodriessen, 2016-04-16:
I think a fix might be on the way, not at all able to decipher what all this is, but sounds like it's in the same ballpark:

PATCH v2 2/2] clk: bcm2835: Fix PLL poweron

In poweroff, we set the reset bit and the power down bit, but only
managed to unset the reset bit for poweron.  This meant that if HDMI
did -EPROBE_DEFER after it had grabbed its clocks, we'd power down the
PLLH (that had been on at boot time) and never recover.

anholt, 2016-04-20:
Not sure what's going on here.  I had to move the CMA region in the overlay down to 256@256.

ric96, 2016-04-23:
works fine on BRANCH=next
4.4.6

popcornmix, 2016-04-23:
@ric96 have you tried latest master branch? That includes an update to gl driver, including the change to CMA.

ric96, 2016-04-23:
@popcornmix yes master branch (4.4.8) works as well.

closed 2016-04-23 by popcornmix

anyc, 2016-04-27:
Is there something I could try if I get this error with a self-built 4.4.8 on a RPi3?
```
dmesg | grep -i cma
[    0.000000] cma: Failed to reserve 256 MiB
[    0.000000] Kernel command line: 8250.nr_uarts=1 cma=256M@256M dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1600 bcm2708_fb.fbheight=900 bcm2709.boardrev=0xa02
082 bcm2709.serial=0xd0e05121 smsc95xx.macaddr=B8:27:EB:E0:51:21 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=
0x3f000000  console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait root=/dev/mmcblk0p2 rw rootwait rootfstyp
e=btrfs console=ttyAMA0,115200 console=tty1 selinux=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 elevator=noop quiet splash logo.nolog
o
[    0.000000] Memory: 945820K/966656K available (7456K kernel code, 462K rwdata, 1996K rodata, 504K init, 780K bss, 20836K reserved, 0K cma-reserved)
[    4.290217] vc-cma: Videocore CMA driver
[    4.290229] vc-cma: vc_cma_base      = 0x00000000
[    4.290238] vc-cma: vc_cma_size      = 0x00000000 (0 MiB)
[    4.290247] vc-cma: vc_cma_initial   = 0x00000000 (0 MiB)
```
I am also experimenting with Uboot and I set the device_tree_address to a fixed location which worked so far. My config.txt:
```
cat /boot/config.txt  | grep -v "^#\|^$"
dtoverlay=vc4-kms-v3d
avoid_warnings=2
device_tree_address=0x17fdd200
dtoverlay=pi3-disable-bt
kernel=uboot.bin
```

popcornmix, 2016-04-27:
Why are you specifying the device_tree_address explicitly?
That conflicts with cma=256M@256M
Just removing that line would probably fix it.

anyc, 2016-04-27:
As far as I know, U-Boot expects the DT at a fixed location and does not recognize the dynamic address that is passed from the firmware. With the old 4.1 kernels, I used the default 0x100 but that didn't work with recent kernels/firmware anymore. So I took one of the dynamic addresses (minus a small number) and set it as fixed location.

It works now indeed if I put it back to 0x100 but I have to check under which circumstances it failed at this address back then (I mentioned this here: https://github.com/raspberrypi/linux/issues/1414#issuecomment-211355704). I guess I have to check if I can fix uboot myself to make this setup reliable.

pelwell, 2016-04-27:
It would have failed with a 4.4 DT at 0x100 because the DTBs grew significantly (the version of dtc was including some unnecessary relocation information). I've patched the 4.4 dtc command so that the size is pretty much as it was before, and the latest firmware will also strip out the unnecessary cruft before moving the DTB into place, so either way with the latest rpi-update kernel+firmware 0x100 should be viable again. I've also confirmed that device_tree_end functions as it should, so with u-boot you can probably set device_tree_end=0x8000 and double your headroom.

anyc, 2016-04-27:
Ah, great. Thank you!