raspberrypi/linux #1409: 4.4.y issues

as-qs, 2016-04-15:
I've run into a couple of headaches with this latest release. 

**1st Issue:**
Disabling the Device Tree kills the image since RPI refuses to boot up.

1) If having DT enabled is a requirement, there shouldn't be an option for disabling it.
2) People who already have the DT disabled, shouldn't be allowed to update to 4.4.y or at the very least the DT should be enabled as part of the update process.

**2nd Issue:**
My custom board that worked with spi-bcm2708 does not work with spi-bcm2835. Any ideas on what I need to do to get SPI working with that board again?

More details [here](https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=143198&p=953736#p953736).

lategoodbye, 2016-04-16:
Hi @as-qs,
even in the mentioned thread there are too little information:

1. What are the SPI slave(s) on the custom board? Microcontroller with custom SPI protocol?
2. What is the SPI protocol driver you are using?
3. What do you mean with SPI does not work (probing failed, no SPI clock, MISO doesn't contain useful data)?
4. Do you have an overlay / devicetree snippet for your custom board?

pelwell, 2016-04-18:
1) The Raspberry Pi 4.4 kernel tree isn't the only kernel that can be loaded on a Pi. Some people use U-boot as the "kernel", then load their own kernel by other means. That is why it is still possible to disable device tree, even though our standard kernel won't boot without it.
2) rpi-update doesn't examine or modify config.txt. It is a tool that comes with a number of warnings, extra in this case because the step to 4.4 is a major one. Perhaps the message could have been more explicit about the implications for Device Tree, but it really shouldn't come as a surprise that our kernels no longer support non-DT usage.

As @lategoodbye says, you should really give more information about your setup and the problems you encounter. If it turns out there this is some feature of the spi-bcm2835 driver that is causing the problem then we would consider making the older spi-bcm2708 driver selecable via an overlay until the upstream driver is fixed/enhanced.

as-qs, 2016-06-09:
First of all, I apologize for the lack of details and the late reply. I am somewhat out of my comfort zone as I'm primarily a software developer with very basic hardware knowledge, but I am very committed to finding the "correct" solution to my problem. If I'm leaving anything out, please point it out and I'll do my best to find the right answers. 

Here's a quick update with where I am and my assumptions. After reverting the Kernal to 4.1, I was able to keep the system operating by using the spi-bcm2708 overlay. However, I didn't want to be stuck with 4.1 & Wheezy, so I downloaded a fresh copy of Jessie and burned it on my SD card. Three assumptions (please correct me if I'm wrong):

1) I can still revert to the 4.1 Kernel if everything else fails.
2) Theoretically, there should be some kind of overlay or setting that would get this problem resolved.
3) The primary reason spi-bcm2708 was used was because it had Direct Memory Access (DMA) capabilities that are now native to spi-bcm2835. I have a very vague understanding of this.

With that out of the way, here is more information about how the two boards communicate over SPI. RPI acts as the master and connects to the custom board using pi4j libraries (the java implementation of Gordon's libraries). Now, as far as I know Gordon's libraries are updated to work with RPI3 and shouldn't lack any features; however, a solid build hasn't been released for pi4j. I am using their latest libraries that should theoretically work with SPI on RPI3 & 4.4 kernel, but I am completely clueless as to what the real issue is or how to isolate the problem.

From my java software, I connect to the custom board on SPI channel 1, mode 0, with the SPI speed of 390000. My understanding is that pins 19, 21, 23, and 24 are used for MOSI, MISO, SCLK, and CE0 respectively. The custom board uses a PIC18F45K80 chip with pins 24, 43, 42, 37 for SPI Select, MOSI, MISO, and SPI clock respectively. 

**Symptoms:**
I am able to "connect" to SPi using "wiringPiSPISetupMode()" function of the pi4j liberary, and I'm able to send data to it, but the data doesn't seem to make it to the custom board & nothing gets back. Unfortunately, I don't really have an oscilloscope or the hardware knowledge to provide any more details on that. Are there any sort of software tools that could listen for these SPI communications and shed some light on what's going on?

I've attached the data dumps/logs generated with the following commands:

dtc -I fs /proc/device-tree
sudo vcdbg log msg

[OverlayLoading.txt](https://github.com/raspberrypi/linux/files/307505/OverlayLoading.txt)
[DT_Data.txt](https://github.com/raspberrypi/linux/files/307506/DT_Data.txt)

Thanks,

lategoodbye, 2016-06-09:
Hi @as-qs,
is this your software stack?

    Java application
    Pi4J
    WiringPi
    /dev/spidev
    Kernel

Did you verify that spidev is available?

as-qs, 2016-06-09:
I am not sure about anything after WiringPi, but the rest of the stack looks about right.
How would I verify that? and what exactly is spidev? I'm gonna google it right now.

Ferroin, 2016-06-09:
/dev/spidev is the Linux character device interface to the SPI subsystem.  It provides a traditional UNIX style abstraction of the controller details so that apps don't have to include drivers for the SPI controller.

as-qs, 2016-06-09:
Alright, so I did the loop test from [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting): 

and got this:

```
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

FF FF FF FF FF FF
FF FF FF FF FF FF
FF FF FF FF FF FF
FF FF FF FF FF FF
FF FF FF FF FF FF
FF FF FF FF FF FF
FF FF

```

I even changed the max speed in the C file to 390000 and got the same results.

as-qs, 2016-06-09:
I disconnected the custom board from RPI and did the same test and got the following results:

```
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

00 00 00 00 00 00
00 00 00 00 00 00
00 00 00 00 00 00
00 00 00 00 00 00
00 00 00 00 00 00
00 00 00 00 00 00
00 00
```

lategoodbye, 2016-06-09:
According to your last output i can't see any issue with the kernel / spi driver. It seems to me userspace related.

as-qs, 2016-06-09:
I'm assuming you mean:
```
Java application
Pi4J
WiringPi
```

Could it be the CS (select) line? Because apparently, that test doesn't take the cs line into account. Is there anything I could do to test the cs line?

as-qs, 2016-06-14:
Looks like the problems I'm dealing with are related to an input pin I'm using rather than SPI.

Are there any reasons for not being able to use pin 7 (GPIO 7) as input? I don't seem to get a callback on the gpio listener I'm using on that pin. Do I need to do anything to disable GPCLK0?

pelwell, 2016-06-14:
Do you mean physical header pin 7 or GPIO 7? Header pin 7 is GPIO4 which can also carry GPCLK0. GPIO7 (on pin 26) is also CE1 for SPI0.

On a standard Raspbian installation GPCLK0 is not used, making it handy for other projects. I think it would help to get a better understanding of the GPIO usage in your system. Straight after booting run `raspi-gpio get > before.txt`, then start your application. If you have time before the app exits try `raspi-gpio get > during.txt', but if not just allow it to complete and run `raspi-gpio get > after.txt`. Now you can use `diff before.txt after.txt` etc. to see what has changed. You will probably see some pin level changes, but they can be ignored - it's the function changes that matter.

as-qs, 2016-06-14:
pi4j calls it pin 7: http://pi4j.com/pins/model-2b-rev1.html

Seems like I'm able to set the pin as input and use it without my board attached to it using this command with "pi4j-1.1-SNAPSHOT":

sudo pi4j /opt/pi4j/examples/GpioListenExample.java --pin 7 --pull down

as-qs, 2016-06-14:
Here is the result of what you asked for:

```
diff before.txt after.txt
19c19
<   GPIO 17: level=0 fsel=0 alt=  func=INPUT
---
>   GPIO 17: level=0 fsel=1 alt=  func=OUTPUT
24,25c24,25
<   GPIO 22: level=0 fsel=0 alt=  func=INPUT
<   GPIO 23: level=0 fsel=0 alt=  func=INPUT
---
>   GPIO 22: level=0 fsel=1 alt=  func=OUTPUT
>   GPIO 23: level=0 fsel=1 alt=  func=OUTPUT
29c29
<   GPIO 27: level=0 fsel=0 alt=  func=INPUT
---
>   GPIO 27: level=0 fsel=1 alt=  func=OUTPUT
```

pelwell, 2016-06-14:
Around here we are prepared to talk about header pin numbers, but the Broadcom "GPIO" pin numbering is the most important one because this is how the chip labels them and how the Linux pinctrl/gpio driver labels them. Trying to make sense of the overlays and tutorials using any other numbering scheme is just going to make your life harder. See page 102 of https://www.raspberrypi.org/wp-content/uploads/2012/02/BCM2835-ARM-Peripherals.pdf for the facts.

as-qs, 2016-06-14:
Trust me, I'm already having a nervous breakdown trying to keep up. The sad part is that I mostly deal with the pi4j numbers as that's how they are referenced from within my application, but I do understand your point.

pelwell, 2016-06-15:
BCM GPIOs 17, 22, 23 & 27 (header pins 11, 15, 16 & 13) are general purpose GPIOs (PI4J calls them GPIOs 0, 3, 4 and 2). The fact that they have become outputs (and nothing else has changed) doesn't tell us anything useful except that something is manipulating GPIOs. Perhaps that same thing is using GPIO4 (pin 7, PI4J 7) as either GPCLK0 or just another GPIO. I can see from your vcdbg log and DT dump that no kernel drivers should be touching them, so it must be up in the userspace/application layer.

as-qs, 2016-06-16:
I've isolated the problem to pin #4 (pi4j #7). SPI seems to be working correctly (this is confirmed) and so  do the output pins. The question is, what the hell is wrong with that input pin, and why is it that I'm not getting a callback from it when it goes low. 

Yesterday, before I had the above conclusions, I measured the voltage on all 40 pins with the following configurations: 

- RPI without the host board. with 4.1 kernel & 4.4 kernel.
- RPI with the host board. with 4.1 kernel & 4.4 kernel.

The results are compiled in this spreadsheet:
[RPI_Measurements.xlsx](https://github.com/raspberrypi/linux/files/319002/RPI_Measurements.xlsx)

**My conclusions from this test:**
- When the host board is NOT attached, GPIO 7 (pin 4) is set to 3.1V when my program is not running. it is set to zero when the program is running, which might be because my program sets it as input with pull down. 
- It is set to ~3.1V when the host board is attached with & without my program running.
- All other pins get identical readings in all every configuration.

pelwell, 2016-06-16:
I think it would help to strip the testing back to the very basics - can the GPIO subsystem detect edges on various pins. There is a simple GPIO watching framework described [here] (http://blog.oddbit.com/2014/07/26/gpiowatch-run-scripts-in-respo) with a git repo for the source. It would be interesting if you could install this on your two systems (4.1 and 4.4) and configure it to watch GPIO 4 (pin 7, PI4J 7) with edges in both directions. Then pull the pin high and low with a switch or a jumper cable and see if it triggers (remove the add-on-board for this testing). You could try a few pins, just to be sure.

pelwell, 2016-06-17:
By coincidence, [this](https://github.com/RPi-Distro/python-gpiozero/issues/382) appeared yesterday - a problem where the onewire (w1) driver clashes with GPIO edge detection on GPIO 4 (the default for w1).

What do you have in your config.txt?

as-qs, 2016-06-17:
Yeah, that pin is evil.

Config file:
[Config.txt](https://github.com/raspberrypi/linux/files/320973/Config.txt)

I'll be trying your test later in the day. I spoke to an EE about this, and he said it might have something to do with the timing or sequencing of how stuff are loaded. And it does seem like the 4.4 loads much faster than the 4.1.

as-qs, 2016-08-23:
I just want to report that my issue is resolved. By reducing the SPI speed from 390,000 to less than 100,000, everything is started working as normal. Apparently the new drivers don't like the higher speeds. I don't know if and how this might affect others, but at least we know one area where the new drivers behave differently from the previous ones.

closed 2016-08-23 by as-qs