raspberrypi/linux #1410: WiFi Inoperative when Bluetooth Deactivated

Ruler2112, 2016-04-15:
I just got my first Raspberry Pi 3 today.  There are several Pi 2's scattered around & I was looking forward to trying out the new model.

Because I have no bluetooth devices to pair with the Pi, I deactivated the bluetooth daemon by changing the first non-comment line in /etc/default/bluetooth from BLUETOOTH_ENABLED=1 to BLUETOOTH_ENABLED=0 and rebooted after tweaking a script I use.

At first I was puzzled because though the wlan0 interface had an address and showed as being up, I couldn't connect anywhere via http.  Then I discovered that the Pi couldn't ping anywhere, internet or local.  (At first I thought it was a problem with my script, until I discovered I couldn't ping.)  A light bulb went on over my head, so I enabled bluetooth and rebooted - network was accessible.  Change it back & can't get anywhere.  Nothing else changed between reboots, so I'm confident that there is a correlation here.

While I understand that both WiFi and bluetooth are wireless communication protocols, they do vastly different things & should operate independently of each other.  This Pi is running Raspian 4.1.18-v7+ #846 SMP installed with NOOBS 1.8.0.

Ruffio, 2017-01-26:
@Ruler2112 has this been fixed? I normally just disable loading of the drivers by: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=138610

By using the above method, the boot time of RPi is reduced significantly :-)

Ruler2112, 2017-01-26:
@Ruffio I honestly have no idea if it's been fixed or not - all the Pi units I have in place are running public-facing TVs, so it's not practical to take one out of service just to test a bug I've already worked around.  Should be easy enough to test if you have a spare Pi3 though... just change the one line above, reboot, and ping something to see if wifi works.

Wish I'd have found that thread before beating my head into a wall for 2 hours trying to figure out why the wifi didn't work... I'm the kind of person who, if I'm not going to be using something on a machine, turns off the daemon responsible for it to save memory/power/cpu time.  Too much stuff comes defaulted to 'on' nowadays IMO & then people wonder why their brand new machine runs slow. ;-)

closed 2017-05-18 by JamesH65