raspberrypi/linux #1417: I2S channel swap on RPi3 on 16-bit playback

bonezuk, 2016-04-19:
Hi

From customer feedback we've found that on playback of a 16-bit sample via an I2S sound card the left and right channels are intermittently being swapped around. This swap seems to be fairly random but it occurs on average once every 45 - 60 seconds. This is only happening on the RPi3 and not seen using the same kernel on a RPi2.

This is only seen with 16-bit sample, and is happening over different playback rates (8000, 22500, 44100, 48000, 192000). It is not seen with sample sizes of 24-bit or 32-bit.

I've been able to reproduce this issue on all our sound cards using the latest kernel so it is independent of the sound card. Also this is seen on cards where the I2S dsp is both the master and the slave.

Also this independent of player and has been seen on a number of distributions.

Thanks for any help in sorting this issue.

Stuart

clivem, 2016-04-19:
@bonezuk Stuart, I've not come across this.... You are able to reproduce with a 4.4.y kernel? Does it matter whether bclk_ratio is being set to 32 or 64 for 16 bit media? This really affects all of your cards, Digi, Light (ES9023) and DAC+ (PCM512x), with their respective drivers? I'll have a go at reproducing myself this afternoon.

pelwell, 2016-04-19:
I can reproduce this on a Pi3. I've not tried a Pi2 yet.

clivem, 2016-04-19:
Yep, I have tried to reproduce (with assorted hardware including HiFiBerry DAC+) and haven't been able to. @pelwell did you see the posts on the HifiBerry forum, that seem to be what made Staurt aware of the issue...

https://support.hifiberry.com/hc/en-us/community/posts/207307985-Raspberry-Pi-3-and-DAC-Pro

https://support.hifiberry.com/hc/en-us/community/posts/207250915-Hifiberry-light-Random-channel-swap

pelwell, 2016-04-19:
I hadn't seen them, but they don't add much except some doubt about whether the issue is Pi3-specific. I'm sure Stuart would have included any significant information.

bonezuk, 2016-04-19:
So yes the issue was initially reported by customers on our forum, particularly

https://support.hifiberry.com/hc/en-us/community/posts/207307985-Raspberry-Pi-3-and-DAC-Pro

I've only been able to reproduce it on the RPi3. I am unable to reproduce it on the RPi2 or RPi. The version of the kernel I reproduced it on was 4.1.19+. I am still to test this using the latest 4.4.y kernel.

The issue only happens when the bclk_ratio is set to 16 bit data. When the bclk_ratio is 24 or 32 the issue is not seen. Also I'll it appears to be independent of playback frequency.

I'll keep testing it and see if I can what other information I can pass on.

pelwell, 2016-04-19:
I've also failed to reproduce it on a Pi2 in 20-30 minutes of testing, and I can see it on the latest 4.4.y.

clivem, 2016-04-19:
@pellwell I cannot reproduce at all on either 3B or 2B. You are using Pi3B and latest 4.4.y to successfully reproduce...... Which hardware (DAC board) and driver have you reproduced it with?

HiassofT, 2016-04-20:
Interesting... I could reproduce it here as well (RPi3, latest 4.4 kernel, iqaudio card, raspbian jessie).

I dug out some older test commits and sure enough I2S signalled a TX FIFO error (underrun).
Here's my testing tree with quick and dirty IRQ error reporting: https://github.com/HiassofT/rpi-linux/commits/rpi-4.4.y-i2stest

I tried playing around with FIFO threshold settings and DMA panic levels but that didn't seem to fix the issue - but I could have missed something.

It's puzzling why the underrun occurred, that'd indicate that DMA data couldn't be transferred fast enough. But even more puzzling is the channel swap. DMA should be done in 32 bits and each frame is 32 bits as well in 16-bit stereo mode. I have no idea what's going on here.

pelwell, 2016-04-20:
I've managed to capture frames with two left or two right channels on a scope, both in slave and master mode. The clocking looks completely normal, which suggests the glitch is internal to the controller.

Interestingly, the normal state for the controller while playing correct (non-swapped) audio is with TXSYNC set to zero, i.e. out of sync. In the period where the audio is swapped, this sync flag goes to one, i.e. in sync.

clivem, 2016-04-20:
I've just figured out why I was unable to reproduce this...... It doesn't happen with @msperl backport of upstream dma to rpi-4.4.y. I forgot I was still running those patches on top of a base rpi-4.4.y. https://github.com/msperl/linux-rpi/commits/rpi-4.4.y-new-dmaengine

pelwell, 2016-04-21:
I can't see why the new upstream driver would behave differently, since the problem seems to be caused by overrunning the I2S TX FIFO and the new driver uses the same DREQ thresholds.

I have a patch which reduces the TX DREQ threshold to 0x20 (half full), and it's looking good. I'll test a bit more then push the patch if it survives.

pelwell, 2016-04-21:
I've pushed the patch after one hour without glitches.

hifiberry, 2016-04-21:
Any idea why this happened only on the RPI3?

clivem, 2016-04-21:
> I can't see why the new upstream driver would behave differently, since the problem seems to be caused by overrunning the I2S TX FIFO and the new driver uses the same DREQ thresholds.

@pelwell Sorry Phil. Can't shed any light on the "why", just that I was never able to reproduce it with the msperl upstream DMA code backport. But as soon as I reverted that patchset in my rpi-4.4.y builds..... I reproduced the "channel swap" and noticed the TX Errors with Mattias patch applied.

bcm2835-i2s 3f203000.i2s: interrupt status: 4
bcm2835-i2s 3f203000.i2s: TX Error IRQ
bcm2835-i2s 3f203000.i2s: RX status: RXF:0 RXD:0 RXR:0 RXERR:0 RXSYNC:0
bcm2835-i2s 3f203000.i2s: TX status: TXE:0 TXD:0 TXW:0 TXERR:1 TXSYNC:0

clivem, 2016-04-21:
@pelwell OK, will revert the 4x 3B's I have in my I2S test group to virgin rpi-4.4.y with your fix and leave them playing 16 bit overnight, and check them again in the morning.

msperl, 2016-04-22:
@clivem : the upstream version creates stable dma transfers really taking the requested length into account - the downstream version can create transfers that are just 4 bytes in size and each generates an interrupt and if that interrupt is not handled in time then you have an under run... (Or the wrong offset is assumed by the interrupt handler - so potentially what you see is the result of a concurrency issues)

pelwell, 2016-04-22:
But lowering the threshold for TX DREQ means that FIFO has to be more empty before it will request more data, making an underrun more likely. What we are seeing is the result of an overflow.

The theory here is that the A53s can keep the bus busy for longer, allowing the DMA pipeline to fill, and it is the surge of data when the DMA is released that overwhelms the FIFO before the DREQ signal has time to go low.

clivem, 2016-04-22:
Thanks guys! The issue wasn't an issue for me anyway with the upstream dma code patched-in, and isn't an issue (or reproducible for me) now with virgin rpi-4.4.y and Phil's "half-full" threshold patch.

bonezuk, 2016-04-22:
Reading the ongoing discussion I guess there is probably a patch (or series) of that I could probably build and test on our various sound cards? Hopefully I will get a bit of time to do some testing on this over the weekend.

clivem, 2016-04-22:
@bonezuk LOL. The original @msperl dma backport patch series has already been well tested on HifiBerry Light and DAC+. https://github.com/msperl/linux-rpi/commits/rpi-4.4.y-new-dmaengine

Hopefully, now this code has been accepted upstream, (IIRC it hadn't been at the time), @msperl  will update the series and it can be applied to downstream rpi-4.4.y. I've been running that patch set since Jan, and it has been tested with more I2S DAC HAT's than I can count on both fingers and toes, including yours.

Apply the last 15 patches from his tree with the exception of number 3. (The Revert "bcm2835-dma: Fix dreq not set for slave transfers", patch is already applied to rpi-4.4.y.)

clivem, 2016-04-22:
Stuart, you'll probably want to apply this patch after the @msperl 15 patch series to avoid an oops...

[msperl_dma_0016_GFP_ATOMIC.txt](https://github.com/raspberrypi/linux/files/231946/msperl_dma_0016_GFP_ATOMIC.txt)

bonezuk, 2016-04-22:
Ok cheers. Thanks for letting me know. I'll do that and see how it goes.

JamesH65, 2017-05-18:
Closing this issue as question answered/issue resolved.

closed 2017-05-18 by JamesH65

lucastanure, 2018-03-27:
I'm seeing channel swap on 4.9. Do we need this patch on 4.9 ?