raspberrypi/linux #1420: Hifiberry DAC+ Pro, driver not 'recognized' when using experimental gl driver on a pi 3

pimvlaar, 2016-04-20:
On a raspberry pi 3, with raspbian 

When i enable the experimental gl driver (done via raspi-config, advanced options), then after a reboot, the dac+pro driver is not recognized anymore.

Do you know a way to both use the dac+ pro for playing hifi music and use the experimental gl driver with the application mixxx for dj-ing.

I use a pi 3, with up-to-date raspbian.
(did not do a rpi-update, because after that my pi won't boot anymore)

output from uname -a:
Linux pi19 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

a piece of dmesg:
   3.828256] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    3.828276] [drm] No driver support for vblank timestamp query.
[    3.896483] Console: switching to colour frame buffer device 240x75
[    3.947201] vc4-drm soc:gpu@7e4c0000: fb0:  frame buffer device
[    3.947211] vc4-drm soc:gpu@7e4c0000: registered panic notifier
[    4.002081] snd-rpi-hifiberry-dacplus sound: ASoC: CPU DAI (null) not registered
[    4.002122] snd-rpi-hifiberry-dacplus sound: snd_soc_register_card() failed: -517
[    4.022814] snd-rpi-hifiberry-dacplus sound: ASoC: CPU DAI (null) not registered
[    4.022851] snd-rpi-hifiberry-dacplus sound: snd_soc_register_card() failed: -517
[    4.041936] systemd-journald[102]: Received request to flush runtime journal from PID 1
[    4.277359] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    4.277391] cfg80211: World regulatory domain updated:
[    4.277401] cfg80211:  DFS Master region: unset
[    4.277410] cfg80211:   (start_f

popcornmix, 2016-04-20:
rpi-update not working suggests you may have something unsuitable in config.txt. Can you post your config.txt?

pimvlaar, 2016-04-20:
Here is my current config.txt
Note: i have not yet done a rpi-update based on this file.

# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtparam=i2c_arm=on
#dtoverlay=vc4-kms-v3d

# hifi berry
#dtoverlay=pi3-disable-bt
#dtparam=spi=off
dtoverlay=hifiberry-dacplus

dtdebug=1

pimvlaar, 2016-04-20:
sorry for the big fonts, next time i will attache a file

popcornmix, 2016-04-20:
I don't see a reason for that to fail to boot with rpi-update. Can you try again?

popcornmix, 2016-04-20:
If it does boot, then post output of ``sudo vcdbg log msg``

pimvlaar, 2016-04-20:
what should i backup, before doing a rpi-update?
assume it fails, how do i recover?
what does a rpi-update give me, a new kernel with a newer 'experimental gl driver'.

pimvlaar, 2016-04-20:
here is output
[dmesg_with_experimental_gl_enabled.txt](https://github.com/raspberrypi/linux/files/228416/dmesg_with_experimental_gl_enabled.txt)
[dmesg_with_experimental_gl_enabled.txt](https://github.com/raspberrypi/linux/files/228418/dmesg_with_experimental_gl_enabled.txt)
[vcdbg_with_experimental_gl_enabled.txt](https://github.com/raspberrypi/linux/files/228417/vcdbg_with_experimental_gl_enabled.txt)

 from dmesg and vcdbg log msg (before a rpi-update)

closed 2016-04-20 by pimvlaar

popcornmix, 2016-04-20:
@pelwell any ideas?

HiassofT, 2016-04-20:
the vc4 3d driver uses the new clock framework but the I2S driver hasn't been (back-)ported to that yet so ATM they can't be used together.

This is why you get that line in dmesg:
bcm2708-i2s 3f203000.i2s: can't request region for resource [mem 0x3f101098-0x3f10109f]

That region has already been registered by the clock framework, so the I2S driver can't request that, too.

msperl posted that his I2S/clk patches already landed in the 4.6rc kernel series, but I diidn't have time to test them yet
https://github.com/raspberrypi/linux/issues/1231#issuecomment-209359399
https://github.com/torvalds/linux/commit/517e7a1537ae4663268be5d0c0ec62c563b9fc99

joaodriessen, 2016-04-21:
> msperl posted that his I2S/clk patches already landed in the 4.6rc kernel series, but I diidn't have time to test them yet

will the patches be made available to 4.4 kernel users?