raspberrypi/linux #1422: Enable DisplayLink Support; as a kernel module, by default

sigpaw, 2016-04-21:
I've located this wonderful little display that is totally USB powered. Its a nice size, reasonable price and evidently CAN work on a Pi (as others have done custom builds to prove this).

Here is what it looks like here:
http://www.amazon.com/AOC-e1659Fwu-1366 ... C+e1659Fwu

With a Pi 3, a USB hub, keyboard and mouse and this display, one could manage a self contained 5 volt only computer system.

With a cell phone battery pack and solar charger, and the Pi's reasonable price point. It would make for a very affordable computer that could be used ANYWHERE.

:-)

pelwell, 2016-04-21:
This is a wish rather than an issue. If you can find a module that does what you want that is part of the kernel tree then we'll consider building it by default, but for now this issue is closed.

closed 2016-04-21 by pelwell

sigpaw, 2016-04-21:
Actually not a wish but a polite request. The DisplayLink module does exist, and there are several inexpensive USB powered monitors "out there".

I've been patching/modding kernels since 2.0.34. Actually pioneered hard RT apps on large Level D flight sims (the LARGE $20,000,000.00 ones - am working on an updated 4.1 patch for RTAI as we speak). Doing this for myself will be a snap, and shouldn't require any memory or resources if it isn't built/loaded.

Apologies for the bother,

'sig

pelwell, 2016-04-21:
If you know of a driver in the kernel tree then by all means point us at it - don't make us search for it.
If it's an out of tree driver then suggest that the authors submit it.

Ferroin, 2016-04-21:
The config option for the DRM module is DRM_UDL,, the equivalent legacy frame-buffer one is FB_UDL, both have no dependencies that aren't already met in the standard kernel build on the Pi.  Note that these only work with the older USB 2.0 DisplayLink devices, the USB 3.0 ones use proprietary encryption on the USB link to satisfy HDCP requirements, and thus don't work with Linux (and are not likely to ever work with Linux).

sigpaw, 2016-04-21:
Ouch!

I loath DRM ( I was SHOCKED to see SlySoft go offline - thank heavens for SlyFox). [😊]

Actually the device I am working with right now works just fine via USB 2 (even though it is advertised as a USB 3 device). But we all know it will eventually be replaced with an "upgraded" version that likely won't. So the window on this may be narrow. Still makes for an intriguing little cheap solar powered computer...

sigpaw, 2016-04-21:
I am using this link as my reference:

https://imkiyoung.wordpress.com/2014/07/25/aoc-e1649fwu-16-usb-powered-portable-with-raspberry-pi/
[http://pds27.egloos.com/pds/201407/25/77/d0009977_53d1b89c3fff4.png]<https://imkiyoung.wordpress.com/2014/07/25/aoc-e1649fwu-16-usb-powered-portable-with-raspberry-pi/>

AOC E1649FWU 16? USB-Powered Portable with Raspberry Pi<https://imkiyoung.wordpress.com/2014/07/25/aoc-e1649fwu-16-usb-powered-portable-with-raspberry-pi/>
imkiyoung.wordpress.com
My old post -    AOC E1649FWU 16? USB-Powered Portable has no HDMI connector but USB 2.0 Y type cable(one is for additional 500mA power). This article is written based on 2014-06-20-whe...

Enjoy!

'sig

Ferroin, 2016-04-21:
Just to clarify, in the context of graphics drivers in the Linux kernel, DRM usually refers to the Direct Rendering Manager framework, which is what almost everything with more than a basic text console is using these days as it's faster, more scalable, and overall better designed than the old framebuffer interfaces.

But yeah, I'm not too happy about DisplayLink's choice there (I understand offering ones that support HDCP, but not offering any that don't is just stupid, and there's no way they can _truly_ provide the secure communications that HDCP requires over USB, because it's trivial to reverse-engineer _anything_ that goes over USB), even more so because that's how most laptop docks do their integrated video outputs these days...

That said, I very much doubt that the old USB 2.0 stuff will go away any time soon, the protocol is well documented, and it's perfectly legal in most jurisdictions to reuse a communications protocol for the purposes of interoperability.

pelwell, 2016-04-21:
@Ferroin Can you (or anybody) vouch for the fact that, at least as far as the kernel is concerned, enabling CONFIG_DRM_UDL is sufficient to allow the use of this display? We don't have one, so can't test it.

sigpaw, 2016-04-21:
I have one on the way (from B&H photo) and will be "playing" with it in short order. [😉]

Ferroin, 2016-04-21:
Assuming that the people who have gotten this working already haven't used third party drivers, then I can say with almost 100% certainty that it will work, because the only other USB display adapters supported in mainline are almost impossible to find these days, and were never really hugely popular to begin with.

pelwell, 2016-04-21:
@popcornmix CONFIG_DRM_UDL just builds a 50KB module. Are you happy for me to add this?

sigpaw, 2016-04-21:
Yes indeedy. Just a coincidence that Digital Rights Management and Direct Rendering Manager both share the same TLA. (many TLAs have fallen along this path, take APM for instaance). Anyway, it was your reference to HDCP that "set me off" so to speak.

I recently upgraded the living room TV to one of those "new fangled" OLED 4k displays. Thus I also had to upgrade my Media center graphics card to insure HDCP 2.2 compliance. As it turns out the Nvidia GTX950/960 ended up being my best bet since the GM206 has hardware H.265 built in (as well as 2.2 compliance).

The whole process bothered me, I was disappointed that AMD had not yet supported HDMI 2.0, and disturbed by the rumors that they never would support HDCP 2.2. It's NUTS that anyone bothers with all this encryption, someone will ALWAYS find a way to break it. In the mean time it forces the rest of us to constantly upgrade and to discard otherwise perfectly good hardware. grrrrrr (I digress)

I hope you are right on the legacy hardware. Heck I can still find ISA buss motherboards if I purchase servers meant for process control and the like.

popcornmix, 2016-04-21:
@pelwell no objections.

pelwell, 2016-04-21:
Pushed.

sigpaw, 2016-04-21:
Awesome. Thanks! [😃]

Once I get RTAI working on the Pi do you think there would be any interest (outside of my research) in hard real-time thread capabilities for the Pi?

pelwell, 2016-04-21:
That sounds like a good question for the ["Advanced users" forum](https://www.raspberrypi.org/forums/viewforum.php?f=29).

hemangjoshi37a, 2017-05-13:
Are the commits solved yet?? Are the modules for USB powered display drivers added to the main Raspbian Kernel??

popcornmix, 2017-05-13:
Yes, CONFIG_DRM_UDL is enabled in kernel.

aqueousboot9256, 2018-07-26:
Any chance someone could write a guide how to use what was discussed here to get the AOC portable monitor working with a Rasberry Pi 3? Preferably for someone very novice in kernel editing or whatever this is called. Thanks!

Moriarte, 2018-12-07:
Hi, Total noob here, I want to use this same USB monitor on my rpi with pihole but i'm rather confused by all the stuff posted in this thread as i'm very new to linux. 

could someone tell me how to get this working on a fresh install of Raspbian Stretch?

sigpaw, 2018-12-09:
I don't believe it is possible. The newer monitors have DRM (Digital Rights Management) built into their USB3 communications. SO if the proper security info isn't sent, the display will not un-blank. From my understanding only Microsoft has been given the proper driver that is able to do this. I don't believe they will ever allow an "open source" OS access to this info. Thank Dollarwood for this! :-(

Moriarte, 2018-12-09:
Aww, thanks anyways Sigpaw!