raspberrypi/linux #1427: Excessive number of "bioset" kernel processes in 4.4 kernel

ncguk, 2016-04-23:
Since upgrading to kernel 4.4.7 (and now 4.4.8) on an otherwise stock 2016-03-18-raspbian-jessie-lite I have 25 kernel processes named bioset showing in top output, each with a nice priority of -20, e.g.:

`60 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 bioset`

This happens on an RPi B+, RPi2 and RPi3, all run headless. As I didn't see this on the 4.1 kernel, am I to assume this is not desired behaviour?

popcornmix, 2016-04-23:
Seems like a kernel working thread involved in block I/O. See:
https://answers.launchpad.net/ubuntu/+question/272922

I don't think there is any issue. It takes no CPU when it is not in use.

ncguk, 2016-04-23:
It's true they don't seem to consume CPU or RAM. It's odd, though, isn't it, that they weren't there in 4.1 and now are in 4.4?

popcornmix, 2016-04-23:
That is a decision the upstream kernel developers have taken.
If you search around for bioset you can see these kernel threads are present on all platforms with newer kernels. I can find no evidence that there is a problem with that.

ncguk, 2016-04-23:
Fair enough. I'll close the ticket.

closed 2016-04-23 by ncguk