raspberrypi/linux #1429: mmc_spi broken?

franzflasch, 2016-04-23:
I am trying to get an SD Card to work via SPI interface. I am using a RaspberryPI Compute Module with kernel Version 4.4, but I can't get it to work properly. I've tried it with the native spi interface as well as with the spi aux interface and also with a gpio bitbang spi interface. It's always the same error:
 
[ 2639.558282] mmc1: new SD card on SPI
[ 2639.566325] mmcblk1: mmc1:0000 SD256 243 MiB 
[ 2639.646120]  mmcblk1: p1 < >
[ 2639.833759] mmcblk1: error -38 sending status command, retrying
[ 2639.839973] mmcblk1: error -38 sending status command, retrying
[ 2639.846170] mmcblk1: error -38 sending status command, aborting
[ 2639.853406] mmc1: SPI card removed
[ 2640.240223] mmc1: host does not support reading read-only switch, assuming write-enable
[ 2640.248328] mmc1: new SD card on SPI
[ 2640.256373] mmcblk1: mmc1:0000 SD256 243 MiB 
[ 2640.336068]  mmcblk1: p1 < >
[ 2640.523545] mmcblk1: error -38 sending status command, retrying
[ 2640.529757] mmcblk1: error -38 sending status command, retrying
[ 2640.535893] mmcblk1: error -38 sending status command, aborting
[ 2640.543129] mmc1: SPI card removed
[ 2640.930176] mmc1: host does not support reading read-only switch, assuming write-enable
[ 2640.938313] mmc1: new SD card on SPI
[ 2640.946679] mmcblk1: mmc1:0000 SD256 243 MiB 
[ 2641.026395]  mmcblk1: p1 < >
[ 2641.213996] mmcblk1: error -38 sending status command, retrying
[ 2641.220208] mmcblk1: error -38 sending status command, retrying
[ 2641.226344] mmcblk1: error -38 sending status command, aborting
[ 2641.233435] mmc1: SPI card removed
[ 2641.620177] mmc1: host does not support reading read-only switch, assuming write-enable
[ 2641.628277] mmc1: new SD card on SPI
[ 2641.636530] mmcblk1: mmc1:0000 SD256 243 MiB 
[ 2641.716114]  mmcblk1: p1 < >
[ 2641.903636] mmcblk1: error -38 sending status command, retrying
[ 2641.909852] mmcblk1: error -38 sending status command, retrying
[ 2641.915987] mmcblk1: error -38 sending status command, aborting
[ 2641.923056] mmc1: SPI card removed
 
I don't have any RO or CD pins configured, just plain SPI interface directly to the SD Card. I've tried it with different SD cards with different capacities. It's always the same error. The above messages appear in a endless loop. The last time I've used the mmc_spi driver was kernel 3.6 and it worked without any problems.

I've already patched the kernel with this patch: https://github.com/msperl/linux-rpi/commit/aef50a19f335ae3d2eb43ac5496c7f03930b394f which solves the problem, that the card was not recognized, however it still does not work.

Does anyone know how to fix this?
 
Thanks,
Franz

pelwell, 2016-04-25:
Is this correct?
```
[ 2639.566325] mmcblk1: mmc1:0000 SD256 243 MiB 
```
That is a peculiar size and very small, so perhaps you have a large card that isn't being recognised. Have you tried with 2GB card? Perhaps SDHC isn't supported.

pelwell, 2016-04-25:
Also, this isn't really a kernel bug, although it could eventually turn out to be one. You would be better off asking this question in the [Interfacing Forum](https://www.raspberrypi.org/forums/viewforum.php?f=44). Be sure to include details of the hardware you are using and any specific configuration you have done.

franzflasch, 2016-04-25:
Hi pelwell!

243 is absolutely correct. I've also tried it with 512MB and 2GB Cards. They always get recognized correctly but the error above appears with every card.

franzflasch, 2016-04-25:
For testing I switched back to kernel 4.1 and it seemed the same error occurs. However after several attempts the kernel made it to mount the device properly:

[   76.422753] mmc1: new SD card on SPI
[   76.431482] mmcblk1: mmc1:0000 SU02G 1.89 GiB 
[   76.483269]  mmcblk1: p1
[   76.706695] mmcblk1: error -38 sending status command, retrying
[   76.712992] mmcblk1: error -38 sending status command, retrying
[   76.719275] mmcblk1: error -38 sending status command, aborting
[   76.726636] mmc1: SPI card removed
[   76.814568] mmc1: host does not support reading read-only switch, assuming write-enable
[   76.822755] mmc1: new SD card on SPI
[   76.831524] mmcblk1: mmc1:0000 SU02G 1.89 GiB 
[   76.883258]  mmcblk1: p1
[   77.131966] mmcblk1: error -38 sending status command, retrying
[   77.138157] mmcblk1: error -38 sending status command, retrying
[   77.145221] mmcblk1: error -38 sending status command, aborting
[   77.151946] mmc1: SPI card removed
[   77.295305] mmc1: host does not support reading read-only switch, assuming write-enable
[   77.303415] mmc1: new SD card on SPI
[   77.312247] mmcblk1: mmc1:0000 SU02G 1.89 GiB 
[   77.363731]  mmcblk1: p1

And I also can mount and access it successfully! Anayway I think there is something wrong with the mmc_spi in the newer kernel versions. I can't be the first person who has this problems...

msperl, 2016-04-27:
It is a know "issue" of the kernel - at least the last time I checked.
This is the offending commit: 0589342c27944e50ebd7a54f5215002b6598b748, but I would not revert that.

This is a workarround patch:
```
diff --git a/drivers/mmc/host/mmc_spi.c b/drivers/mmc/host/mmc_spi.c
index 1c1b45e..f882bd7 100644
--- a/drivers/mmc/host/mmc_spi.c
+++ b/drivers/mmc/host/mmc_spi.c
@@ -1386,7 +1386,7 @@ static int mmc_spi_probe(struct spi_device *spi)
        host->data = kmalloc(sizeof(*host->data), GFP_KERNEL);
        if (!host->data)
                goto fail_nobuf1;
-
+#if 0
        if (spi->master->dev.parent->dma_mask) {
                struct device   *dev = spi->master->dev.parent;

@@ -1402,7 +1402,7 @@ static int mmc_spi_probe(struct spi_device *spi)
                                host->data_dma, sizeof(*host->data),
                                DMA_BIDIRECTIONAL);
        }
-
+#endif
        /* setup message for status/busy readback */
        spi_message_init(&host->readback);
        host->readback.is_dma_mapped = (host->dma_dev != NULL);
```

As the new spi framework does dma on its own we do not have to rely on mmc_spi doing the right thing with dma mapping itself.

franzflasch, 2016-04-27:
Thanks for your response! As I already mentioned above I've already patched the kernel with your patch. Without this patch it doesn't come that far at all. 

In the meantime I've got new findings regarding this issue. It seems it has something to do with udev. If I do:
`/etc/init.d/udev stop`
The card gets recognized correctly, and I can mount it manually. I also can access the data on the card. So it seems to me it is a udev issue in my case.

closed 2016-07-25 by franzflasch