raspberrypi/linux #1440: bcm2835-v4l2 Manual ISO not working properly

Austriker, 2016-04-25:
Following this issue : #1251 @6by9 and the golden rule of bug tracking 😄 
And this post on the raspberry forum : [https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=145397](url)

I am opening a issue because I found out that manual iso control of the V4L2 driver does not work.

I launch the camera like this : 
```sh
v4l2-ctl --set-ctrl=video_bitrate=25000000
v4l2-ctl -v width=2592,height=1944,pixelformat=5
v4l2-ctl -p 1
v4l2-ctl --set-ctrl=auto_exposure=1
v4l2-ctl --set-ctrl=iso_sensitivity_auto=0
```

When I change the iso with : `v4l2-ctl --set-ctrl=iso_sensitivity=4` the driver accepts the change and if I check the status I can see the good value : 

```sh
pi@raspberrypi ~ $ v4l2-ctl --log-status

Status Log:

   [23366.234897] bcm2835-v4l2: =================  START STATUS  =================
   [23366.234928] bcm2835-v4l2: Saturation: 0
   [23366.234951] bcm2835-v4l2: Sharpness: 0
   [23366.234968] bcm2835-v4l2: Contrast: 0
   [23366.234984] bcm2835-v4l2: Brightness: 50
   [23366.234999] bcm2835-v4l2: ISO Sensitivity: 800000
   [23366.235016] bcm2835-v4l2: ISO Sensitivity, Auto: Manual
   [23366.235032] bcm2835-v4l2: Image Stabilization: false
   [23366.235048] bcm2835-v4l2: Auto Exposure: Manual Mode
   [23366.235065] bcm2835-v4l2: Exposure Time, Absolute: 100
   [23366.235081] bcm2835-v4l2: Auto Exposure, Bias: 0
   [23366.235097] bcm2835-v4l2: Exposure, Dynamic Framerate: false
   [23366.235113] bcm2835-v4l2: Exposure, Metering Mode: Average
   [23366.235129] bcm2835-v4l2: White Balance, Auto & Preset: Auto
   [23366.235144] bcm2835-v4l2: Red Balance: 1000
   [23366.235160] bcm2835-v4l2: Blue Balance: 1000
   [23366.235176] bcm2835-v4l2: Color Effects: None
   [23366.235193] bcm2835-v4l2: Color Effects, CbCr: 32896
   [23366.235209] bcm2835-v4l2: Rotate: 0
   [23366.235228] bcm2835-v4l2: Horizontal Flip: false
   [23366.235243] bcm2835-v4l2: Vertical Flip: false
   [23366.235259] bcm2835-v4l2: Video Bitrate Mode: Variable Bitrate
   [23366.235275] bcm2835-v4l2: Video Bitrate: 25000000
   [23366.235291] bcm2835-v4l2: Compression Quality: 30
   [23366.235308] bcm2835-v4l2: Power Line Frequency: 50 Hz
   [23366.235326] bcm2835-v4l2: Repeat Sequence Header: false
   [23366.235342] bcm2835-v4l2: H264 Profile: High
   [23366.235357] bcm2835-v4l2: H264 Level: 4
   [23366.235373] bcm2835-v4l2: Scene Mode: None
   [23366.235389] bcm2835-v4l2: H264 I-Frame Period: 60
   [23366.235405] bcm2835-v4l2: ==================  END STATUS  ==================
```

When I check the video file created with `v4l2-ctl --stream-mmap=3 --stream-to=$(date +"%Y%m%d_%H%M%S")_v4l2.mjpg` there is no modification of the brightness of the video. 

It seems that changing changing the ISO has no impact on the bightness of the video. An other user reported having this issue.

Delgan, 2016-06-21:
@6by9 Using the new camera v2.1 and v4l2-ctl, I made the initial steps described here and I tried to set `iso_sensitivity` to `0` and then to `4`.

Unfortunately, the images looked the same, without any noticeable change in the brightness.

Some examples (`0` is on the left, `4` on the right):

[image] vs [image]

[image] vs [image]

JamesH65, 2017-05-18:
@6by9 ANy thoughts?

6by9, 2017-05-18:
Increasing ISO whilst allowing automatic exposure control just means that AE will choose a shorter exposure time to compensate. It does not adjust the target brightness for the scene (that would be exposure compensation, aka EV, or V4L2_CID_AUTO_EXPOSURE_BIAS).

You'd need to be able to see the tuner state to confirm, or capture JPEGs as the EXIF in those would show the ISO value.

JamesH65, 2017-05-18:
@Delgan  Can this be closed?

Delgan, 2017-05-18:
Thanks for the explanation, I guess it makes sense to close this issue then.

closed 2017-09-13 by JamesH65

dstarke, 2018-03-22:
I think this is a real problem. When `auto_exposure` is set to manual (value 1) with some exposure time configured, changing the `iso_sensitivity` value has no apparent effect. But if I set `auto_exposure` to auto (value 0), then change `iso_sensitivity`, wait briefly, and then set `auto_exposure` back to manual, the images captured after this set of changes are different than the ones before the change.

biasedlogic, 2021-01-19:
This is still a problem and still hasn't been fixed. This issue should not be closed!

biasedlogic, 2021-01-19:
https://github.com/showmewebcam/showmewebcam/issues/86 there's a discussion here with more details on how this can be observed / reproduced. Basically after setting all parameters to manual changing ISO should directly be visible as changing registered image brightness. But it doesn't, you need to change some other irrelevant parameter, like image flip, for the setting to 'take'.