raspberrypi/linux #1441: recipe for target 'include/generated/timeconst.h' failed

NoHomey, 2016-04-26:
Today i installed latest linux kernel (4.4.8-v7+) for my new RPI 2 but when trying to get the linux header files I'm geting an error. Tried with rpi-source and linux-headers-4.4.8-v7+.deb and all resulted in this error:

```
root@raspberrypi:/usr/src/linux# make prepare
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/msi.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/seccomp.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
  CHK     include/generated/bounds.h
  UPD     include/generated/bounds.h
  CHK     include/generated/timeconst.h
/bin/sh: 1: bc: not found
Kbuild:66: recipe for target 'include/generated/timeconst.h' failed
make[1]: *** [include/generated/timeconst.h] Error 127
Makefile:986: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
root@raspberrypi:/usr/src/linux# 
```

MrEngman, 2016-04-26:
Install `bc`

```sudo apt-get update```
```sudo apt-get install bc```

NoHomey, 2016-04-26:
Got new error:

```
pi@raspberrypi:~ $ sudo apt-get install  bc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 linux-headers-4.4.8-v7+ : Depends: gcc-4.7 but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
pi@raspberrypi:~ $ sudo apt-get install -f
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following extra packages will be installed:
  cpp-4.7 gcc-4.7 libgcc-4.7-dev
Suggested packages:
  gcc-4.7-locales libmudflap0-4.7-dev gcc-4.7-doc libgcc1-dbg libgomp1-dbg libitm1-dbg libquadmath-dbg libmudflap0-dbg libcloog-ppl0 libppl-c4 libppl9 libpwl5
The following NEW packages will be installed:
  cpp-4.7 gcc-4.7 libgcc-4.7-dev
0 upgraded, 3 newly installed, 0 to remove and 76 not upgraded.
1 not fully installed or removed.
Need to get 6798 kB of archives.
After this operation, 23.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main cpp-4.7 armhf 4.7.3-11+rpi1 [3252 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main libgcc-4.7-dev armhf 4.7.3-11+rpi1 [113 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main gcc-4.7 armhf 4.7.3-11+rpi1 [3433 kB]
Fetched 6798 kB in 4s (1478 kB/s)  
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_TIME = "bg_BG.UTF-8",
	LC_MONETARY = "bg_BG.UTF-8",
	LC_ADDRESS = "bg_BG.UTF-8",
	LC_TELEPHONE = "bg_BG.UTF-8",
	LC_NAME = "bg_BG.UTF-8",
	LC_MEASUREMENT = "bg_BG.UTF-8",
	LC_IDENTIFICATION = "bg_BG.UTF-8",
	LC_NUMERIC = "bg_BG.UTF-8",
	LC_PAPER = "bg_BG.UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_ALL to default locale: No such file or directory
Selecting previously unselected package cpp-4.7.
(Reading database ... 140650 files and directories currently installed.)
Preparing to unpack .../cpp-4.7_4.7.3-11+rpi1_armhf.deb ...
Unpacking cpp-4.7 (4.7.3-11+rpi1) ...
Selecting previously unselected package libgcc-4.7-dev:armhf.
Preparing to unpack .../libgcc-4.7-dev_4.7.3-11+rpi1_armhf.deb ...
Unpacking libgcc-4.7-dev:armhf (4.7.3-11+rpi1) ...
Selecting previously unselected package gcc-4.7.
Preparing to unpack .../gcc-4.7_4.7.3-11+rpi1_armhf.deb ...
Unpacking gcc-4.7 (4.7.3-11+rpi1) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up cpp-4.7 (4.7.3-11+rpi1) ...
Setting up libgcc-4.7-dev:armhf (4.7.3-11+rpi1) ...
Setting up gcc-4.7 (4.7.3-11+rpi1) ...
Setting up linux-headers-4.4.8-v7+ (4.4.8-v7+-2) ...
make: Entering directory '/usr/src/linux-headers-4.4.8-v7+'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
*** Default configuration is based on 'bcm2709_defconfig'
[... 35 lines cut ...]
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
  CHK     include/generated/bounds.h
  UPD     include/generated/bounds.h
make[1]: *** No rule to make target 'kernel/time/timeconst.bc', needed by 'include/generated/timeconst.h'.  Stop.
Makefile:986: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
make: Leaving directory '/usr/src/linux-headers-4.4.8-v7+'
/var/lib/dpkg/info/linux-headers-4.4.8-v7+.postinst: 1: /var/lib/dpkg/info/linux-headers-4.4.8-v7+.postinst: dkms: not found
pi@raspberrypi:~ $ 
```

NoHomey, 2016-04-26:
Executing apt-get install bc from /usr/src worked. I ran again and make prepare module_prepare all went ok but when I try to build my kernel module this is what i get:
```
make -C /lib/modules/4.4.8-v7+/build M=/home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin clean
make[1]: Entering directory '/usr/src/linux-headers-4.4.8-v7+'
  CLEAN   /home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin/.tmp_versions
make[1]: Leaving directory '/usr/src/linux-headers-4.4.8-v7+'
make -C /lib/modules/4.4.8-v7+/build M=/home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.8-v7+'
  CC [M]  /home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin/input_pin.o
In file included from include/linux/ktime.h:25:0,
                 from include/linux/rcupdate.h:47,
                 from include/linux/srcu.h:33,
                 from include/linux/notifier.h:15,
                 from include/linux/memory_hotplug.h:6,
                 from include/linux/mmzone.h:735,
                 from include/linux/gfp.h:5,
                 from include/linux/kmod.h:22,
                 from include/linux/module.h:13,
                 from /home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin/input_pin.h:5,
                 from /home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin/input_pin.c:1:
include/linux/jiffies.h:10:33: fatal error: generated/timeconst.h: No such file or directory
compilation terminated.
scripts/Makefile.build:264: recipe for target '/home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin/input_pin.o' failed
make[2]: *** [/home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin/input_pin.o] Error 1
Makefile:1384: recipe for target '_module_/home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin' failed
make[1]: *** [_module_/home/pi/Linux_Device_Driven_LED_Game/kernel_modules/input_pin] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.8-v7+'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
```

NoHomey, 2016-04-26:
removing all linux files from /usr/src and runing ```rpi-source``` now lets me successfully compile kernel modules

closed 2016-04-26 by NoHomey

eberhab, 2016-06-09:
Hi all, I still get this error even after installing bc:

`CC      kernel/bounds.s
  CHK     include/generated/bounds.h
  UPD     include/generated/bounds.h
make[1]: *** Keine Regel vorhanden, um das Ziel „kernel/time/timeconst.bc“, 
  benötigt von „include/generated/timeconst.h“, zu erstellen.  Schluss.
Makefile:986: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
make: Leaving directory '/usr/src/linux-headers-4.4.11-v7+'
`

ben423423n32j14e, 2016-07-27:
I also installed BC, even tried uninstalling and re-installing it and get error:

`make[1]: *** No rule to make target 'kernel/time/timeconst.bc', needed by 'include/generated/timeconst.h'.  Stop.
Makefile:986: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2`