raspberrypi/linux #1442: RPi 3 WiFi - kernel NULL pointer dereference

ali1234, 2016-04-26:
Using 2016-03-18-raspbian-jessie-lite. This is reproducible every single time, within a minute of booting.

These files were modified:
`
/etc/network/interfaces - "wireless-power off" was added:

# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet manual
    wireless-power off
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

allow-hotplug wlan1
iface wlan1 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
`

wpa_supplicant.conf - ssid and psk were added:
`
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="myssid"
    psk="mypassphrase"
}
`

No other files in the system were modified.

Full dmesg from boot to crash:

`
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.19-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #858 SMP Tue Mar 15 15:56:00 GMT 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a400000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80880fc0, node_mem_map b9bb4000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9420->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 12 pages/cpu @bafb0000 s20416 r8192 d20544 u49152
[    0.000000] pcpu-alloc: s20416 r8192 d20544 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa02082 bcm2709.serial=0x9a5f0c37 smsc95xx.macaddr=B8:27:EB:5F:0C:37 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[... 464 lines cut ...]
[   52.488245] [<800091e8>] (do_DataAbort) from [<805783dc>] (__dabt_svc+0x3c/0x60)
[   52.497686] Exception stack(0xb8cefc88 to 0xb8cefcd0)
[   52.503772] fc80:                   0000000d b668097c 000000c9 00000000 b668098d b6680160
[   52.513964] fca0: 000000b8 000000c9 00000000 00000000 00000064 b8cefcf4 00000003 b8cefcd0
[   52.524144] fcc0: fffffff6 7f10223c 60000013 ffffffff
[   52.530223] [<805783dc>] (__dabt_svc) from [<7f10223c>] (cfg80211_get_bss_channel+0xf8/0x190 [cfg80211])
[   52.541736] [<7f10223c>] (cfg80211_get_bss_channel [cfg80211]) from [<7f10310c>] (cfg80211_inform_bss_width+0x70/0x274 [cfg80211])
[   52.555585] [<7f10310c>] (cfg80211_inform_bss_width [cfg80211]) from [<7f22c6fc>] (brcmf_inform_single_bss+0xd0/0x140 [brcmfmac])
[   52.569370] [<7f22c6fc>] (brcmf_inform_single_bss [brcmfmac]) from [<7f22d150>] (brcmf_inform_bss+0x70/0xac [brcmfmac])
[   52.582269] [<7f22d150>] (brcmf_inform_bss [brcmfmac]) from [<7f22eac0>] (brcmf_cfg80211_escan_handler+0x6c/0x354 [brcmfmac])
[   52.595707] [<7f22eac0>] (brcmf_cfg80211_escan_handler [brcmfmac]) from [<7f234ddc>] (brcmf_fweh_call_event_handler+0x48/0xa4 [brcmfmac])
[   52.610209] [<7f234ddc>] (brcmf_fweh_call_event_handler [brcmfmac]) from [<7f234ea0>] (brcmf_fweh_event_worker+0x68/0x2d0 [brcmfmac])
[   52.624339] [<7f234ea0>] (brcmf_fweh_event_worker [brcmfmac]) from [<8003e0b8>] (process_one_work+0x154/0x458)
[   52.636448] [<8003e0b8>] (process_one_work) from [<8003e410>] (worker_thread+0x54/0x500)
[   52.646628] [<8003e410>] (worker_thread) from [<80043bec>] (kthread+0xec/0x104)
[   52.655049] [<80043bec>] (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)
[   52.663384] Code: e24cb004 e52de004 e8bd4000 e59032c0 (e5130014) 
[   52.670564] ---[ end trace 2952b3331628e3de ]---
[   52.676221] Fixing recursive fault but reboot is needed!
[   73.718559] INFO: rcu_sched detected stalls on CPUs/tasks:
[   73.725101] 	3: (1 GPs behind) idle=ca1/140000000000000/0 softirq=1213/1213 fqs=1852 
[   73.734913] 	(detected by 2, t=2104 jiffies, g=184, c=183, q=119)
[   73.742041] Task dump for CPU 3:
[   73.746221] kworker/3:1     D 80573e10     0    53      0 0x00000000
`

ali1234, 2016-04-26:
This appears to be a very similar crash to the crash with the official USB wifi dongle which I reported last year:

https://github.com/raspberrypi/linux/issues/1234

ali1234, 2016-04-26:
Here is another report of the same crash: https://github.com/guysoft/OctoPi/issues/207

ali1234, 2016-04-26:
This also looks like the same crash, even though it is completely different hardware: https://bugzilla.kernel.org/show_bug.cgi?id=100201

pelwell, 2016-04-26:
I've followed those steps on a clean Jessie-Lite installation and I'm not seeing a crash. How is your WiFi network configured?

SirVer, 2016-04-27:
I see the same issue. For me it depends on the WiFi I connect to though. With a Vodafone box, everything was fine. Now I switched to a "o2 HomeBox 6441" and I see crashes. My analysis leads me to believe that the channel for the 2.4GHz network is the culprit. See below.

The new box supports 5Ghz 802.11ac/n/a AND 802.11b/g/n which can be turned on and off independently and on different SSIDs. Also the channel can be AUTO or manually set. 

I played around with the settings and figured the following out: If I let the channel on AUTO for 2.4Ghz, the router chooses 13, this leads to a crash on the raspberry PI. Choosing manually the channel gives these results: CH 12 and 13 crash the RaspberryPi, 11 and everything below are working fine.

pelwell, 2016-04-27:
Thank you, @SirVer, that's useful feedback.

pelwell, 2016-04-27:
For some reason the brcmfmac driver in the 4.1 kernel doesn't support channels 12 and 13, even though the regulations in some regions allow them.

I'm in the UK, and on 4.1 I see:
```
pi@raspberrypi:~$ iwlist wlan0 channel
wlan0     11 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Current Frequency:2.437 GHz (Channel 6)
```
but 4.4 shows:
```
pi@raspberrypi:~$ iwlist wlan0 channel
wlan0     11 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Current Frequency:2.437 GHz (Channel 6)
```
Perhaps the crash occurs because the lack of support is as the result of a bug, and in some code paths that leads to a NULL pointer.

ali1234, 2016-04-27:
Channel 13 here too, also with router supporting AC and multiple SSIDs (both are in use).

Klamath233, 2016-05-02:
```
[   13.127109] Bluetooth: BNEP socket layer initialized
[   51.057048] Unable to handle kernel NULL pointer dereference at virtual address 00000000
```
Did the kernel freeze in the interval between these two messages?

pelwell, 2016-05-02:
There is no reason to think that there was a freeze. Kernels aren't normally very chatty.

The solution to this problem is to upgrade to a 4.4 kernel. All rpi-update kernels are now 4.4-based, and an apt-get 4.4 kernel will be released soon.

Klamath233, 2016-05-02:
Thank you @pelwell

ali1234, 2016-05-16:
On 4.4.9+ #884 I still can't connect to channel 12 or 13 APs with brcmfmac devices.

pelwell, 2016-05-16:
What is your regulatory domain set to? What output do you get from `iw list`?

ali1234, 2016-05-16:
I'll get all that info for you tomorrow. Note when I said "I cant connect" I meant that I can't connect because the Pi still crashes. It does connect and get an IP address and set the clock correctly with NTP... then it just dies a couple of minutes later. I'll have to connect up an ethernet cable to see what is going on.

ali1234, 2016-05-17:
It turns out that the access point set to channel 13 also has wireless isolation enabled. It actually works fine now.

pelwell, 2016-05-17:
Does that mean we can close this now?

ali1234, 2016-05-17:
Fixed in 4.4

closed 2016-05-17 by ali1234