raspberrypi/linux #1445: Latest git release build fails at "LD      init/built-in.o"

jdb2, 2016-04-30:
I'm running on Linux Mint 17.3 Cinnamon 64-bit and I've tried building ( via a cross-compile ) the latest git snapshot of the Raspberry Pi Linux kernel as described in the official Raspberry Pi build guide [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md).

My system information is as follows ( uname -a ) : 

`Linux jdb2-Satellite-M305 3.19.0-32-generic #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux`

Here is the build output : 

```
  LD      init/built-in.o
drivers/built-in.o: In function `hcd_init_fiq':
:(.text+0xec550): undefined reference to `claim_fiq'
:(.text+0xec5ec): undefined reference to `set_fiq_handler'
:(.text+0xec63c): undefined reference to `__set_fiq_regs'
:(.text+0xec6f4): undefined reference to `enable_fiq'
make: *** [vmlinux] Error 1
```

Here is the "git log" output : 

```
commit f6f3129ad7b49499004510b6a252f222c029efcb
Author: popcornmix <popcornmix@gmail.com>
Date:   Fri Apr 29 17:30:00 2016 +0100

    Revert "bcm2835: extend allowed range of channels and samplerates"
    
    This reverts commit 688a5f0daa45e0a51b324707768d472e1d715c13.
```

pelwell, 2016-04-30:
It just built for me with no difficulties. Start from clean and list your build commands.

jdb2, 2016-04-30:
I used the procedure that I usually use from the official Raspberry Pi kernel build instructions. Here are my build commands, in the order that they are issued : 

```
cd ~/src/rpi4
git clone https://github.com/raspberrypi/tools
PATH=$PATH:/home/jdb2/src/rpi4/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64:/home/jdb2/src/rpi4/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin
git clone --depth=1 https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel7
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
make -j 3 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs
```

The above results in an : 

```
  LD      init/built-in.o
drivers/built-in.o: In function `hcd_init_fiq':
:(.text+0xec550): undefined reference to `claim_fiq'
:(.text+0xec5ec): undefined reference to `set_fiq_handler'
:(.text+0xec63c): undefined reference to `__set_fiq_regs'
:(.text+0xec6f4): undefined reference to `enable_fiq'
make: *** [vmlinux] Error 1
```

jdb2

pelwell, 2016-05-01:
I rewrote your commands slightly to be user-agnostic and to use the downloaded tools in preference of any others:
```
mkdir -p ~/src/rpi4
cd ~/src/rpi4
git clone https://github.com/raspberrypi/tools
PATH=$HOME/src/rpi4/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64:$HOME/src/rpi4/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin:$PATH
git clone --depth=1 https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel7
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
make -j 12 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs
```
(Yes, I also bumped the job count on the make to save time.)

These commands completed without errors.

What do you get from running (within the linux directory):
```
grep FIQ .config
ls arch/arm/kernel/fiq*
git status
```

jdb2, 2016-05-01:
Thank you for replying :)

As per your question, the output of the commands that you listed is : 

"grep FIQ .config" -- No output - perhaps this is the problem?
"ls arch/arm/kernel/fiq*" : arch/arm/kernel/fiqasm.S  arch/arm/kernel/fiq.c
The "git status" command output is quite lengthy and it follows : 

```
On branch rpi-4.4.y
Your branch is up-to-date with 'origin/rpi-4.4.y'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   Documentation/kernel-parameters.txt
	modified:   Documentation/sysrq.txt
	modified:   Makefile
	modified:   arch/Kconfig
	modified:   arch/arm/Kconfig
	modified:   arch/arm/include/asm/switch_to.h
	modified:   arch/arm/include/asm/thread_info.h
	modified:   arch/arm/kernel/asm-offsets.c
	modified:   arch/arm/kernel/entry-armv.S
	modified:   arch/arm/kernel/entry-common.S
	modified:   arch/arm/kernel/process.c
	modified:   arch/arm/kernel/signal.c
	modified:   arch/arm/kernel/smp.c
	modified:   arch/arm/kernel/unwind.c
	modified:   arch/arm/kvm/arm.c
	modified:   arch/arm/kvm/psci.c
	modified:   arch/arm/mach-at91/Kconfig
	modified:   arch/arm/mach-at91/at91rm9200.c
	modified:   arch/arm/mach-at91/at91sam9.c
	modified:   arch/arm/mach-at91/generic.h
	modified:   arch/arm/mach-at91/pm.c
	modified:   arch/arm/mach-at91/sama5.c
	modified:   arch/arm/mach-exynos/platsmp.c
	modified:   arch/arm/mach-hisi/platmcpm.c
	modified:   arch/arm/mach-imx/Kconfig
	modified:   arch/arm/mach-omap2/omap-smp.c
	modified:   arch/arm/mach-prima2/platsmp.c
	modified:   arch/arm/mach-qcom/platsmp.c
	modified:   arch/arm/mach-spear/platsmp.c
	modified:   arch/arm/mach-sti/platsmp.c
	modified:   arch/arm/mm/fault.c
	modified:   arch/arm/mm/highmem.c
	modified:   arch/arm/plat-versatile/platsmp.c
	modified:   arch/arm64/Kconfig
	modified:   arch/arm64/include/asm/thread_info.h
	modified:   arch/arm64/kernel/asm-offsets.c
	modified:   arch/arm64/kernel/entry.S
	modified:   arch/mips/Kconfig
	modified:   arch/mips/kvm/mips.c
	modified:   arch/powerpc/Kconfig
	modified:   arch/powerpc/include/asm/kvm_host.h
	modified:   arch/powerpc/include/asm/thread_info.h
	modified:   arch/powerpc/kernel/asm-offsets.c
	modified:   arch/powerpc/kernel/entry_32.S
	modified:   arch/powerpc/kernel/entry_64.S
	modified:   arch/powerpc/kernel/irq.c
	modified:   arch/powerpc/kernel/misc_32.S
	modified:   arch/powerpc/kernel/misc_64.S
	modified:   arch/powerpc/kvm/Kconfig
	modified:   arch/powerpc/kvm/book3s_hv.c
	modified:   arch/powerpc/platforms/ps3/device-init.c
	modified:   arch/s390/include/asm/kvm_host.h
	modified:   arch/s390/kvm/interrupt.c
	modified:   arch/sh/kernel/irq.c
[... 332 lines cut ...]
	include/linux/rwsem_rt.h
	include/linux/spinlock_rt.h
	include/linux/spinlock_types_nort.h
	include/linux/spinlock_types_raw.h
	include/linux/spinlock_types_rt.h
	include/linux/swait.h
	include/linux/work-simple.h
	include/trace/events/hist.h
	include/trace/events/latency_hist.h
	kernel/locking/rt.c
	kernel/sched/swait.c
	kernel/sched/work-simple.c
	kernel/trace/latency_hist.c
	localversion-rt

no changes added to commit (use "git add" and/or "git commit -a")
```

I'm going to run "make clean" and "make distclean" and try building the kernel ( again ) but this time with the .config file from my running Raspberry Pi 3 model B. First, though, I'm going to peruse anything to do with FIQ after a "make menuconfig" command.

Thanks for helping,

Regards,

jdb2

pelwell, 2016-05-01:
You should really start with seeing `git reset --hard HEAD` and `git clean -fd` get your git tree back into a clean state (run `git status` again to check). If not, then no amount of make cleaning will fix the problem.

jdb2, 2016-05-01:
Thanks for the advice :) I'm building the kernel using the default config as we speak. I'll close this issue if it succeeds.

Regards,

jdb2

jdb2, 2016-05-01:
I attempted to re-build the kernel with the following commands :

"git reset --hard HEAD"
"git clean -fd"
"git status" :

```
On branch rpi-4.4.y
Your branch is up-to-date with 'origin/rpi-4.4.y'.

It took 8.71 seconds to enumerate untracked files. 'status -uno'
may speed it up, but you have to be careful not to forget to add
new files yourself (see 'git help status').
nothing to commit, working directory clean
```

make -j 3 clean
make -j 3 distclean
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
make menuconfig # ( Changed the kernel preemption model to "Preemptible Kernel (Low-Latency Desktop)" )
make -j 3 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs
 This results in the same error as before :

```
  LD      init/built-in.o
drivers/built-in.o: In function `hcd_init_fiq':
:(.text+0xebb84): undefined reference to `claim_fiq'
:(.text+0xebc20): undefined reference to `set_fiq_handler'
:(.text+0xebc70): undefined reference to `__set_fiq_regs'
:(.text+0xebd28): undefined reference to `enable_fiq'
make: *** [vmlinux] Error 1
```

Regards,

jdb2

pelwell, 2016-05-01:
You need ARCH=arm in your menuconfig command, otherwise it will assume x86.

You also haven't said what happened when you ran with an unmodified bcm2709_defconfig. Running menuconfig to alter the configuration is a significant step and should have been mentioned in your list of build steps.

jdb2, 2016-05-02:
Thanks for pointing that out :)

It was my omission of the "ARCH=arm" in "make menuconfig" that was causing the problem, apparently.

I'm closing this issue.

Thanks

jdb2

closed 2016-05-02 by jdb2

Mpropre, 2017-03-27:
This helped me too. I think they should point that out in some tutorials. The "arch=ARM" is often clearly indicated to generate the basic configuration or to build the kernel, but not to configure the kernel.

pelwell, 2017-03-27:
It's already in the documentation: https://www.raspberrypi.org/documentation/linux/kernel/configuring.md

### USING MENUCONFIG

> Once you've got everything set up and ready to go, you can compile and run the  menuconfig utility as follows:

```
$ make menuconfig
```
> If you're cross-compiling,:

```
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- menuconfig
```