raspberrypi/linux #1448: IPv6 static address assignment with dhcpcd

stg39, 2016-05-01:
My Raspberry Pi B+ is currently running Raspbian GNU/Linux 8 (jessie) which I currently use as a DNS/NTP server for my IPv4 and IPv6 clients. I have configured /etc/dhcpcd.conf so that the Raspberry Pi should be allocated a static IPv4 address, an IPv6 Unique Local Unicast and Global Unicast address using the  "static ip6_address=" directive. 
However in order for this functionality to work with dhcpcd a version 6.10.2 or later is required.
Can dhcpcd be upgraded to an upto date version.

grigorig, 2016-05-01:
You should be able to simply use the Debian network configuration in /etc/network/interfaces for static addressing.

XECDesign, 2016-05-01:
Please note that this repo is for the linux kernel and is not related to dhcpcd.

The correct place to report things like this would be https://github.com/RPi-Distro/repo/issues

closed 2016-05-01 by XECDesign