raspberrypi/linux #1453: How does kernel 4.4 get entropy from `/dev/hwrng` ?

Mausy5043, 2016-05-03:
After upgrading from 3.18 to 4.4 I noticed that  `/dev/hwrng` exists but the module that I used to need to load in the past (bcm2708-rng) is not found. So, I'm guessing that this functionality is now compiled into the kernel and nolonger supplied as a module.

The question that I have is: does the kernel also use `/dev/hwrng` as a source of entropy or do I still need to use `/usr/sbin/rngd`  (provided by the package `rng-tools`) to feed the HWRNG source to the kernel?

popcornmix, 2016-05-03:
Probably best to ask these questions on the forum.
Yes, the hw rng module is now built in.
Yes, some scheme for moving data from /dev/hwrng to /dev/random is required (like rng-tools).

closed 2016-05-03 by popcornmix

Mausy5043, 2016-05-03:
forum?

popcornmix, 2016-05-03:
This github is for reporting kernel bugs.
The forum is best for "how to" discussions:
https://www.raspberrypi.org/forums