raspberrypi/linux #1455: 4.4.x does not load VC4

schnitzeltony, 2016-05-05:
Up to now I used 4.1.x with VC4 enabled and it worked quite nice. Now that I try to update to 4.4.8 it seems that vc4 drm isn't started: On boot with HDMI connected I see the four raspberries - not the rainbow and dmesg does not display anything regarding vc4. config.txt contains

dtoverlay=vc4-kms-v3d-overlay,cma-256
mask_gpu_interrupt0=0x400
avoid_warnings=2

As mentioned in the commit introducing CMA memory size I tried also

dtoverlay=vc4-kms-v3d,cma-256

but result was same. Any ideas?

pelwell, 2016-05-05:
You don't need the:
```
mask_gpu_interrupt0=0x400
avoid_warnings=2
```
but they won't stop the overlay from working. The "-overlay" would, but you've removed that now.

What does `sudo vcdbg log msg |& grep -v -E "(gpioman|HDMI|clock):"` say?

pelwell, 2016-05-05:
And `dmesg`?

pelwell, 2016-05-06:
By the way, it starts for me on 4.4.8.

schnitzeltony, 2016-05-06:
OK thanks for responses.

* forgot to mention: I am testing on Raspi 2
* I am following the yocto adventure so I build my images from sources and don't have vcdbg
* have now 4.4.9
* dmesg attached (4.4.9)

Interesting: When restarting by shutdown -r now I see 'rainbow' screen for short time and then for raspberries.

[dmesg.txt](https://github.com/raspberrypi/linux/files/253116/dmesg.txt)

schnitzeltony, 2016-05-06:
Ehm I meant four (not 'for') raspberries (as I saw before trying vc4)

pelwell, 2016-05-07:
How could you fail to mention it was a custom kernel in a non-Raspbian distribution? We don't really support that, but I'll have a go.

Is your firmware recent? What does /boot/.firmware_revision contain? And /boot/config.txt? And `ls /boot/overlays`?

ED6E0F17, 2016-05-08:
There is no CMA in your command line. You are going to have a hard time if your kernel overrides the firmware supplied commandline.

     Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=1024 bcm2709.boardrev=0xa01041 bcm2709.serial=0xd937f2cb smsc95xx.macaddr=B8:27g

chbae, 2016-05-15:
I thought that dtoverlay is not loaded. As ED6E0F17 said, CMA is not allocated.

See below message in your dmesg.txt.
[    0.000000] cma: Reserved 8 MiB at 0x3a800000

schnitzeltony, 2016-05-15:
On Sun, May 15, 2016 at 6:35 PM, Changhyeok Bae <notifications@github.com>
wrote:

> I thought that dtoverlay is not loaded. As ED6E0F17 said, CMA is not
> allocated.
>
> See below message in your dmesg.txt.
> [ 0.000000] cma: Reserved 8 MiB at 0x3a800000
>
> Thanks everybody and sorry for late response - have lots of daily coding
issues.

I am using recent firmware and thought by enabling vc4 devicetree with
'dtoverlay=vc4-kms-v3d,cma-256' the magic should happen.

By the way: The kernel I use is nothing special: it is build from unpatched
sources of the 4.4 tree found here.

So the next step I'll do is to install a raspian and see what's different.

chbae, 2016-05-19:
Please check your /boot/overlays. In there, vc4-kms-v3d.dtbo file should be existed. (not dtb file)

schnitzeltony, 2016-08-31:
1. Thanks for help
2. dt was not loaded (later firmware requires *.dtbo)

closed 2016-08-31 by schnitzeltony