raspberrypi/linux #1466: (RPI3) Pulling list of GATT services/characteristics the same on Bluez & Bleno

johnrussellernest, 2016-05-11:
I'm running BT on the internal BCM43xx, and get an LE connection fine from my iPhone running either Bluez or Bleno.
Currently running Bluez 5.39 from source tarball with Phil Elwell's bcm43xx patches applied to hciattach. 
Behavior is the same as it was when I ran Bluez 5.23-2+rpi1.
Behavior is also the same if I stop the bluetooth service and run Bleno.
GATT service will start and appear as though it loaded my custom services/characteristics (I even see this in the debug logs for bluetoothd that it's adding my custom services/characteristics).

However, it will appear as the following each time on the client:

Link Loss
Tx Power
Immediate Alert
Current Time Service
Reference Time Update Service
Phone Alert Service
Alert Notification Service

(Note: Current Time Service doesn't work now that I've upgraded to Bluez 5.39, but it's still listed when the central device (iPhone) scans for services/characteristics, even from a custom app I made in Xamarin.IOS to use CoreBluetooth directly. I don't care if it just doesn't work, I just want what should be there to be there and what shouldn't to be gone.)

In Bleno it says "serviceSet" was successful, debugged the files and it looks like it's setting those services, but every time I pull this is the list of services I get even though I'm trying to implement a custom GATT service with custom read/write characteristics.

Is this a D-Bus issue? Firmware issue? Write permissions issue? Not sure where to look, wasted a day looking through the Bluez source for clues but I couldn't find where the info is getting presented to the central device.

pelwell, 2016-05-11:
I'm pretty sure this isn't a kernel question, but I'd like to at least understand your question.

Can you please restate the issue, listing the steps and at each one saying what you expect to see and what you actually get (if it is different). If there is a point where BlueZ and Bleno diverge, also make that clear.

johnrussellernest, 2016-05-11:
I'm not really sure where it belongs but I figured out the issue this morning.

It would appear that IOS's CoreBluetooth API caches GATT services/characteristics on IOS9, and you have to do a full power cycle of the phone to discover new/removed services to a server after the first discover. Worked perfectly fine on Android. I hadn't power cycled my phone in weeks so it was the last thing I thought of during testing.

I thought originally it might be a bluez or kernel/firmware issue. I was mistaken.

Also Phil just noting that your patches to hciattach from 5.23-2+rpi1 have to be applied to all later versions in order for the internal BT driver to work on the RPI3. I could find little to nothing about this online without days of searching. Where's the best place for that info to be communicated?

pelwell, 2016-05-12:
That's hard to say. Creators of other distributions have approached me directly to find out, and with the Raspbian BlueZ sources available we are clearly not hiding anything, but should we go beyond that?

clivem, 2016-05-12:
> but should we go beyond that?

You mean, should you submit your patches to upstream bluez and get Broadcom to submit the firmware to the linux-firmware git repo? ;)

johnrussellernest, 2016-05-13:
Yes I think it would be good to submit it upstream. I'm fairly new to the Raspberry Pi (but have a Linux background) and spent a few days on this. This part is more a Raspbian issue than a Linux kernel one, but the default source repository in sources.list (goes to archive.org) gets 5.23_2 not 5.23_2-rpi1 from the release. It would be great if apt-get could be updated to get 5.39+rpi1. 

Not sure who the responsible parties are for updating what but it should get fixed.

closed 2017-05-18 by JamesH65