raspberrypi/linux #1467: i2c-gpio module has the problem of slow response

FishTest, 2016-05-11:
When I enable devictree i2c-gpio in /boot/config.txt , after run sudo i2cdetect -y 3 , the response is very slow , about 1 second each scan when display '--'.

pi@raspberrypi:~ $ sudo i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- ^Cpi@raspberrypi:~ $

pi@raspberrypi:~ $ cat /boot/config.txt | grep i2c-gpio
dtoverlay=i2c-gpio

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.9-v7+ #884 SMP Fri May 6 17:28:59 BST 2016 armv7l GNU/Linux

What could be the causes ?

6by9, 2016-05-11:
Have you left the SCL line either floating or with a pull down? If so then the driver will be doing clock stretching on every bit.
Looking at https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/i2c/busses/i2c-gpio.c#L208 there appears to be a 100ms per bit timeout, so 1sec total sounds plausible for each transaction from i2c-detect.

6by9, 2016-05-11:
30 second test doing as you reported.

Add the dtoverlay - defaults to using GPIOs 23&24. i2cdetect -y 3 takes a long time.

```
sudo raspi-gpio set 23 pu
sudo raspi-gpio set 23 pu
```
and i2cdetect -y 3 completes very quickly with all dashes. So my guess looks to be correct - you have no pullups.
Do not rely on the SoC internal pullups if you want reliable I2C - use external ones.

FishTest, 2016-05-12:
Thank you very much!
everything go back after 
pi@raspberrypi:~ $ sudo raspi-gpio set 23 pu
pi@raspberrypi:~ $ sudo raspi-gpio set 24 pu
pi@raspberrypi:~ $ sudo i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberrypi:~ $

closed 2016-05-12 by FishTest