raspberrypi/linux #1473: Feature request - overlay for remapping audio to GPIO pins

stefandz, 2016-05-13:
Hello - this is a follow-up to me showing my ignorance on the [forums](https://www.raspberrypi.org/forums/viewtopic.php?p=973756#p973756) (and PhilE - who I imagine is @pelwell here) suggesting that I post a request here.

I am trying to remap the PWM audio from the pins connected to the onboard 3.5mm socket to pins 32 and 33 on the GPIO socket. So far I was achieving this by adding the following to /boot/config.txt

`dtoverlay=pwm-2chan,pin=12,pin2=13,func=4,func2=4`

However, as Phil points out this loads the driver twice (I think) and certainly also leaves audio coming out of the 3.5mm socket. Would it be possible to ask for someone's help in writing an overlay that actually remaps the audio properly? (and that the user can disable if they would like audio back in the normal place)

All help gratefully received.

P33M, 2016-06-03:
The preferred method is to rewrite the relevant section in dt-blob.bin (which is primarily Videocore-based GPIO setup on a per-model basis).

The source DTS is here: https://github.com/raspberrypi/documentation/blob/master/configuration/images/dt-blob.dts and it's simple enough to change pin 40/45 to 12/13.

pelwell, 2016-06-03:
I woudn't say it was preferred - the dt-blob is a monolith monstrosity. Edit once and forever have to keep it up-to-date.

All you have to do is choose a DT node that doesn't have an existing pinctrl section and add one - the "audio" node is the obvious one.

It would look something like this (untested):
```
/dts-v1/;
/plugin/;

/ {
        compatible = "brcm,bcm2708";

        fragment@0 {
                target = <&gpio>;
                __overlay__ {
                        audio_pins: audio_pins {
                                brcm,pins = <40 45 12 13>;
                                brcm,function = <0 0 4 4>; /* in in alt0 alt0 */
                        };
                };
        };

        fragment@1 {
                target = <&audio>;
                __overlay__ {
                        pinctrl-name = "default";
                        pinctrl-0 = <&audio_pins>;
                };
        };
};
```

stefandz, 2016-06-14:
Thanks for this @pelwell (sorry for the delay in replying!). Am I correct in thinking that I would use the code snippet as a .dts file, before compiling it to a .dtb (with the -@ argument to allow unresolved references), and then getting it onto the `/boot` partition, along with adding `dtoverlay=name-of-device` to `/boot/config.txt` - is that everything? Once I have this debugged I can obviously submit a PR to have it included in future releases of Raspberry Pi Linux I guess, which should filter down to Raspbian.

If I am getting any of this drastically wrong please let me know!

pelwell, 2016-06-16:
@stefandz That looks broadly right, with the proviso that different models use different pins for PWM audio. Having said that, I haven't got it to work yet. For reasons I haven't managed to figure out the pinctrl section is being ignored or at least not taking effect. I'll report back when I make some progress.

karrika, 2016-06-17:
I have a tested setup for mapping the audio pins to 12, 13. There is also step by step instructions for creating the dt blob and programming it into an eeprom so it works as soon as the HAT is put on top of the RPi. See at the end of the page in [http://discohat.com/miscap/](url)

pelwell, 2016-06-17:
Thanks for the suggestion. By the looks of it I'd say that works almost by accident. What you are doing is enabling the PWM driver, which is not the same as the PWM audio driver - just basic perod+duty cycle stuff. But by getting it to request the PWM function on pins 12 & 13 you are achieving the objective. What I'm trying to do is get the pins assigned to the audio node.

I'm also concerned that your overlay leaves the PWM functions multiply mapped, although it appears to be working for you. Once I've got pins allocated for the audio node I'll get the standard DTBs to request the per-platform standard pins, then all an overlay would have to do is overwrite those pin assignments.

karrika, 2016-06-17:
Thank you for the feedback. As a normal user I did not understand the difference between pwm and audio node. I have to change the src to use audio node instead. Perhaps later today when I get home. It would be nice to do things correctly.

stefandz, 2016-06-17:
Thank you everyone for your valuable (and generous) time contributions to this. If there's anything I can be doing to help, please let me know! BTW, I really like your HAT, @karrika 👍

pelwell, 2016-06-19:
Duh! It really helps if you call the property `pinctrl-names` (note the plural); with that change, it works as expected. I've edited the post above to correct it.

I'll have a think about the best way to make this work on all platforms. My current proposal is:
1) Add audio_pins to the base DTBs.
2) Add parameters to modify them.
3) Get the firmware to use those parameters to select the pins.
4) Make the firmware use the DT audio_pins values when setting the initial pin functions. The intention is make the system act as if the audio pins were inputs before the kernel starts (i.e. no need to restore the usual pins to inputs), but still set the pin functions in the firmware in case other OSs rely on it.

pelwell, 2016-07-11:
The firmware change for this went in at the end of last month. I've now pushed the DTB changes that define the audio_pins node for all the Pi models with audio outputs, to be picked up by the next build.

With these in place, the overlay required to remap the output to 12 & 13 now looks like this:
```
/dts-v1/;
/plugin/;

/ {
        compatible = "brcm,bcm2708";

        fragment@0 {
                target = <&audio_pins>;
                __overlay__ {
                        brcm,pins = <12 13>;
                        brcm,function = <4>; /* alt0 */
                };
        };
};
```
Note that you can also disable the PWM output with an empty "brcm.pins" property.

stefandz, 2016-07-14:
Thanks very much for this @pelwell (and everyone else involved). I will try testing this out today on as many platforms as I can get my hands on.

Ruffio, 2016-08-08:
@stefandz as a fix for your issue has been implemented, please try a firmware upgrade and confirm that is has been resolved by closing this issue. Thanks.

stefandz, 2016-08-09:
@Ruffio will try to test this during the week - sorry for the delay at this end

stefandz, 2016-08-10:
Tested and confirmed working on Raspberry Pi 2 B and Raspberry Pi 3. On the Zero, however, the audio channels are reversed - this is true for both versions 1.2 and 1.3. What is the recommended way of allowing for this - something within the overlay or loading different overlays for different models?

I don't have access to a Pi 1 A+ or B+ for testing - should I expect those to behave like the Zero or the 2/3? Also, I am using a single .dtbo file for all testing, compiled on a Pi3 - is this best practice?

Thank you for all your help on this so far, and apologies that it has taken me so long to get testing finished.

pelwell, 2016-08-10:
On early Pis (Models A & B), onboard audio uses PWM0 to drive the left channel of the audio jack and PWM1 to drive the right. With the appearance of the B+ this mapping got reversed (essentially by accident), but the firmware has the ability to swap the channels over so the end result is correct. This swapping is controlled by the pin allocations as read from the dt-blob, either an external dt-blob.bin or the built-in default version, but PiZero normally has no audio jack so there is no swap.

When you use the overlay, pins 12 and 13 are mapped to PWM0 and PWM0 respectively, so audio left should always be sent to 12 and right to 13 for all models EXCEPT that the above swapping is still applied because the firmware doesn't notice that it is no longer needed. The end result is that A+, B+, 2B and 3B have right on 12 and left on 13, while the other Pis have the channels the right way round.

I'll have a think about a fix.

stefandz, 2016-08-10:
Thanks for the explanation, @pelwell. I am just as happy to make the fix two overlays, one for the Zero and one for all other Pis if that helps (although from what you're saying, that might get broken in future updates).

pelwell, 2016-08-10:
That's actually not an option because you can't map PWM0 to pin 13 and PWM1 to 12. You could use a custom dt-blob.bin, but that can lead to problems with future releases so is best avoided if possible.

stefandz, 2016-08-10:
Ah sorry, my misunderstanding. I feel this might be useful to resolve for others who wish to have audio out on the Zero. Do let me know if I can help testing (I promise to be on it quicker this time around - although I am out on holiday for the last two weeks of August).

stefandz, 2016-08-18:
Since using this overlay gives the same L-R swapping as my cludgy bit of `/boot/config.txt` `dtoverlay=...` nonsense and we're about to release a Debian package for our product, I feel I should get the overlay working for now (with a user note to swap L-R some other way as a workaround). 

What's the best way to do this so that the overlay is automatically included if I (as a user) run `sudo apt-get dist-upgrade`? Or is it equally good form to put the compiled overlay inside the Debian package - and if so, where within the package should it live?

pelwell, 2016-08-18:
The rpi-update firmware released on the 16th includes a fix for the L-R swap on 12 & 13 ("firmware: platform: Don't swap audio L&R if using GPIOs 12&13") - see if it works for you.

I can't help you with Debian packaging matters - @XECDesign is our expert. We have always been very wary of editing config.txt during upgrades, but appending a line if it doesn't exist is about as safe as it gets.

stefandz, 2016-08-18:
Thanks @pelwell - I shall ask @XECDesign. I wouldn't want to goof around with config.txt during an upgrade - merely ensure that the correct overlay is part of the latest upgrade / release and then appending a line if it doesn't exist as part of the Debian package postinst. I shall also test the L-R swap now - thanks again for everything.

stefandz, 2016-08-18:
OK @pelwell - strangely now when I add this overlay to `/boot/config.txt` I get no audio out at all. Pretty sure everything is correctly written / located / compiled etc - using your snippet from [here](https://github.com/raspberrypi/linux/issues/1473#issuecomment-231711087) named as `picap-overlay.dts` and then compiling it to `picap.dtbo` using `dtc -@ -I dts -O dtb -o picap.dtbo picap-overlay.dts`. Going back to the example in my first comment works (albeit with audio reversed on the Zero). Am I doing something silly here?

pelwell, 2016-08-18:
Are you on the latest rpi-update kernel? I've just tested this on a Pi3 and got exactly the behaviour I expected:
1) Without the overlay (and provided the audio jack is selected using raspi-config), audio plays from the audio jack with correct channel assignment.
2) Without the overlay (and provided the audio jack is selected using raspi-config), and using raspi-gpio to select function a0 on 12 & 13, the left channel is sent to pin 13 and the right to pin 12.
3) Rebooting with the overlay (shown above) loaded, audio plays with the left channel on pin 12 and the right on pin 13.
4) With the overlay loaded and raspi-gpio used to select a0 on 40 & 41 (for a Pi3),  audio plays from the jack with channels swapped.

karrika, 2016-08-19:
I just made a Miscap yesterday, updated the software to the latest version and implemented the audio_pins instead of pwm pins. The dt-blob worked nicely. The audio was activated automatically and the channels were correct when putting Miscap on RPi2. On a PiZero the channels were swapped. The sound was ok. I will shortly update the example on discohat.com/miscap but I wonder if there is something I could add to the eeprom to get the channels correct on the PiZero?

pelwell, 2016-08-19:
Do you mean dt-blob or overlay? If you methodically go through the steps you have taken, with details of pin numbers etc., and what you have observed then I may be able to help. However, the current rpi-update image firmware with the overlay above gives the correct channel assigments to 12 & 13 on a Pi3 and a PiZero, and I would expect it to also work on the other Pi models.

pelwell, 2016-08-19:
@stefandz XECDesign forwarded your email - sorry, I misunderstood your question.

Since other applications may require the same functionality, so I've added an audremap overlay to the kernel tree, which will automatically be included in future releases.

stefandz, 2016-08-19:
Hi again @pelwell - spent a bit of time testing this. I have updated to latest firmware using rpi-update. I started with a blank microSD, put Jessie 2016-05-27 on it, then apt-get update, dist-upgrade, reboot, rpi-update, reboot. At this stage I placed the card into a Pi3. Channel test command is `speaker-test -c 2 -t wav`

1. With configuration as above, audio is output on correct channels on the Pi socket
2. After calling `raspi-gpio set 12 a0; raspi-gpio set 13 a0` left channel outputs on pin 13, right on pin 12 - I would consider this "correct". Audio is also still audible (with correct channel mapping) on the Pi socket.
3. Loading the overlay, no audio is output on the Pi socket, and left channel outputs on pin 12, right on pin 13 - I would consider this reversed.

Taking this same microSD card out of the Pi 3 and placing it in a Pi Zero results in

1. With no raspi-gpio calls or overlays, no audio is output on pins 12 or 13.
2. After calling `raspi-gpio set 12 a0; raspi-gpio set 13 a0` left channel outputs on pin 12, right on pin 13 - I would consider this "reversed".
3. Loading the overlay, left channel outputs on pin 12, right on pin 13 - I would consider this "reversed".

To my best understanding here, the overlay / its interaction with lower down software is functioning to make behaviour identical between the Pi 3 and the Pi 0, but the identical behaviour is the opposite of what I would expect. Is it possible to create an overlay / firmware change that causes left on 13, right on 12 behaviour?

pelwell, 2016-08-19:
PWM0 is considered to be the left channel, and PWM1 the right. This is how boards before the B+ were wired, and it fits with the way the PWM hardware deals with interleaved data: by convention the first sample in a pair is for the left channel, and sends the first sample to PWM0.

With the B+ the pins got swapped, requiring software mod to restore the correct channel mapping. Subsequent boards have stuck to that wiring configuration.

Since the positions of PWM0 and PWM1 on the header is fixed, I wanted to standardise on the correct ordering - L -> 12 and R -> 13, but this does lead to the situation you have seen. Plus, it means that if headphone output was enabled at the same time (is this part of your use case?) then one would be swapped with respect to the other.

Although I don't like it, I now think we have to standardise on what I would consider the "reversed" mapping, i.e. R ->12, L -> 13, to match the headphone output. This has the happy side effect of meaning your boards won't have to be scrapped. At the same time I'll add some DT properties and overlay parameters to allow the swapped-ness to be overridden (this will affect all outputs), and to control whether the headphone output is active or not (it will default to being disabled).

stefandz, 2016-08-19:
This sounds great - and I agree it would be preferable to have standardised on PWM0 -> L and PWM1 -> R (from a hardware perspective) but I can understand how we got to where we are.

Bringing the two board types into line (with the overlay) and then allowing a global swappedness (affecting all outputs equally) seems like a nice feature-add to me. Please do let me know when there's a version of this that I can test using a system with the latest `rpi-update` - I am guessing that the reversal parameter documentation will be in `/boot/overlays/README`.

Thanks so much for looking at this, Phil - it really is appreciated!

pelwell, 2016-08-19:
The updated overlay has been pushed, but it won't do anything until the firmware is updated to match - probably in the next few days. The README says:
```
Name:   audremap
Info:   Switches PWM sound output to pins 12 (Right) & 13 (Left)
Load:   dtoverlay=audremap,<param>=<val>
Params: swap_lr                 Reverse the channel allocation, which will also
                                swap the audio jack outputs (default off)
        enable_jack             Don't switch off the audio jack output
                                (default off)
```

stefandz, 2016-08-19:
Perfect, thank you so much! I am out of the office now for two weeks, but I shall test it on my return as a priority and then hopefully close the issue. Have a great weekend!

popcornmix, 2016-08-19:
latest ``rpi-update`` firmware has this in

stefandz, 2016-08-19:
Thanks @popcornmix - sadly I am now out! Didn't take a Pi on holiday with me purposefully! Will make this first order of business on the 5th of September though. Thanks for the heads up.

SylvainGarrigues, 2016-08-21:
Hi, excuse my ignorance: would what is being talked about here enable me to have sound out of a stereo jack soldered on some GPIO pins of my Raspberry Pi Zero?

karrika, 2016-08-22:
Yes. You need a few resistors and capacitors to make it happen, It works fine on PiZero. I try to keep [http://discohat.com/miscap](url) updated with the steps required to get the sound working on PiZero. It helps if you also have an eeprom for storing this info for assigning sound to the gpio pins.

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65