raspberrypi/linux #1475: Pi3 hang when attaching network

spali, 2016-05-16:
Hi

Just bought two Pi3.
Tried "2016-05-10-raspbian-jessie-lite.img" and even the full image. Everytime I connect the network cable the Pi3 freezes after a couple of seconds.
First I thought it must be the power supply, but I'm absolutely sure it has nothing to do with the power, tried Pi PoE Switch, USB 1.5A adapter and even a labor power supply with limit set to 2A. Measuring the current during usage didn't show a voltage drop.
I can reproduce this on the second Pi3 ... exactly the same behavior.

I tried to watch kernel messages during attaching network, it shows eth0 up and then some seconds later the Pi freezes without any additional message.

I tried the image from march "2016-03-18-raspbian-jessie-lite.img". First I thought it works with them, could start pi, login and connect the network cable. ping to gateway works for some time.
Sometimes after seconds sometimes after minutes. Sometimes starts failing with "No buffer space available" during pinging and then it freezes after a couple of these messages.

If I boot with already attached network, it almost instantly freeze as soon as the login prompt is shown.

Any idea or what can I do to isolate the problem?

pelwell, 2016-05-16:
That kind of lock-up without warning is almost always an indication of a power supply fault. We recommend a 2.5A power supply for Pi3, especially with other peripherals attached. Keep an eye on the power LED when you connect the cable - if it flickers off then the firmware has detected an under-voltage situation.

Is it possible that the ethernet cable or the hub/switch has a fault? Try changing cables and ports to see if it makes a difference.

popcornmix, 2016-05-16:
After the issue what does:
``vcgencmd get_throttled`` report?

spali, 2016-05-16:
@popcornmix 
returned 0

@pelwell 
You was right... a couple of times during boot, the led flickers. And it looks like too before getting freeze. Also I think even the device has enough power, there seems to be an instability with my power sources.
I updated the firmware and tried some other power sources and it seems to be a bit more stable.

close this issue, I think I can solve it by myself. Thought I eliminated the power stuff, but it seems to be the problem.

closed 2016-05-16 by spali