raspberrypi/linux #1478: 7" touch screen with drm (vc4)

chbae, 2016-05-19:
I can use "1920x1080" touch monitor with vc drm and drm is initialized successfully.

However, When I use 7" touch screen (official raspberrypi monitor), I didn't get the "drm initialized" message in dmesg and I got the below errors.

Do you have any idea?

`[   11.314518] [drm] No driver support for vblank timestamp query.
[   11.327805] vc4-drm soc:gpu@7e4c0000: No connectors reported connected with modes
[   11.327835] [drm] Cannot find any crtc or sizes - going 1024x768
[   11.445197] vc4-drm soc:gpu@7e4c0000: fb0:  frame buffer device`

chbae, 2016-05-19:
I thought that current official kernel (4.4) for raspberrypi doesn't support DSI right?

popcornmix, 2016-05-20:
No DSI/DPI display support in the raspberrypi kernel with vc4-kms-drm overlay enabled. 
We'll get updates from @anhol when they are stable.
For now bug reports or feature requests are best reported here:
https://github.com/anholt/linux/issues

closed 2016-05-20 by popcornmix

chbae, 2016-05-24:
@popcornmix Ok. Thanks.

tom--pollard, 2016-08-08:
Hi, do we know if this support was merged yet to 4.4.y ?

popcornmix, 2016-08-08:
Keep an eye on https://github.com/raspberrypi/linux/pull/1556