raspberrypi/linux #1481: Rpi2 is hungs up after hot plug usb cabel ( Kernel Panic Unable to handle kernel paging request at virtual address)

olllivier, 2016-05-20:
Hi,

Short description:
I have problem with my environment to remote control (redrat3 vII). Redrat is connected to RPi2 via external usb power hub Dlink.

logs after hungs up:
[ 300.430390] Unable to handle kernel paging request at virtual address 2f815486
Remote Transceiv[ 300.438962] pgd = a1374000
er (112a:0005) a[ 300.442985] [2f815486] *pgd=00000000s /devices/platf
orm/soc/3f980000.[ 300.449331] Internal error: Oops: 5 [#1] SMP ARM
usb/usb1/1-1/1-1.[ 300.455386] Modules linked in:3/1-1.3.5/1-1.3.5 uinput.4/1-1.3.5.4:1.0 evdev/rc/rc0
[ ir_xmp_decoder +0.003054] ir_lirc_codecinput: ir_sanyo_decoderMCE IR Keyboard/ ir_sharp_decoderMouse (redrat3) ir_mce_kbd_decoderas /d evices/virt ir_sony_decoderual/input/input4 ir_jvc_decoder
[ +0.004 ir_rc6_decoder804] rc ir_rc5_decoder rc0: lirc_d ir_nec_decoderev: driver ir-li rc_hauppaugerc-codec (redrat redrat33) registered at ftdi_sio minor = 1
l2303m[ +0.021245] sbserial[0mUnab sgle to handle ker i2c_bcm2708nel paging reque bcm2835_gpiomemst at virtual ad spi_bcm2835dress 2f815486 cm2835_wdt0m
[ +0.0 lirc_rpi(C)08572] io_pdrv_genirq[31mpgd = a13740 uio00
[ usbtv+0.004023] ideobuf2_v4l2[7m[2f81548 v4l2_common6] *pgd=00000000 videobuf2_vmalloc
[ +0 videobuf2_memops.006346] ideobuf2_core3mInternal error videodev: Oops: 5 [# media1] SMP ARM
lirc_dev rc_core snd_bcm2835 snd_pcm snd_timer snd fuse
[ 300.546750] CPU: 3 PID: 3699 Comm: modprobe Tainted: G C 4.4.11-v7+ #887
[ 300.554644] Hardware name: BCM2709
[ 300.558076] task: a125b3c0 ti: a139c000 task.ti: a139c000
[ 300.563526] PC is at ext4_htree_store_dirent+0xa4/0x11c
[ 300.568794] LR is at 0x666e6f63
[ 300.571962] pc : [<801dd0ec>] lr : [<666e6f63>] psr: 20000013
[ 300.571962] sp : a139dd90 ip : 00000000 fp : a139ddb4
[ 300.583533] r10: afc28ff8 r9 : afc28018 r8 : b0e28400
[ 300.588800] r7 : 83633c71 r6 : b0e28400 r5 : 2f81548e r4 : 2f81548e
[ 300.595380] r3 : b0e28400 r2 : b0e2840c r1 : 00000000 r0 : b0e2841e
[ 300.601961] Flags: nzCv IRQs on FIQs on Mode SVC_32 ISA ARM Segment user
[ 300.609154] Control: 10c5387d Table: 2137406a DAC: 00000055
[ 300.614947] Process modprobe (pid: 3699, stack limit = 0xa139c210)
[ 300.621177] Stack: (0xa139dd90 to 0xa139e000)
[ 300.625574] dd80: afc28018 b5664620 a139de40 afc28020
[ 300.633822] dda0: b5640800 00000000 a139de0c a139ddb8 801ee3c0 801dd054 a139ddd8 b5640800
[ 300.642069] ddc0: afc28000 00001000 00000018 a139ddd8 a10d1300 00000003 afc28020 0000000c
[ 300.650318] dde0: b670eff0 b0e28400 b5664620 00000000 a10d1300 0000000c 00000000 b5664620
[ 300.658565] de00: a139de94 a139de10 801ef0d0 801ee248 00000000 00000000 8012e798 8001e55c
[ 300.666813] de20: 80164970 80160bc4 a139de40 b0e28420 203d90ad 00000000 b2494015 7efa89f0
[ 300.675062] de40: 2f81548e 83633c71 00000001 b5b4209c 8014864c 801486bc 00000000 b5802800
[ 300.683309] de60: 00000043 0000630a a139de94 b0e28400 b5664620 a139df60 b5664620 0000000c
[ 300.691557] de80: 00000000 a10d1300 a139df24 a139de98 801dcdfc 801ef04c a139defc a139dea8
[ 300.699804] dea0: 805b8b68 800692a0 80166a34 8016098c b5b37610 b2c7eb28 203d90ad 0000000a
[ 300.708052] dec0: b2494015 00000000 00000800 00000000 b5664620 80865584 b5b41c00 805b89c8
[ 300.716301] dee0: 54da729c a139dfb0 00008020 000001ff a139dfac a139df00 800091e8 a10d1300
[ 300.724548] df00: 00000000 a10d1308 b5664690 a139df60 a10d1300 00000000 a139df5c a139df28
[ 300.732796] df20: 80169570 801dc7f8 a139df4c a139df38 80165d18 54d9f278 00008000 54d9f278
[ 300.741042] df40: 000000d9 a10d1300 a10d1300 00000000 a139dfa4 a139df60 80169bd4 801694f0
[ 300.749290] df60: 80169860 00000000 00000000 00000000 54d9f298 00000000 00008000 00000000
[ 300.757541] df80: 54d9f278 00000000 54d9f278 000000d9 8000fd08 a139c000 00000000 a139dfa8
[ 300.765793] dfa0: 8000fb40 80169b58 54d9f278 00000000 00000000 54d9f298 00008000 00000000
[ 300.774044] dfc0: 54d9f278 00000000 54d9f278 000000d9 00000002 54d9f298 54b5c224 54b49574
[ 300.782294] dfe0: 54d9f298 7efa8920 76e99c34 76e99b50 60000010 00000000 00000000 00000000
[ 300.790562] <801dd0ec> from <801ee3c0>
[ 300.800228] <801ee3c0> from <801ef0d0>
[ 300.809541] <801ef0d0> from <801dcdfc>
[ 300.818057] <801dcdfc> from <80169570>
[ 300.825696] <80169570> from <80169bd4>
[ 300.833425] <80169bd4> from <8000fb40>
[ 300.841502] Code: e1a0c004 e5934000 e3540000 0a00000c (e5143008)
[ 300.847747] ---[ end trace c3d05cb8edff7379 ]---
[ +0.006055] Module[ 300.853888] Unable to handle kernel NULL pointer dereference at virtual address 00000001
s linked in 300.857428] Unable to handle kernel paging request at virtual address 2f815496
[ 300.857434] pgd = 80004000
[ 300.857441] [2f815496] *pgd=00000000
[ 300.857450] Internal error: Oops: 5 [#2] SMP ARM
[ 300.857526] Modules linked in: uinput evdev ir_xmp_decoder ir_lirc_codec ir_sanyo_decoder ir_sharp_decoder ir_mce_kbd_decoder ir_sony_dec oder ir_jvc_decoder ir_rc6_decoder ir_rc5_decoder ir_nec_decoder rc_hauppauge redrat3 ftdi_sio pl2303 usbserial sg i2c_bcm2708 bcm2835_gpiom em spi_bcm2835 bcm2835_wdt lirc_rpi(C) uio_pdrv_genirq uio usbtv videobuf2_v4l2 v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_cor e videodev media lirc_dev rc_core snd_bcm2835 snd_pcm snd_timer snd fuse
1m: uinput evdev ir_xmp_decode[ 300.857537] CPU: 1 PID: 3699 Comm: modprobe Tainted: G D C 4.4.11-v7+ #887
[ 300.857540] Hardware name: BCM2709
[ 300.857544] task: a125b3c0 ti: a139c000 task.ti: a139c000
[ 300.857560] PC is at rb_first_postorder+0x18/0x44
[... 196 lines cut ...]
[ 308.502644] Exception stack(0xb593bf48 to 0xb593bf90)
[ 308.536220] bf40: 00000000 b60003c8 00000000 00000000 b593a000 808605dc
[ 308.573088] bf60: 10c0387d 80860580 805bce5c 808c4718 00000000 b593bfa4 808614f8 b593bf98
[ 308.610388] bf80: 80010b34 80010b38 60000013 ffffffff
[ 308.644279] <805b8384> from <80010b38>
[ 308.680694] <80010b38> from <80063da8>
[ 308.718025] <80063da8> from <80063fd0>
[ 308.755978] <80063fd0> from <80015c88>
[ 308.794474] <80015c88> from <000095ac>
[ 309.321172] SMP: failed to stop secondary CPUs
[ 309.355122] ---[ end Kernel panic - not syncing: Fatal exception in interrupt

Env cmdline.txt:
"logo.nologo dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait rootdelay=5 dwc_otg.fiq_fsm_mask=0x7"

without "dwc_otg.fiq_fsm_mask=0x7"- issue is still occured.

Kernel version:
"Linux raspberrypi 4.4.11-v7+ #887 SMP Thu May 19 16:24:03 BST 2016 armv7l GNU/Linux"
additional info- on previous kernel version this issue also occures (from 4.0.7+).

thanks for your help !

Br,
Ollivier

olllivier, 2016-05-20:
*Reproduce this issue: 100 %(always)

P33M, 2016-05-20:
`[ 300.546750] CPU: 3 PID: 3699 Comm: modprobe Tainted: G C 4.4.11-v7+ #887`

Why is the kernel tainted with a broken module? As far as I can tell, standard Raspberry Pi kernels do not ship any modules that taint the kernel.

Does the pi crash on hot-*plug* or hot-*unplug*? If it's on hot-unplug, then please post the output of lsmod.

popcornmix, 2016-05-20:
The lack of function names in backtrace is also suspicious.
@olllivier If this meant to be an untouched kernel and set of modules from raspbian?
Or have you built your own module, or downloaded one from elsewhere?

olllivier, 2016-05-20:
Thanks for your quick feedback on my issue ! bellow deatils:
@P33M On my previous Kernel version build module for V4L2loopback. Otherwise all modules is from raspbian. But before build module for V4L2loopback this issue also occure.
Crash is after hot-unplug (lsmod I will be able to add near Monday).
@popcornmix On my previous Kernel version build module for V4L2loopback. All additioanls env i always download from jessy repo. 

Additional infromation:
1.I was perform this test on oryginal raspbian jessy sw from raspberry.org. Problem is also occures. 
2. This problem is not occure in case when: (actual i have default for lirc0 and lirc1 for redrat device). When i changed lirc1 to lirc0 and next perform hot-plug, problem(few times) is not still occure.

popcornmix, 2016-05-20:
Can you post kernel dmesg log when using a clean raspbian Jessie image?
Hopefully that will have a usable backtrace after the panic.

olllivier, 2016-05-23:
@P33M Sorry for the late in my response. This my lsmod:
"Module                  Size  Used by
uinput                  7454  1
evdev                  11396  3
ir_xmp_decoder          2903  0
ir_lirc_codec           4863  3
ir_mce_kbd_decoder      3921  0
ir_sharp_decoder        2177  0
ir_sony_decoder         2175  0
ir_sanyo_decoder        2433  0
ir_rc6_decoder          2813  0
ir_jvc_decoder          2173  0
ir_rc5_decoder          2493  0
ir_nec_decoder          2621  0
rc_hauppauge            2161  0
redrat3                 8531  0
pl2303                  8393  0
ftdi_sio               31193  0
usbserial              22115  2 pl2303,ftdi_sio
sg                     18319  0
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
lirc_rpi                6478  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
usbtv                  10968  0
videobuf2_v4l2         16428  1 usbtv
v4l2_common             3929  2 usbtv,videobuf2_v4l2
videobuf2_vmalloc       5187  1 usbtv
videobuf2_memops        1268  1 videobuf2_vmalloc
videobuf2_core         20861  2 usbtv,videobuf2_v4l2
videodev              132913  4 usbtv,v4l2_common,videobuf2_core,videobuf2_v4l2
media                  11573  1 videodev
lirc_dev                8310  2 lirc_rpi,ir_lirc_codec
rc_core                16468  14 ir_sharp_decoder,ir_xmp_decoder,lirc_dev,redrat3,ir_lirc_codec,rc_hauppauge,ir_rc5_decoder,ir_nec_decoder,ir_sony_decoder,ir_mce_kbd_decoder,ir_jvc_decoder,ir_rc6_decoder,ir_sanyo_decoder
snd_bcm2835            20511  0
snd_pcm                75698  2 snd_bcm2835,usbtv
snd_timer              19160  1 snd_pcm
snd                    51844  4 snd_bcm2835,usbtv,snd_timer,snd_pcm
fuse                   83461  1"
@popcornmix I will try send log from dmesg to near Wednesday.

olllivier, 2016-05-24:
Logs from clear raspbian jessy (+dmesg):
"[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.9-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #884 SMP Fri May 6 17:28:59 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 808c0d80, node_mem_map b9fa6000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b9f63000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2709.boardrev=0x2a01041 bcm2709.serial=0xbb0c017a smsc95xx.macaddr=B8:27:EB:0C:01:7A bcm2708_fb.fbswap=1 bcm2709.uart_clock=3000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939088K/966656K available (6340K kernel code, 432K rwdata, 1712K rodata, 480K init, 764K bss, 19376K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807e5330   (8053 kB)
      .init : 0x807e6000 - 0x8085e000   ( 480 kB)
      .data : 0x8085e000 - 0x808ca088   ( 433 kB)
       .bss : 0x808cd000 - 0x8098c1ac   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000012] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000032] Switching to timer-based delay loop, resolution 52ns
[    0.000349] Console: colour dummy device 80x30
[    0.001558] console [tty1] enabled
[    0.001612] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001689] pid_max: default: 32768 minimum: 301
[    0.002101] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002153] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003361] Disabling cpuset control group subsystem
[    0.003443] Initializing cgroup subsys io
[    0.003504] Initializing cgroup subsys memory
[    0.003583] Initializing cgroup subsys devices
[    0.003634] Initializing cgroup subsys freezer
[    0.003686] Initializing cgroup subsys net_cls
[    0.003788] CPU: Testing write buffer coherency: ok
[    0.003900] ftrace: allocating 21193 entries in 63 pages
[    0.056317] CPU0: update cpu_capacity 1024
[    0.056399] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.056439] [bcm2709_smp_prepare_cpus] enter
[    0.056591] Setting up static identity map for 0x8240 - 0x8274
[    0.058991] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.059423] [bcm2709_secondary_init] enter cpu:1
[    0.059482] CPU1: update cpu_capacity 1024
[    0.059490] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.060062] [bcm2709_boot_secondary] cpu:2 started (0) 17
[    0.060414] [bcm2709_secondary_init] enter cpu:2
[    0.060447] CPU2: update cpu_capacity 1024
[... 685 lines cut ...]
 kernel:[  135.284756] 1f00: 573270f0 2c8dfaff 00000253 00000000 5625756c 76fdf068 01200011 76f88000

Message from syslogd@raspberrypi at May 24 14:46:53 ...
 kernel:[  135.298451] 1f20: 00000000 8000fd08 00000000 00000000 b8891f8c b8891f40 80024520 80022d60

Message from syslogd@raspberrypi at May 24 14:46:53 ...
 kernel:[  135.312144] 1f40: 00000000 00000000 00000000 000081ed 00000001 00000000 00000000 00000000

Message from syslogd@raspberrypi at May 24 14:46:53 ...
 kernel:[  135.325841] 1f60: 0002a624 76fdf068 7ef7b9a0 76f88000 00000078 8000fd08 b8890000 00000000

Message from syslogd@raspberrypi at May 24 14:46:53 ...
[... cut ...]
 kernel:[  135.353230] 1fa0: 8000fb40 8002495c 76fdf068 7ef7b9a0 01200011 00000000 00000000 00000000

Message from syslogd@raspberrypi at May 24 14:46:53 ...
 kernel:[  135.366925] 1fc0: 76fdf068 7ef7b9a0 76f88000 00000078 76fdf000 00000000 000003af 7ef7b9dc

Message from syslogd@raspberrypi at May 24 14:46:53 ...
 kernel:[  135.380620] 1fe0: 76fdf4c0 7ef7b9a0 00000000 76eea4ac 60000010 01200011 00000000 00000000

Message from syslogd@raspberrypi at May 24 14:46:53 ...
 kernel:[  135.453219] Code: e7905002 e3550000 0a00004c e5943014 (e795c003)"

olllivier, 2016-05-25:
I have workaround on this problem- remove from kernel module uinput module (on the side python). After removed this mod problem is not still occure.

P33M, 2017-05-04:
@olllivier is your issue still present on an updated Pi?

closed 2017-09-13 by JamesH65