raspberrypi/linux #1494: Bluetooth D-Bus config is missing a policy section that enable root-less access

WayneKeenan, 2016-05-26:
In the Feb'16 Raspbian release it was possible to use the bluetooth DBus API without having to be root or using `sudo`.  
The latest Raspbian (May'16) has a `policy` section missing from `/etc/dbus-1/system.d/bluetooth.conf`

```xml
<!-- allow users of bluetooth group to communicate -->
<policy group="bluetooth">
    <allow send_destination="org.bluez"/>
</policy>
```
More info in a follow-up here https://github.com/ukBaz/python-bluezero/issues/30

Ruffio, 2017-01-26:
@pelwell Do you know if this (removal of policy) is by design?

pelwell, 2017-01-26:
This is a Raspbian question rather than a kernel issue, but since it's been sitting here for so long:

I doubt it would have been removed intentionally by us, but @XECDesign and @spl237 might know otherwise.

spl237, 2017-01-26:
Looking at the current release, the section you refer to above is in bluetooth.conf anyway - 

`  
  <policy group="bluetooth">
    <allow send_destination="org.bluez"/>
  </policy>

`

Ruffio, 2017-02-05:
@WayneKeenan  this has been resolved? Right?

JamesH65, 2017-05-18:
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65